realsenseai/realsense-ros #2400: Robot Localization ekf_node: frame ID "camera_imu_optical_frame" does not exist

RNorlie, 2022-07-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515  | 
| Operating System & Version |   Linux (Ubuntu 20.04LTS) Jetpack 5.01| 
| ROS2 Distro | Foxy (Built from source) |
| Kernel Version                        |  5.10.65-tegra                                         | 
| Platform                                 | NVIDIA Jetson Xavier NX         |
| Architecture | Arm64 |
| SDK Version                            |  2.50.0                           | 

Hello,

I'm working to set up odometry for my robot that utilizes the L515 and its internal imu. I don't have my wheel odometry set up yet, but I wanted to see if I could run the ekf_node from the robot localization package with just the imu first, but I encountered a barrage of warnings when I launched the node in conjunction with the realsense camera node. This message was repeatedly printed in the terminal:

```
[ekf_node-1] Warning: Invalid frame ID "camera_imu_optical_frame" passed to canTransform argument source_frame - frame does not exist
[ekf_node-1]      	at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
```

My ekf.yaml config file is as follows:

```
### ekf config file ###
ekf_filter_node:
	ros__parameters:
    	frequency: 30.0
    	sensor_timeout: 0.1
    	two_d_mode: false
    	transform_time_offset: 0.0
    	transform_timeout: 0.0
    	print_diagnostics: true
    	debug: false
    	debug_out_file: /path/to/debug/file.txt
    	publish_tf: true
    	publish_acceleration: false
    	map_frame: map          	# Defaults to "map" if unspecified
    	odom_frame: odom        	# Defaults to "odom" if unspecified
    	base_link_frame: camera_link  # Defaults to "base_link" if unspecified
    	world_frame: odom       	# Defaults to the value of odom_frame if unspecified

    	odom0: example/odom
    	odom0_config: [true,  true,  false,
                   	false, false, false,
                   	false, false, false,
                   	false, false, true,
                   	false, false, false]
    	odom0_queue_size: 2
    	odom0_nodelay: false
    	odom0_differential: false
    	odom0_relative: false
    	odom0_pose_rejection_threshold: 5.0
    	odom0_twist_rejection_threshold: 1.0

    	odom1: example/odom2
    	odom1_config: [false, false, true,
                   	false, false, false,
                   	false, false, false,
                   	false, false, true,
                   	false, false, false]
    	odom1_differential: false
    	odom1_relative: true
    	odom1_queue_size: 2
    	odom1_pose_rejection_threshold: 2.0
    	odom1_twist_rejection_threshold: 0.2
    	odom1_nodelay: false
    	pose0: example/pose
    	pose0_config: [true,  true,  false,
                   	false, false, false,
                   	false, false, false,
                   	false, false, false,
[... 168 lines cut ...]
- 0.01
linear_acceleration:
  x: -0.019613299518823624
  y: -9.639936447143555
  z: 0.16671304404735565
linear_acceleration_covariance:
- 0.01
- 0.0
- 0.0
- 0.0
- 0.01
- 0.0
- 0.0
- 0.0
- 0.01

```

Notice the **frame_id** is notated as **camera_imu_optical_frame**. 

So my question is how can I help the ekf_node see my imu data when the frames are currently split into camera_accel_optical_frame and camera_gyro_optical_frame?

I have tried using different values for **unite_imu_method**, but both copy and linear interpolation result in the same warning from the ekf_node terminal.

Thank you!

closed 2022-07-18 by RNorlie

madgrizzle, 2022-08-20:
I'm having this same issue and curious as to how to solve it.

RNorlie, 2022-08-20:
> I'm having this same issue and curious as to how to solve it.

@madgrizzle,
Using the complementary or madgwick filter from [CCNYRoboticsLab/imu_tools](https://github.com/CCNYRoboticsLab/imu_tools) generates the desired frame!

madgrizzle, 2022-08-20:
Thanks.  There's still a lot to learn about ROS (and ROS2 in particular) and I managed to get it to work with your lead.  It seems odd that its not generated by the realsense ros package.

adamanov, 2022-08-22:
> I'm having this same issue and curious as to how to solve it.

alternatively you can edit your URDF model of realsense camera by adding that frame attached to camera_gyro_optical_frame

madgrizzle, 2022-08-22:
@adamanov, @RNorlie was the OP.  But, since you bring it up, this is what's in the example on this repo:

[image]

https://github.com/IntelRealSense/realsense-ros/blob/e4938bbbfd7b285fe71f81fe69cebcc76dbf72fc/realsense2_camera/launch/opensource_tracking.launch#L42

Is this correct or is that for the T265 and not the D435i?

adamanov, 2022-08-22:
> @adamanov, @RNorlie was the OP. But, since you bring it up, this is what's in the example on this repo:
>
> [image] [...]

@madgrizzle  sorry for the wrong mentioning. Yes, I saw them but was not sure if they were correct.