realsenseai/realsense-ros #2403: D405 Color stream

levmarki, 2022-07-08:
Hello all,

I'm new with the D405 (worked before on D435 for a long time).
How can I get through ROS wrapper (I have Dashing right now) get the **color** stream from both cameras?
As I see it now, I can get only one camera **color** stream aside both camera "infra" stream, that comes - I guess - from the same cameras, but afrer RGB2Gray conversion.

Thank you in advance!

MartyG-RealSense, 2022-07-08:
Hi @levmarki  RealSense camera models that support color from the infrared sensor (D405, D415, D455) only provide it on the left sensor, not both left and right.

levmarki, 2022-07-08:
> Hi @levmarki  RealSense camera models that support color from the infrared sensor (D405, D415, D455) only provide it on the left sensor, not both left and right.

Thank you @MartyG-RealSense !
Is there some way to get it **not** using ROS? Through pyrealsense or some other way?
And another question, if I use only D401 cameras module, can I connect it through MIPI to Rpi/Jetson and have the both cameras together?

Thank you once more!

MartyG-RealSense, 2022-07-08:
Color from right infrared is also not supported in the librealsense SDK and is therefore unavailable in pyrealsense2.

 Access to the camera sensors through MIPI is not supported at this time, unfortunately.

levmarki, 2022-07-08:
> Color from right infrared is also not supported in the librealsense SDK and is therefore unavailable in pyrealsense2.
> 
>  Access to the camera sensors through MIPI is not supported at this time, unfortunately.

Ok, understood, thank you.

It this on plan to provide the access to both cameras in the future?
By the way, does the right camera have color channel at all?

MartyG-RealSense, 2022-07-08:
I will enquire to my Intel RealSense colleagues about your question regarding the possibility of future support of both left and right color from infrared.

The right sensor has no RGB color.  It is possible to stream RGB8 from left infrared and RGB8 from RGB on the D405, though they are the same viewpoint rather than one of the images being offset from the left infrared sensor position like Infra2 is.

[image]

levmarki, 2022-07-11:
Thank you, @MartyG-RealSense !
Looking forward for your's colleagues' answer.

MartyG-RealSense, 2022-07-11:
Hi again @levmarki  My colleagues have informed me that there are no plans to add support for color from Infrared 2.  Thanks very much for your patience and understanding!

levmarki, 2022-07-12:
Thank you, @MartyG-RealSense!
It was very informative and helpful!

Zephyr69, 2022-09-13:
Very useful and important note on the D405!

gcaccia1, 2023-05-31:
> Color from right infrared is also not supported in the librealsense SDK and is therefore unavailable in pyrealsense2.
> 
> Access to the camera sensors through MIPI is not supported at this time, unfortunately.

@MartyG-RealSense do you have any update on the possibility to access the D401 sensors through mipi?
Thanks,
Guido