realsenseai/realsense-ros #2407: Problems with topics /camera/aligned_depth_to_infra1/*

IzzyCoder, 2022-07-11:
Hello,
I'm using RealSense-435i camera on Ubuntu 20.04 with ROS Galactic.
I'm running the the camera node with following command line:
`ros2 launch realsense2_camera rs_launch.py align_depth.enable:=true enable_depth:=true enable_infra1:=true enable_infra2:=true`
The topic list is:
```
/camera/aligned_depth_to_color/camera_info
/camera/aligned_depth_to_color/image_raw
/camera/aligned_depth_to_color/image_raw/compressed
/camera/aligned_depth_to_color/image_raw/compressedDepth
/camera/aligned_depth_to_infra1/camera_info
/camera/aligned_depth_to_infra1/image_raw
/camera/aligned_depth_to_infra1/image_raw/compressed
/camera/aligned_depth_to_infra1/image_raw/compressedDepth
/camera/color/camera_info
/camera/color/image_raw
/camera/color/image_raw/compressed
/camera/color/image_raw/compressedDepth
/camera/color/metadata
/camera/depth/camera_info
/camera/depth/image_rect_raw
/camera/depth/image_rect_raw/compressed
/camera/depth/image_rect_raw/compressedDepth
/camera/depth/metadata
/camera/extrinsics/depth_to_color
/camera/extrinsics/depth_to_infra1
/camera/extrinsics/depth_to_infra2
/camera/imu
/camera/infra1/camera_info
/camera/infra1/image_rect_raw
/camera/infra1/image_rect_raw/compressed
/camera/infra1/image_rect_raw/compressedDepth
/camera/infra1/metadata
/camera/infra2/camera_info
/camera/infra2/image_rect_raw
/camera/infra2/image_rect_raw/compressed
/camera/infra2/image_rect_raw/compressedDepth
/camera/infra2/metadata
/parameter_events
/rosout
/tf_static

```
I'm interested in `/camera/aligned_depth_to_infra1/*` topics, however there seem to be no messages published on those topics - ros2 topic hz/echo commands give nothing. Topics `/camera/aligned_depth_to_color/*` ,on the other hand, work just fine.

Am I missing something?
Thanks in advance.

MartyG-RealSense, 2022-07-11:
Hi @IzzyCoder  There is one previous past case of this issue at https://github.com/IntelRealSense/realsense-ros/issues/1060 regarding no messages being published for the **/camera/aligned_depth_to_infra1** topic when align_depth is true.  It was thought that the cause of the problem might be the kernel version that was being used.  Which kernel version do you have, please?

IzzyCoder, 2022-07-11:
> Hi @IzzyCoder There is one previous past case of this issue at #1060 regarding no messages being published for the **/camera/aligned_depth_to_infra1** topic when align_depth is true. It was thought [...]

Hi,
My kernel version is `5.13.0-52-generic`.

IzzyCoder, 2022-07-11:
Thanks for your help!

MartyG-RealSense, 2022-07-18:
Hi @IzzyCoder  Do you require further assistance with this case, please?  Thanks!

IzzyCoder, 2022-07-19:
Hi @MartyG-RealSense 
I’ve ran recently into the same issue on Jetson NX, I’m assuming for the same reasons.
Other than that, no, thank you.

MartyG-RealSense, 2022-07-19:
Thanks very much for the update.  Have you tried the methods suggested in https://github.com/IntelRealSense/realsense-ros/issues/2407#issuecomment-1180487683 please (using kernel 5.11, applying the draft kernel patch or using an RSUSB source code build of librealsense)

IzzyCoder, 2022-07-26:
@MartyG-RealSense, not for now, thanks!

MartyG-RealSense, 2022-07-26:
Thanks very much @IzzyCoder for the update!  As you do not require further asssistance at this time, I will close this case.  Thanks again!

closed 2022-07-26 by MartyG-RealSense