realsenseai/realsense-ros #2412: Official method to build ros-kinetic-*realsense*.deb file?

byhang, 2022-07-14:
Dear Intel,
        Is there a official method to build ros-kinetic-*realsense***.deb** file?
        I wanna build **kinetic** librealsense**2-2.50.0** and realsense-**2.3.2** from source code in Ubuntu16，but no good idea.

MartyG-RealSense, 2022-07-14:
Hi @byhang  Does using the command below to install librealsense SDK 2.50.0 and ROS1 wrapper 2.3.2 together at the same time from Debian packages work for you, please?

**sudo apt-get install ros-$ROS_DISTRO-realsense2-camera**

https://github.com/IntelRealSense/realsense-ros#method-1-the-ros-distribution

**$ROS_DISTRO** should enable the installation process to detect that you are using ROS Kinetic and install the appropriate packages.

MartyG-RealSense, 2022-07-14:
@byhang  I see that you raised this question a couple of weeks ago at https://github.com/IntelRealSense/realsense-ros/issues/2394#issuecomment-1171084084 and https://github.com/IntelRealSense/librealsense/issues/10624#issuecomment-1167123479

It reminded me that ROS Kinetic was made End of Life (EOL) by ROS around the time of wrapper 2.3.0.   

https://discourse.ros.org/t/preparing-for-final-kinetic-sync/19805

You should still be able to build the 2.3.2 wrapper for Kinetic from source code using the **Method 2** instructions of the ROS wrapper.

https://github.com/IntelRealSense/realsense-ros#step-2-install-intel-realsense-ros-from-sources

byhang, 2022-07-15:
Thank you very much@MartyG-RealSense,
I try build deb use dpkg-deb after catkin_nake install.
But i have a little question.
what is 0 of 0xenial in ros-kinetic-librealsense2_2.50.0-0xenial_amd64.deb mean?
what is 1 of 1focal in ros-noetic-librealsense2_2.50.0-1focal.20211115.134455_amd64.deb mean?

byhang, 2022-07-15:
get√
Thank you dear MartyG @MartyG-RealSense

closed 2022-07-15 by byhang