realsenseai/realsense-ros #2415: Python Dig Tutorial on point_cloud_registration not working

Fyfth, 2022-07-15:
I was following this tutorial : https://pythondig.com/r/python-package-uses-a-turtlebot-robot-with-an-intel-realsense-di-depth-camera-and-tries-to-create-point-clouds-of-gazebo-environments
I got to this command: rosdep install -i --from-path src --rosdistro $ROS_DISTRO -y
The code errored and said : ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
realsense2_description: Cannot locate rosdep definition for [catkin]
realsense2_camera: Cannot locate rosdep definition for [ddynamic_reconfigure]

I am quite new to this but I know this step is critical to progress down the tutorial. 

Thanks for helping in advance!

MartyG-RealSense, 2022-07-15:
Hi @Fyfth  The official version of the ROS wrapper instructions for installing the ROS2 wrapper from source code recommend including a **--skip-keys** instruction.

**rosdep install -i --from-path src --rosdistro $_ros_dist --skip-keys=librealsense2 -y**

https://github.com/IntelRealSense/realsense-ros/tree/ros2#step-4-install-dependencies

Fyfth, 2022-07-15:
Ok thanks I will follow the instructions on the official git repository! thx

Fyfth, 2022-07-15:
Hello! I entered in the code and re done the tutorial once over but this time with the code provided. It still is not working. 

rosdep install -i --from-path src --rosdistro $_ros_dist --skip-keys=librealsense2 -y
WARNING: given --rosdistro --skip-keys=librealsense2 but ROS_DISTRO is "foxy". Ignoring environment.
WARNING: ROS_PYTHON_VERSION is unset. Defaulting to 3
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
realsense2_description: Cannot locate rosdep definition for [rosunit]
realsense2_camera: Cannot locate rosdep definition for [librealsense2]

Am I missing something? Thx for the patience

MartyG-RealSense, 2022-07-15:
On another case where this problem occurred, the developer of the ROS wrapper suggested changing $_ros_dist to $ROS_DISTRO 

**rosdep install -i --from-path src --rosdistro $ROS_DISTRO --skip-keys=librealsense2 -y**

closed 2022-07-28 by MartyG-RealSense