realsenseai/realsense-ros #2416: Pointcloud with infrared intensity values being all in the e-43 range

gcaccia1, 2022-07-16:
Hi everyone, 

I am following up on @FrankLin9981 's last comment about the intensity values being all in the e-43 range. How can I specify the mapping format (e.g. Y8) for the texture so that they appear in the 256 range as if I was subscribing to the /infra/image_raw topic?

@doronhi 
I am using L515, ROS Noetic, Wrapper 2.0, SDK v2.50.

Best,
Guido

_Originally posted by @gcaccia1 in https://github.com/IntelRealSense/realsense-ros/issues/1046#issuecomment-1126345110_

MartyG-RealSense, 2022-07-16:
Hi @gcaccia1  Are you able to achieve your goal if you edit the launch file to set **pointcloud_texture_stream** to use `default="RS2_STREAM_INFRARED` instead of the default RS2_STREAM_COLOR setting for this parameter, please?  The infrared format in the ROS wrapper is Y8.

Here is the original line in the rs_camera launch file for ROS1:

https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/rs_camera.launch#L47

gcaccia1, 2022-07-16:
@MartyG-RealSense thank you for your reply! 
pointcloud_texture_stream is already set as RS2_STREAM_INFRARED.
I can see the IR intensity values attached to the pointcloud, but they are all in the e-43 range and not in the Y8 scale. This is both in rviz and checking manually the intensity values (on the XYZI matrix) with a script that subscribes to the pointcloud2 topic.
It seems like the Y8 format is applied to the IR raw data when published as an image, but not when attached to a pointcloud.

Any ideas?
Thanks,
Guido

MartyG-RealSense, 2022-07-16:
There are not any references to e-43 on this support forum other than at https://github.com/IntelRealSense/realsense-ros/issues/1046#issuecomment-1126345110 unfortunately.  The other discussion about obtaining a point cloud with intensity information is at https://github.com/IntelRealSense/realsense-ros/issues/1386

gcaccia1, 2022-07-16:
@MartyG-RealSense thank you for your reply! 
I am sorry I closed by mistake!

pointcloud_texture_stream is already set as RS2_STREAM_INFRARED.
I can see the IR intensity values attached to the pointcloud, but they are all in the e-43 range and not in the Y8 scale. This is both in rviz and checking manually the intensity values (on the XYZI matrix) with a script that subscribes to the pointcloud2 topic.
It seems like the Y8 format is applied to the IR raw data when published as an image, but not when attached to a pointcloud.

Any ideas?
Thanks,
Guido

gcaccia1, 2022-07-16:
Thank you I will look into this a bit more and send updates.

Guido

MartyG-RealSense, 2026-10-23:
This open GitHub issue is being closed because the L515 model is End of Life (EOL) and no longer supported.  We welcome the creation of new issues relating to supported cameras.

closed 2026-10-23 by MartyG-RealSense