realsenseai/realsense-ros #2418: Float RGB pixel get depth

Michsh, 2022-07-19:
Hi.
 Is there a solution to use float RGB pixels in de projection of pixels to point ?
I think float pixels will increase accuracy.
Thanks

MartyG-RealSense, 2022-07-19:
Hi @Michsh  Are you are asking about the librealsense SDK instruction **rs2_deproject_pixel_to_point** please?

MartyG-RealSense, 2022-07-26:
Hi @Michsh  Do you require further assistance with this case, please?  Thanks!

Michsh, 2022-07-26:
Hi , i need assistance or an example of using  rs2_deproject_pixel_to_point  with foat pixels, int pixels lose some accuracy.
 Thanks !

MartyG-RealSense, 2022-07-26:
If your project is using depth-color alignment then int pixels may not be the problem in regard to accuracy, as rs2_deproject_pixel_to_point naturally has some inaccuracy if it is used with alignment.  Using **aligned intrinsics** instead of depth intrinsics can help, as advised by a RealSense team member at https://github.com/IntelRealSense/librealsense/issues/4315

MartyG-RealSense, 2022-08-01:
Hi @Michsh  Was the advice in the comment above helpful to you, please?  Thanks!

closed 2022-08-14 by MartyG-RealSense