realsenseai/realsense-ros #2420: cant record "Motion Module" on realsense-viewer V2.50.0 with D435i ?!

Tcleslie, 2022-07-19:
I was trying to record data by realsense-viewer,  it did work when I turn on "Stereo Module" and "RGB camera". But when I tried to turn on "Motion Module", the realsense-viewer immediately shutdown and quit. It showed up a warning and a segmentation fault on teminal  as below. Why dose the problem occer and how can I fix it?  I will appreciate all the answers.

`
WARNING [0x1140c9600] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
zsh: segmentation fault  sudo ./realsense-viewer
`

MartyG-RealSense, 2022-07-19:
Hi @Tcleslie  There is a known issue for some RealSense users with realsense-viewer where enabling depth, RGB and Motion Module simultaneously causes problems.  Most commonly this takes the form of no image on the RGB stream and the message No Frames Received, though more rarely the depth stream or the IMU streams may stop working.

There is not a fix for this issue in realsense-viewer and changing the order that the streams are enabled in does not resolve the problem as the enabling of the Motion Module is the trigger.  The workaround is to enable only two of the three streams simultaneously (such as depth-RGB, depth-IMU or RGB-IMU).

It is possible to display all three streams simultaneously in the **rs-capture** SDK example program as it is not affected by the Motion Module issue.

When creating your own application, a workaround for the issue on Python is to place the streams on two separate pipelines (IMU alone on one pipeline and depth + RGB on the other pipeline).  In C++ the issue can be worked around by using **callbacks**.

Tcleslie, 2022-07-20:
Hi @MartyG-RealSense,  I really appreciate for your timely response. I think I have got your idea. Now since it is not a fixed issue, maybe I should try some way else. Thanks again.

closed 2022-08-01 by MartyG-RealSense