realsenseai/realsense-ros #2422: Simulated d435 model does not work

PhilipAmadasun, 2022-07-23:
I have installed realsense2_description and wish to use it for my simulated turtlebor2. I can see the d435 camera model connected to my turtelbot2in gazebo but the /scan topic does not receive any data so gmapping doesnt work. Is there some parameter in some file somewhere that I'm supposed to change?

MartyG-RealSense, 2022-07-23:
Hi @PhilipAmadasun  May I first ask whether the /scan topic works if you perform roslaunch with the RealSense ROS wrapper's **rs_d435_camera_with_model** launch file?  For example:

`roslaunch realsense2_camera rs_d435_camera_with_model.launch`

PhilipAmadasun, 2022-07-23:
@MartyG-RealSense  When I just run what you suggested on it's own, rviz opens up and I ee the model pf the camera but on the terminal:

```
PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/confidence_fps: 30
 * /camera/realsense2_camera/confidence_height: 480
 * /camera/realsense2_camera/confidence_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_rgb: False
 * /camera/realsense2_camera/infra_width: 640
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: 
[... 29 lines cut ...]
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [3538]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 6135ba68-0ac5-11ed-a576-d43b04207196
process[rosout-1]: started with pid [3551]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [3554]
process[camera/realsense2_camera-3]: started with pid [3555]
process[robot_state_publisher-4]: started with pid [3556]
process[rviz-5]: started with pid [3557]
[ INFO] [1658607842.797853041]: Initializing nodelet with 12 worker threads.
[ INFO] [1658607843.944294542]: RealSense ROS v2.3.0
[ INFO] [1658607843.944333571]: Built with LibRealSense v2.45.0
[ INFO] [1658607843.944353220]: Running with LibRealSense v2.45.0
[ WARN] [1658607844.095673901]: No RealSense devices were found!
[ WARN] [1658607850.118074833]: No RealSense devices were found!
[ WARN] [1658607856.128003039]: No RealSense devices were found!
[ WARN] [1658607862.140627946]: No RealSense devices were found!
[ WARN] [1658607868.165867939]: No RealSense devices were found!
[ WARN] [1658607874.176789030]: No RealSense devices were found!

```and rostopic echo /scan gives me nothing

MartyG-RealSense, 2022-07-24:
Thanks very much @PhilipAmadasun for your patience.

Are you using **depthimage_to_laserscan** to publish the /scan topic, please?

http://wiki.ros.org/depthimage_to_laserscan

MartyG-RealSense, 2022-07-24:
Thanks very much @PhilipAmadasun four your patience.

After researching your issue, I concluded that the best place to start would be to try to resolve the **No RealSense devices were found!** warnings at the bottom of your log.

Please first try adding **initial_reset:=true** to your roslaunch instruction or setting initial_reset to True instead of false in your launch file.  This will reset the camera hardware at launch.

PhilipAmadasun, 2022-07-24:
@MartyG-RealSense  I added initial_reset:=true and still get the same problem. I see the model on rviz but no realsense devices found. Just a reminder I want to get the simulated camera to work.

MartyG-RealSense, 2022-07-24:
Are you using a Gazebo plugin for D435 like the one at https://github.com/pal-robotics/realsense_gazebo_plugin

The subject of RealSense Gazebo plugins is discussed further at https://github.com/IntelRealSense/realsense-ros/issues/1170

PhilipAmadasun, 2022-07-24:
@MartyG-RealSense No I have not, if I clone this to my workspace or something the camera starts working?

MartyG-RealSense, 2022-07-24:
I have not tried a RealSense Gazebo plugin myself, so I do not have advice about what the outcome of using one may be.

PhilipAmadasun, 2022-07-24:
so then the if the camera can work without that package what could be the
other possible problems?

MartyG-RealSense, 2022-07-24:
I will continue to work with you on this problem tomorrow.  Please do update me about any progress that you make in the meantime.  Thanks for your patience!

MartyG-RealSense, 2022-07-25:
Thanks very much for your patience.  I conducted extensive further research but the number of different elements involved in this particular case (RealSense, Gazebo, gmapping, turtlebot2 and no data on the /scan topic) means that there are no references that match up closely to your problem.

Some references recommended that if a simulated camera was being used with Gazebo and /scan had no data then try resetting Gazebo, as Gazebo might nt provide data for the /scan topic on the first attempt.  An example of this advice is at:

https://get-help.robotigniteacademy.com/t/ros-basics-real-robot-project-nothing-published-to-scan/11575/3

MartyG-RealSense, 2022-07-31:
Hi @PhilipAmadasun  Do you require further assistance with this case, please?  Thanks!

PhilipAmadasun, 2022-08-01:
I need to try out some of your advice this monday and get back to you.

PhilipAmadasun, 2022-08-02:
@MartyG-RealSense  I installed the gazebo_realsense_plugin thing in my realsense_ws workspace. I don't know how exactly to use it. How will this help my simulated d435?

MartyG-RealSense, 2022-08-02:
The plugin has a pal-robotics GitHub support forum like this one.

https://github.com/pal-robotics/realsense_gazebo_plugin/issues/

This case in particular from that forum may be helpful: https://github.com/pal-robotics/realsense_gazebo_plugin/issues/7

MartyG-RealSense, 2022-08-08:
Has there been any progress on your Gazebo-related issue please?

PhilipAmadasun, 2022-08-09:
@MartyG-RealSense  I follow the instructions in the forum and I'm sorta stuck on the last step of what urdf file to change. I cam accross this file "turtlebot_gazebo.urdf.xacro":
```
<?xml version="1.0"?>
<robot name="turtlebot_gazebo" xmlns:xacro="http://ros.org/wiki/xacro">
  <!-- Microsoft Kinect / ASUS Xtion PRO Live for simulation -->
  <xacro:macro name="turtlebot_sim_3dsensor">
    <gazebo reference="camera_link">  
      <sensor type="depth" name="camera">
        <always_on>true</always_on>
        <update_rate>20.0</update_rate>
        <camera>
          <horizontal_fov>${60.0*M_PI/180.0}</horizontal_fov>
          <image>
            <format>B8G8R8</format>
            <width>640</width>
            <height>480</height>
          </image>
          <clip>
            <near>0.05</near>
            <far>8.0</far>
          </clip>
        </camera>
        <plugin name="kinect_camera_controller" filename="libgazebo_ros_openni_kinect.so">
          <cameraName>camera</cameraName>
          <alwaysOn>true</alwaysOn>
          <updateRate>10</updateRate>
          <imageTopicName>rgb/image_raw</imageTopicName>
          <depthImageTopicName>depth/image_raw</depthImageTopicName>
          <pointCloudTopicName>depth/points</pointCloudTopicName>
          <cameraInfoTopicName>rgb/camera_info</cameraInfoTopicName>
          <depthImageCameraInfoTopicName>depth/camera_info</depthImageCameraInfoTopicName>
          <frameName>camera_depth_optical_frame</frameName>
          <baseline>0.1</baseline>
          <distortion_k1>0.0</distortion_k1>
          <distortion_k2>0.0</distortion_k2>
          <distortion_k3>0.0</distortion_k3>
          <distortion_t1>0.0</distortion_t1>
          <distortion_t2>0.0</distortion_t2>
          <pointCloudCutoff>0.4</pointCloudCutoff>
        </plugin>
      </sensor>
    </gazebo>
  </xacro:macro>
</robot>
```
Should I change the plugin here, which seems to be a controller for a kinect camera, would that change anything?

MartyG-RealSense, 2022-08-09:
The ROS wrapper provides a set of urdf.xacro files for RealSense cameras at the link below.

https://github.com/IntelRealSense/realsense-ros/tree/development/realsense2_description/urdf

The file for the D435 camera model is **_d435.urdf.xacro**

I do not know if changing the controller from the Kinect one makes a difference.  You could try changing the urdf.xacro file first and see what difference it makes.

We mentioned plugins earlier in this discussion at https://github.com/IntelRealSense/realsense-ros/issues/2422#issuecomment-1193393072

MartyG-RealSense, 2022-08-15:
Hi @PhilipAmadasun  Was the comment above helpful to you, please?  Thanks!

MartyG-RealSense, 2022-08-26:
Case closed due to no further comments received.

closed 2022-08-26 by MartyG-RealSense

PhilipAmadasun, 2022-08-31:
@MartyG-RealSense  Sorry for late respnse, I still have issues with the simulated gazebo, I realize that my rostopic list doesnt even show any camera topics. So I'm guessing I need to use that realsense gazebo plugin package. I built it in a workspace but have no clue how to use it.The repo does not give any instructions on how. would you know?

MartyG-RealSense, 2022-09-01:
A user of the RealSense Gazebo plugin created their own instructions at their forked GitHub version of the plugin.

https://github.com/AzulRadio/realsense_gazebo_plugin

These were based on instructions at the pal-robotics support forum at https://github.com/pal-robotics/realsense_gazebo_plugin/issues/7