realsenseai/realsense-ros #2423: [ROS1 Melodic ]When "rs_multiple_devices.launch" is launched, "Hardware Notification: Depth stream start failure,1.65875e+12,Error, Hardware Error" and then freezes.

ImaiKota, 2022-07-25:
When I run

`roslaunch realsense2_camera rs_multiple_devices.launch`

it displays

> **Hardware Notification: Depth stream start failure,1.65875e+12,Error, Hardware Error**

and then the Laptop is frozen.

**Environment:**
- Camera model:  D435i 
- OS: Ubuntu 18.04 
- Kernel version: 5.4.0-100-generic
- ROS: Melodic
- RealSense SDK: I installed it by
 `sudo apt-get install librealsense2-dkms`
 `sudo apt-get install librealsense2-utils`
  `sudo apt-get install librealsense2-dev`
  `sudo apt-get install librealsense2-dbg`
- realsense-ros: I installed it by
`sudo apt-get install ros-melodic-realsense2-camera`
`sudo apt-get install ros-melodic-realsense2-description`

The following is a screen shot of the warning message.
[image]

ImaiKota, 2022-07-25:
I tried to run it with **initial_reset:=true** option, but the situation did not change.

ImaiKota, 2022-07-25:
The error **"Hardware Notification: Depth stream start failure,1.65875e+12,Error, Hardware Error"** sometimes appeared even when running 

`roslaunch realsense2_camera rs_camera.launch`

with a single D435i.

MartyG-RealSense, 2022-07-25:
Hi @ImaiKota  I note that you first installed the librealsense SDK from packages and then used the RealSense ROS wrapper's Method 1 instructions, which install both librealsense and the ROS wrapper from packages at the same time.  The librealsense SDK will therefore have been installed twice, which may lead to conflicts between the installations.

I would recommend first uninstalling all RealSense related packages and then re-performing the Method 1 procedure to install librealsense and the ROS wrapper together from packages.

RealSense-related packages can be uninstalled with the instruction below:

**dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge**

ImaiKota, 2022-07-26:
After uninstalling all packages related to realsense, I installed the ros package and SDK using method 1 and the problem was solved. Thank you very much!.

ImaiKota, 2022-07-26:
Sorry, the same problem occurred again.

ImaiKota, 2022-07-26:
After running the launch file with **initial_reset:=true**, I no longer encounter the error. 
Is this a good solution?

MartyG-RealSense, 2022-07-26:
Yes, using initial_reset:=true is a fine solution to use to resolve your problem.

ImaiKota, 2022-07-26:
Okay, Thanks!

closed 2022-07-27 by ImaiKota