realsenseai/realsense-ros #2424: Include Camera's Rear Mounting Points in TF

acbuynak, 2022-07-25:
I am looking for a URDF defined TF frame for the rear mounting points on the camera body. Checking a few camera models, I do not see this frame currently added. 

Is this intentional? Am I just not finding the frame by accident?

### Why Needed?
I think many users use the back plane of the camera body as a flush mounting surface. Often the screw mount locations are already defined in CAD and it would be easy to define the camera's mount position in the URDF.

### Documentation
Below is a quick example of two cameras from the D400 series, both of which include a TF frame for the bottom 1/4-20 mount but do not for the rear mounting holes. 
| Iso View | Top View |
| ----- | ------- |
| [image]  |[image] |

[image]

RealSenseCustomerSupport, 2022-09-09:
----------------------------------------------

Hi
What about the STEP files for D435/D435i and D455? they all have the FOV cones and mounting holes in 3D models. Are they helpful?

D455 STEP file:
https://www.intelrealsense.com/wp-content/uploads/2020/10/Intel_RealSense_D455_CAD_external_v2.zip?_ga=2.196577839.2042015434.1662738414-419342904.1659560894

D435/D435i STEP file: see the attached.

Thanks

Attachment(s):
d435_external_release_asm_v3.zip - https://realsensesupport.intel.com/attachments/token/wxyeZ9I6YKUJNTk7qG8CsSHlS/?name=d435_external_release_asm_v3.zip

acbuynak, 2022-09-12:
Hello! Yes, while the mounting holes are detailed in the 3D models a TF transform is not published at the 1/4-20 mounting head. This could be detailed in the URDF. Does that make sense?

zacharyyamaoka, 2024-03-28:
I would find this helpful if this was included in the URDF by default. Here are the values I measured for the D435.

[image]

zacharyyamaoka, 2024-03-28:
[image]

MartyG-RealSense, 2026-10-31:
Case closed due to inactivity for over two years.

closed 2026-10-31 by MartyG-RealSense