realsenseai/realsense-ros #2425: control_transfer returned error, index: 768, error: No data available, number: 61

PhilipAmadasun, 2022-07-26:
trying to get d435i camera to work with turtlebot2. For some reason the '/scan' does not get any data. This is what I get when I `roslaunch turtlebot_navigation gmapping_demo.launch`:

```
PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_rgb: False
 * /camera/realsense2_camera/infra_width: 848
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: 
[... 256 lines cut ...]
[ WARN] [1658795585.035680824]: Param '/camera/stereo_module/auto_exposure_limit' has value 0 that is not in range [1, 200000]. Removing this parameter from dynamic reconfigure options.
[ WARN] [1658795585.049658644]: Param '/camera/stereo_module/auto_gain_limit' has value 0 that is not in range [16, 248]. Removing this parameter from dynamic reconfigure options.
[ INFO] [1658795585.213533316]: Done Setting Dynamic reconfig parameters.
[ INFO] [1658795585.215467172]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1658795585.219940414]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1658795585.220032719]: setupPublishers...
[ INFO] [1658795585.233124700]: Expected frequency for depth = 30.00000
[ INFO] [1658795585.350240086]: Expected frequency for color = 30.00000
[ INFO] [1658795585.404484548]: setupStreams...
[ INFO] [1658795585.415673234]: insert Depth to Stereo Module
[ INFO] [1658795585.415784275]: insert Color to RGB Camera
[ INFO] [1658795585.470978728]: SELECTED BASE:Depth, 0
[ INFO] [1658795585.482786048]: RealSense Node Is Up!
[ WARN] [1658795585.591586406]: 
 25/07 17:33:05,592 WARNING [140669272176384] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 25/07 17:33:05,643 WARNING [140669272176384] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 25/07 17:33:05,896 WARNING [140669272176384] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 25/07 17:33:05,947 WARNING [140669272176384] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 25/07 17:33:05,998 WARNING [140669272176384] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 25/07 17:33:06,149 WARNING [140669272176384] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 25/07 17:33:06,301 WARNING [140669272176384] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 25/07 17:34:30,131 WARNING [140669272176384] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

I don't understand this "control transfer return error` I'm assuming it has something to do with the librealsense wrapper?
```

MartyG-RealSense, 2022-07-26:
Hi @PhilipAmadasun  Is this a separate issue to your earlier recent Gazebo case at https://github.com/IntelRealSense/realsense-ros/issues/2422, where you are trying to access a real D435i camera this time instead of a simulated D435 please?

If you are using a live camera and the **control_transfer returned** error that you experienced is generating in the log continously then this can indicate that there is a serious communication problem with the device.

You have already likely tried **initial_reset:=true** as it is referenced in your other case.  Can you confirm that this have been tried with the D435i camera?

Do you experience the same control_transfer returned warnings if you launch with **roslaunch realsense2_camera rs_camera.launch**

PhilipAmadasun, 2022-07-27:
@MartyG-RealSense  Yes I get the same exact issue when I launch this, even with initial_rest:=true

MartyG-RealSense, 2022-07-27:
Do you have another USB cable that you can try with the camera to confirm whether or not the apparent communication problem demonstrated by the control_transfer returned warnings is related to the cable that is currently being used with the camera?

PhilipAmadasun, 2022-07-28:
i still get the same problem

MartyG-RealSense, 2022-07-28:
Next, we should consider the possibility that if a basic **rs_camera** launch does not work then there may be a problem in the installation of the librealsense SDK 2.45.0 and ROS wrapper 2.3.0 that is causing the warnings.  If you are using those versions, it looks as though you installed the SDK and ROS wrapper together from packages on ROS Kinetic using the wrapper's **Method 1** instructions.  Is that correct, please?

If you did use Method 1, did you install librealsense **before** running the Method 1 installation?  If for example you installed librealsense from packages or source code and then ran the Method 1 installation, you would have installed librealsense twice, because Method 1 installs both librealsense and the wrapper.  This could cause a conflict between librealsense installations.

PhilipAmadasun, 2022-07-29:
@MartyG-RealSense  I can't recall but I believe did method 1 first, I honestly can't remember. But you are saying doing [Method1](https://github.com/IntelRealSense/realsense-ros) will also install a wrapper?

MartyG-RealSense, 2022-07-29:
Method 1 uses packages to install both librealsense and the ROS wrapper at the same time, whereas Method 2 (a source code build) only builds the ROS wrapper.  

If you used Method 1 first and then installed librealsense 2.33 from source, it is probable that you have two librealsense installations on the same computer that are conflicting with each other.  Installing from source and packages on the same computer can lead to a **Multiple udev rules detected** problem that could prevent the SDK from working correctly.

If you have multiple udev rules then the simple solution is to delete the rule that is at the **/etc/udev/rules.d/** folder location, as described by a RealSense team member at https://github.com/IntelRealSense/librealsense/issues/6153#issuecomment-605687547

PhilipAmadasun, 2022-08-02:
Hi! So I removed librealsense 2.33 from my laptop redid method 1 from scratch. I still get the same error. I looked at /etc/udev/rules.d, and I have 99-realsense-libusb.rules. Do I delete this? I'm confused. This s the error:```

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_rgb: False
 * /camera/realsense2_camera/infra_width: 848
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/linear_accel_cov: 0.01
[... 131 lines cut ...]
SUBSYSTEMS=="usb", ENV{DEVTYPE}=="usb_device", ATTRS{idVendor}=="8087", ATTRS{idProduct}=="0af3", MODE="0666", GROUP="plugdev"
SUBSYSTEMS=="usb", ENV{DEVTYPE}=="usb_device", ATTRS{idVendor}=="8087", ATTRS{idProduct}=="0b37", MODE="0666", GROUP="plugdev"
SUBSYSTEMS=="usb", ENV{DEVTYPE}=="usb_device", ATTRS{idVendor}=="03e7", ATTRS{idProduct}=="2150", MODE="0666", GROUP="plugdev"

KERNEL=="iio*", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0ad5", MODE:="0777", GROUP:="plugdev", RUN+="/bin/sh -c 'chmod -R 0777 /sys/%p'"
DRIVER=="hid_sensor_custom", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0ad5", RUN+="/bin/sh -c 'chmod -R 0777 /sys/%p && chmod 0777 /dev/%k'"
KERNEL=="iio*", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0af2", MODE:="0777", GROUP:="plugdev", RUN+="/bin/sh -c 'chmod -R 0777 /sys/%p'"
DRIVER=="hid_sensor*", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0af2", RUN+="/bin/sh -c 'chmod -R 0777 /sys/%p && chmod 0777 /dev/%k'"
KERNEL=="iio*", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0afe", MODE:="0777", GROUP:="plugdev", RUN+="/bin/sh -c 'chmod -R 0777 /sys/%p'"
DRIVER=="hid_sensor_custom", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0afe", RUN+="/bin/sh -c 'chmod -R 0777 /sys/%p && chmod 0777 /dev/%k'"
KERNEL=="iio*", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0aff", MODE:="0777", GROUP:="plugdev", RUN+="/bin/sh -c 'chmod -R 0777 /sys/%p'"
[... cut ...]
KERNEL=="iio*", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0b4b", MODE:="0777", GROUP:="plugdev", RUN+="/bin/sh -c 'chmod -R 0777 /sys/%p'"
DRIVER=="hid_sensor*", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0b4b", RUN+="/bin/sh -c ' chmod -R 0777 /sys/%p && chmod 0777 /dev/%k'"
KERNEL=="iio*", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0b4d", MODE:="0777", GROUP:="plugdev", RUN+="/bin/sh -c 'chmod -R 0777 /sys/%p'"
DRIVER=="hid_sensor*", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0b4d", RUN+="/bin/sh -c ' chmod -R 0777 /sys/%p && chmod 0777 /dev/%k'"
```

PhilipAmadasun, 2022-08-02:
There is something wrong wth my "depth_image_to_laserscan" node, its not connected t oanything on my rqt_graph. There is a launch file in turtlebot_bringup called "3dsensor.launch" that activates this node. Maybe you can look at this file and help me. Here it is:
```
<!-- 
  Turtlebot is a bit low on horsepower...

  We use openni_camera here, turn everything on by default
  (allows the user to conveniently see everything when
  launching this on its own - use with 
  turtebot_rviz_launchers/view_robot.launch to visualise)
  and provide args to disable them so you can optimise the
  horsepower for your application.
  
  For an example of disabling processing modules, check
  any of the turtlebot_rapps (e.g. android_make_a_map.launch
  only enables scan_processing for depthimage_to_laserscan and
  rgb_processing for the android tele-view).
-->
<launch>
  <!-- "camera" should uniquely identify the device. All topics are pushed down
       into the "camera" namespace, and it is prepended to tf frame ids. -->
  <arg name="camera"      default="camera"/>
  <arg name="publish_tf"  default="false"/>
  <arg name="3d_sensor"   default="$(env TURTLEBOT_3D_SENSOR)"/>  <!-- kinect, asus_xtion_pro -->

  <!-- Factory-calibrated depth registration -->
  <arg name="depth_registration"              default="true"/>
  <arg     if="$(arg depth_registration)" name="depth" value="depth_registered" />
  <arg unless="$(arg depth_registration)" name="depth" value="depth" />

  <!-- Processing Modules -->
  <arg name="rgb_processing"                  default="true"/>
  <arg name="ir_processing"                   default="true"/>
  <arg name="depth_processing"                default="true"/>
  <arg name="depth_registered_processing"     default="true"/>
  <arg name="disparity_processing"            default="true"/>
  <arg name="disparity_registered_processing" default="true"/>
  <arg name="scan_processing"                 default="true"/>

  <!-- Worker threads for the nodelet manager -->
  <arg name="num_worker_threads" default="4" />

  <!-- Laserscan topic -->
  <arg name="scan_topic" default="scan"/>

  <include file="$(find turtlebot_bringup)/launch/includes/3dsensor/$(arg 3d_sensor).launch.xml">
    <arg name="camera"                          value="$(arg camera)"/>
    <arg name="publish_tf"                      value="$(arg publish_tf)"/>
    <arg name="depth_registration"              value="$(arg depth_registration)"/>
    <arg name="num_worker_threads"              value="$(arg num_worker_threads)" />

    <!-- Processing Modules -->
    <arg name="rgb_processing"                  value="$(arg rgb_processing)"/>
    <arg name="ir_processing"                   value="$(arg ir_processing)"/>
    <arg name="depth_processing"                value="$(arg depth_processing)"/>
    <arg name="depth_registered_processing"     value="$(arg depth_registered_processing)"/>
    <arg name="disparity_processing"            value="$(arg disparity_processing)"/>
    <arg name="disparity_registered_processing" value="$(arg disparity_registered_processing)"/>
  </include>

   <!--                        Laserscan 
     This uses lazy subscribing, so will not activate until scan is requested.
   -->
  <group if="$(arg scan_processing)">
    <node pkg="nodelet" type="nodelet" name="depthimage_to_laserscan" args="load depthimage_to_laserscan/DepthImageToLaserScanNodelet $(arg camera)/$(arg camera)_nodelet_manager">
      <!-- Pixel rows to use to generate the laserscan. For each column, the scan will
           return the minimum value for those pixels centered vertically in the image. -->
      <param name="scan_height" value="10"/>
      <param name="output_frame_id" value="/$(arg camera)_depth_frame"/>
      <param name="range_min" value="0.45"/>
      <remap from="image" to="camera/depth/image_rect_raw"/>
      <remap from="scan" to="$(arg scan_topic)"/>

      <!-- Somehow topics here get prefixed by "$(arg camera)" when not inside an app namespace,
           so in this case "$(arg scan_topic)" must provide an absolute topic name (issue #88).
           Probably is a bug in the nodelet manager: https://github.com/ros/nodelet_core/issues/7 -->
      <remap from="$(arg camera)/image" to="camera/depth/image_rect_raw"/>
      <remap from="$(arg camera)/scan" to="$(arg scan_topic)"/>
    </node>
  </group>
</launch>
```

MartyG-RealSense, 2022-08-02:
If you have a rule at **/lib/udev/rules.d/60-librealsense2-udev-rules.rule** then you can delete the other rule at etc/udev/rules.d

In regard to your second question, I do not have knowledge about use of depth_image_to_laserscan unfortunately.

PhilipAmadasun, 2022-08-04:
@MartyG-RealSense  I dont have any such rules in my lib/udev, how do I make one?

MartyG-RealSense, 2022-08-04:
The SDK installation guide at the link below advises to use these two instructions to install the /etc udev rule after navigating to the root directory of the librealsense source code folder.

**sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/
sudo udevadm control --reload-rules && udevadm trigger**

https://dev.intelrealsense.com/docs/compiling-librealsense-for-linux-ubuntu-guide

PhilipAmadasun, 2022-08-04:
@MartyG-RealSense      Isn't it supposed to be a 60-realsense-libusb..... file and not a 99-realsense-lib?

PhilipAmadasun, 2022-08-05:
@MartyG-RealSense  I also don't have a config file. in my librealsense, its supposed to be in my ros/kinetic/shared directory right?

MartyG-RealSense, 2022-08-05:
If you have a source code build then the udev rule installed is /etc/udev/rules.d/99-realsense-libusb.rules

If you build from packages then the udev rule is /lib/udev/rules.d/60-librealsense2-udev-rules.rules

If both rules are installed on the same computer then the /etc rule is the one that is deleted, because removing the /lib rule can break linkages to packages.

In your case though, you had neither of these rules installed.

If a system is in a tangle, with both package and source code builds and missing / conflicting files, it can be easier just to perform a complete wipe of the computer (including Ubuntu) and re-install everything from the beginning, making sure to only choose one SDK install method this time (such as the ROS wrapper's Method 1).

If it is not possible to wipe the whole computer then remove as much of librealsense as possible.

*. If you build the ROS wrapper from source code, delete the entire catkin workspace folder.

*. If you built librealsense from packages, remove them.

**dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge**

*. If you have a source code installation of librealsense on your computer, go to its root directory and run the CMake instruction below to uninstall it.

**sudo make uninstall && make clean**

PhilipAmadasun, 2022-08-05:
ROS seems to recognize the camera though, its just the depthimage to laser
scan nodelette that doesn't connect to anything, Thus why I can do SLAM.

MartyG-RealSense, 2022-08-06:
Some resources about using D435 with depthimage_to_laserscan are listed at https://github.com/IntelRealSense/realsense-ros/issues/2440#issuecomment-1206655830

PhilipAmadasun, 2022-08-06:
THe initial issue is that the scan topic is not receiving any messages. the depdthimage-to-laserscan nodelet only works when the scan topic is receiving data(as I now realize). How do I get the scan topic topic to recieve data, I think this is the main issue that needs to be solved.

MartyG-RealSense, 2022-08-07:
I found a case at https://github.com/IntelRealSense/realsense-ros/issues/1163 where **move_base** was not receiving depth images when using depthimage_to_laserscan, so the RealSense ROS user in that case created their own adapted 'fork' version of the ROS wrapper that published directly to the /camera/scan topic.

https://github.com/dschnabel/realsense-ros

Because of the age of that case though, the forked wrapper matches to librealsense SDK **2.33.1** instead of 2.50.0, so the edits made to the wrapper would likely have to be recreated in your own fork of 2.50.0 unless you were happy to use the old 2.33.1 SDK.

PhilipAmadasun, 2022-08-07:
@MartyG-RealSense  Okay, because this is the 2.33.1 version I will try to build depthimage_to_laser scan in a workspace. Although, I have the ros-kinetic-depthimage-to-laserscan package in my opt/ros directory(which does not seem to have some of the files [here](https://github.com/ros-perception/depthimage_to_laserscan),do I just remove that and build another one in the workspace? How do I make sure my the launch file that includes the depthimage-to-laserscan I build?
This is my launch file:
```
<launch>
  <!-- "camera" should uniquely identify the device. All topics are pushed down
       into the "camera" namespace, and it is prepended to tf frame ids. -->
  <arg name="camera"      default="camera"/>
  <arg name="publish_tf"  default="false"/>
  <arg name="3d_sensor"   default="$(env TURTLEBOT_3D_SENSOR)"/>  <!-- kinect, asus_xtion_pro -->

  <!-- Factory-calibrated depth registration -->
  <arg name="depth_registration"              default="true"/>
  <arg     if="$(arg depth_registration)" name="depth" value="depth_registered" />
  <arg unless="$(arg depth_registration)" name="depth" value="depth" />

  <!-- Processing Modules -->
  <arg name="rgb_processing"                  default="true"/>
  <arg name="ir_processing"                   default="true"/>
  <arg name="depth_processing"                default="true"/>
  <arg name="depth_registered_processing"     default="true"/>
  <arg name="disparity_processing"            default="true"/>
  <arg name="disparity_registered_processing" default="true"/>
  <arg name="scan_processing"                 default="true"/>

  <!-- Worker threads for the nodelet manager -->
  <arg name="num_worker_threads" default="4" />

  <!-- Laserscan topic -->
  <arg name="scan_topic" default="scan"/>

  <include file="$(find turtlebot_bringup)/launch/includes/3dsensor/$(arg 3d_sensor).launch.xml">
    <arg name="camera"                          value="$(arg camera)"/>
    <arg name="publish_tf"                      value="$(arg publish_tf)"/>
    <arg name="depth_registration"              value="$(arg depth_registration)"/>
    <arg name="num_worker_threads"              value="$(arg num_worker_threads)" />

    <!-- Processing Modules -->
    <arg name="rgb_processing"                  value="$(arg rgb_processing)"/>
    <arg name="ir_processing"                   value="$(arg ir_processing)"/>
    <arg name="depth_processing"                value="$(arg depth_processing)"/>
    <arg name="depth_registered_processing"     value="$(arg depth_registered_processing)"/>
    <arg name="disparity_processing"            value="$(arg disparity_processing)"/>
    <arg name="disparity_registered_processing" value="$(arg disparity_registered_processing)"/>
  </include>

   <!--                        Laserscan
     This uses lazy subscribing, so will not activate until scan is requested.
   -->
  <group if="$(arg scan_processing)">
    <node pkg="nodelet" type="nodelet" name="depthimage_to_laserscan" args="load depthimage_to_laserscan/DepthImageToLaserScanNodelet $(arg camera)/$(arg camera)_nodelet_manager">
      <!-- Pixel rows to use to generate the laserscan. For each column, the scan will
           return the minimum value for those pixels centered vertically in the image. -->
      <param name="scan_height" value="10"/>
      <param name="output_frame_id" value="/$(arg camera)_depth_frame"/>
      <param name="range_min" value="0.45"/>
      <remap from="image" to="/camera/depth/image_rect_raw"/>
      <remap from="scan" to="$(arg scan_topic)"/>

      <!-- Somehow topics here get prefixed by "$(arg camera)" when not inside an app namespace,
           so in this case "$(arg scan_topic)" must provide an absolute topic name (issue #88).
           Probably is a bug in the nodelet manager: https://github.com/ros/nodelet_core/issues/7 -->
      <remap from="$(arg camera)/image" to="/camera/depth/image_rect_raw"/>
      <remap from="$(arg camera)/scan" to="$(arg scan_topic)"/>
    </node>
  </group>
</launch>```
```

MartyG-RealSense, 2022-08-07:
The impression that I get from https://github.com/IntelRealSense/realsense-ros/issues/1163 is that the RealSense ROS user incorporated the depthimage_to_laserscan code into their forked RealSense ROS wrapper so that depthimage_to_laserscan does not have to be launched separately.

If the separate depthimage_to_laserscan package does not appear to be outputting anything then it may be worth first trying leaving the existing package in place and running the forked wrapper to see what is output.

PhilipAmadasun, 2022-08-08:
@MartyG-RealSense  If I build the editted wrapper package in a workspace, how do I make sure the already installed "ros-kinetic-realsense2-camera" does not fight with this newly installed wrapper? wouldn't I have two wrappers then fighting against each other like last time?

MartyG-RealSense, 2022-08-09:
If **ros-kinetic-realsense2-camera** was built from packages using Method 1 then using the package uninstall command should remove all packages that include 'realsense' as part of their name.

**dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge**

It would also remove the librealsense SDK though.  So you could then install librealsense 2.33.1 from source and then build the edited wrapper.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.33.1

MartyG-RealSense, 2022-08-21:
Case closed due to no further comments received.

closed 2022-08-21 by MartyG-RealSense

PhilipAmadasun, 2022-08-26:
@MartyG-RealSense I found the solution. The depthimage_to_laserscan package wasn't working because the realsense camera could not share information with the nodelets the package uses for some reason. So the problem is fixed by modifying the `3dsensor.launch` file:
```
<!--
  Turtlebot is a bit low on horsepower...

  We use openni_camera here, turn everything on by default
  (allows the user to conveniently see everything when
  launching this on its own - use with
  turtebot_rviz_launchers/view_robot.launch to visualise)
  and provide args to disable them so you can optimise the
  horsepower for your application.
 
  For an example of disabling processing modules, check
  any of the turtlebot_rapps (e.g. android_make_a_map.launch
  only enables scan_processing for depthimage_to_laserscan and
  rgb_processing for the android tele-view).
-->
<launch>
  <!-- "camera" should uniquely identify the device. All topics are pushed down
   	into the "camera" namespace, and it is prepended to tf frame ids. -->
  <arg name="camera"  	default="camera"/>
  <arg name="publish_tf"  default="false"/>
  <arg name="3d_sensor"   default="$(env TURTLEBOT_3D_SENSOR)"/>  <!-- kinect, asus_xtion_pro -->

  <!-- Factory-calibrated depth registration -->
  <arg name="depth_registration"          	default="true"/>
  <arg 	if="$(arg depth_registration)" name="depth" value="depth_registered" />
  <arg unless="$(arg depth_registration)" name="depth" value="depth" />

  <!-- Processing Modules -->
  <arg name="rgb_processing"              	default="true"/>
  <arg name="ir_processing"               	default="true"/>
  <arg name="depth_processing"            	default="true"/>
  <arg name="depth_registered_processing" 	default="true"/>
  <arg name="disparity_processing"        	default="true"/>
  <arg name="disparity_registered_processing" default="true"/>
  <arg name="scan_processing"             	default="true"/>

  <!-- Worker threads for the nodelet manager -->
  <arg name="num_worker_threads" default="4" />

  <!-- Laserscan topic -->
  <arg name="scan_topic" default="scan"/>

  <include file="$(find turtlebot_bringup)/launch/includes/3dsensor/$(arg 3d_sensor).launch.xml">
	<arg name="camera"                      	value="$(arg camera)"/>
	<arg name="publish_tf"                  	value="$(arg publish_tf)"/>
	<arg name="depth_registration"          	value="$(arg depth_registration)"/>
	<arg name="num_worker_threads"          	value="$(arg num_worker_threads)" />

	<!-- Processing Modules -->
	<arg name="rgb_processing"              	value="$(arg rgb_processing)"/>
	<arg name="ir_processing"               	value="$(arg ir_processing)"/>
	<arg name="depth_processing"            	value="$(arg depth_processing)"/>
	<arg name="depth_registered_processing" 	value="$(arg depth_registered_processing)"/>
	<arg name="disparity_processing"        	value="$(arg disparity_processing)"/>
	<arg name="disparity_registered_processing" value="$(arg disparity_registered_processing)"/>
  </include>

   <!--                    	Laserscan
 	This uses lazy subscribing, so will not activate until scan is requested.
   -->
  <!--
  <group if="$(arg scan_processing)">
	<remap from="image" to="/camera/depth/image_rect_raw"/>
	<remap from="scan" to="$(arg scan_topic)"/>
	<remap from="camera_info" to="/camera/depth/camera_info"/>
	<remap from="$(arg camera)/image" to="/camera/depth/image_rect_raw"/>
	<remap from="$(arg camera)/scan" to="$(arg scan_topic)"/>
	<remap from="$(arg camera)/camera_info" to="/camera/depth/camera_info"/>
	<node pkg="nodelet" type="nodelet" name="depthimage_to_laserscan" args="load depthimage_to_laserscan/DepthImageToLaserScanNodelet $(arg camera)/$(arg camera)_nodelet_manager">
  	<param name="scan_height" value="10"/>
  	<param name="output_frame_id" value="/$(arg camera)_depth_frame"/>
  	<param name="range_min" value="0.45"/>
	</node>
  </group>
  -->
  <remap from="image" to="/camera/depth/image_rect_raw"/>
  <remap from="camera_info" to="/camera/depth/camera_info"/>
  <node pkg="depthimage_to_laserscan" type="depthimage_to_laserscan" name="depthimage_to_laserscan" args="">
  <param name="scan_height" 	type="int"	value="1"/> <!-- default: 1 pixel. Number of pixel rows used to generate laser scan. -->
<param name="scan_time"   	type="double" value="0.033"/> <!-- default:0.033, 30 FPS . Time between scans. -->
<param name="range_min"   	type="double" value="0.45"/> <!--default:0.45m. Ranges less than this are considered -Inf. -->
<param name="range_max"   	type="double" value="10.0"/> <!--default: 10m. Ranges less than this are considered +Inf. -->
<param name="output_frame_id" type="str"	value="camera_depth_frame"/> <!--default: camera_depth_frame. Frame id of the laser scan. -->
  </node>
</launch>
```
Maybe this can help someone else, thanks!

MartyG-RealSense, 2022-08-26:
It's awesome to hear that you found a solution!  Thanks so much for the update and for sharing your solution with the RealSense ROS community  :)

adamzr2000, 2022-11-22:
Hello, I am trying to use the turtlebot2 with the d435i camera.

I am using this file as 3D sensor turtlebot_bringup/launch/includes/3dsensor/**d435i.launch.xml** but the maps created are not very accurate:

<launch>

  <arg name="camera"      default="camera"/>
  <arg name="publish_tf"  default="false"/>
  
  <arg name="depth_registration"              default="false"/>
  <arg     if="$(arg depth_registration)" name="depth" value="depth_registered" />
  <arg unless="$(arg depth_registration)" name="depth" value="depth" /> 

                                      
  <arg name="rgb_processing"                  default="true"/>                                      
  <arg name="ir_processing"                   default="true"/>                                      
  <arg name="depth_processing"                default="true"/>                                      
  <arg name="depth_registered_processing"     default="true"/>                                      
  <arg name="disparity_processing"            default="true"/>                                      
  <arg name="disparity_registered_processing" default="true"/>                                      
  <arg name="scan_processing"                 default="true"/>                                      
                                                                                                    
                                                     
  <arg name="num_worker_threads" default="4" />

  <include file="$(find realsense2_camera)/launch/rs_rgbd.launch">
    <arg name="camera"                          value="$(arg camera)"/>                                                             
    <arg name="publish_tf"                      value="$(arg publish_tf)"/>                                    
                                                                                                  
                                                                         
    <arg name="rgb_processing"                  value="$(arg rgb_processing)"/>                     
    <arg name="ir_processing"                   value="$(arg ir_processing)"/>                      
    <arg name="depth_processing"                value="$(arg depth_processing)"/>                   
    <arg name="depth_registered_processing"     value="$(arg depth_registered_processing)"/>        
    <arg name="disparity_processing"            value="$(arg disparity_processing)"/>               
    <arg name="disparity_registered_processing" value="$(arg disparity_registered_processing)"/>    
  </include>                                                                                        
</launch>

Could you show me the modifications you have made to gmapping with the turtlebot and the d435i?

Thanks in advance.