realsenseai/realsense-ros #2426: `/camera/aligned_depth_to_color/image_raw` not published

Olimoyo, 2022-07-27:
Hi,

I am running the following example:
`ros2 launch realsense2_camera rs_launch.py align_depth:=true`

The rgb image `/camera/color/image_raw` and the depth image `/camera/depth/image_rect_raw` are both published but not the aligned depth image `/camera/aligned_depth_to_color/image_raw`. 

I am on Ubuntu 20.04 with the kernel `5.14.0-1024-oem`. 

Does anyone know why this is happening?

Thank you,
Oliver

MartyG-RealSense, 2022-07-27:
Hi @Olimoyo  Which computer or computing device are you using, please (Raspberry Pi, Nvidia Jetson, desktop / laptop PC, etc)

Olimoyo, 2022-07-27:
Hi @MartyG-RealSense, I am using a Thinkstation desktop. The realsense works with another almost identical desktop except that it has a different slightly bigger GPU and kernel `5.15.0-41-generic`.

MartyG-RealSense, 2022-07-27:
Use of kernel versions newer than 5.13 has been very rare up until now, so there is almost no data about their performance with the RealSense SDK at the time of writing this.   An unofficial test patch that adds support for kernels 5.13 and 5.15 was recently released though at https://github.com/IntelRealSense/librealsense/issues/10439#issuecomment-1171965106

It is notable that the patch skips kernel 5.14.

Olimoyo, 2022-07-27:
That's very helpful! I am going to try and test with kernel 5.15 and report back.

Olimoyo, 2022-07-27:
I changed the kernel and that alone didn't fix it, but building from source did.

closed 2022-07-27 by Olimoyo

MartyG-RealSense, 2022-07-27:
That's excellent to hear that you were successful, @Olimoyo - thanks very much for the update!

omgaikwad08, 2023-11-18:
> That's very helpful! I am going to try and test with kernel 5.15 and report back.

> I changed the kernel and that alone didn't fix it, but building from source did.

Hi, I am facing exactly the same issue, can you please elaborate the procedure you followed to fix it.

I would really appreciate it.

Thanks,
Om