realsenseai/realsense-ros #2428: realsense2_camera node stops working occasionally

robotdream, 2022-07-28:
I met this weird issue when I was testing a vslam algorithm with realsense D455 using the realsense2-ros package. 

The following is the output from bringing up rs_camera.launch. It almost always gives warning or error messages as shown below. The weird part is that if the robot is kept still, the node works fine (～12h in a row) without any problems in spite of the error messages. However, when the ground robot moves in a general way that I assume will introduce some vibrations and more computational cost from localization and navigation nodes, the camera node fails to stream image data occasionally. 

This might be caused by usb cable connection, usb cable length, or any issues with the realsense node? Any suggestions would be appreciated. Thanks.

**Environment setting:** 
ubuntu 16.04 LTS，ros1 kinetic

**executed command and output:**
roslaunch realsense2_camera rs_camera.launch 
... logging to /home/jw/.ros/log/2f08cb20-0d4f-11ed-a20e-00e75c682c66/roslaunch-jw-Default-string-16079.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.100.11:42993/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
[... 142 lines cut ...]
 27/07 15:57:27,825 WARNING [140700167431936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 27/07 15:57:27,876 WARNING [140700167431936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 27/07 18:35:35,940 WARNING [140700167431936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 27/07 19:03:00,882 WARNING [140700167431936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 27/07 19:03:00,932 WARNING [140700167431936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 27/07 19:37:47,952 WARNING [140700167431936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 28/07 02:39:16,835 WARNING [140700167431936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 28/07 08:27:05,712 WARNING [140700167431936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 28/07 08:27:11,772 WARNING [140700167431936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 28/07 08:27:11,822 WARNING [140700167431936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 28/07 08:27:12,872 WARNING [140700167431936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 28/07 08:27:12,922 WARNING [140700167431936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 28/07 08:27:28,980 WARNING [140700167431936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 28/07 08:27:29,031 WARNING [140700167431936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 28/07 08:27:32,516 ERROR [140699930846976] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130
 28/07 08:27:33,087 WARNING [140700167431936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 28/07 08:27:48,143 WARNING [140700167431936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 28/07 08:27:48,194 WARNING [140700167431936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 28/07 08:27:52,834 ERROR [140699930846976] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130
 28/07 08:27:53,245 WARNING [140700167431936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 28/07 08:27:54,412 ERROR [140699974498048] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132
 28/07 08:28:00,740 ERROR [140699974498048] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132
 28/07 08:28:04,163 ERROR [140699930846976] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130
 28/07 08:28:05,303 WARNING [140700167431936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 28/07 08:28:26,726 ERROR [140699974498048] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132

MartyG-RealSense, 2022-07-28:
Hi @robotdream  If the camera is functioning normally when the robot is still then I would not expect cable length to be a factor in the problem, as the data will be travelling along the same cable length and at the same resolution / FPS speed no matter whether the robot is stationary or in motion.

The stereo depth algorithm of RealSense 400 Series cameras is robust against vibration, so it would take severe vibration such as travelling over rough agricultural land or being attached to a robot / drone that was transmitting vibration through its body framework to have a negative effect on the camera.

If the cable is moving around and bending during robot motion then the connector may move in the USB port and cause a temporary disconnection.  Using a USB cable with a screw-lock connector to firmly fix the micro-sized end of the cable into the side of the camera may help to avoid this.

[image]

robotdream, 2022-07-29:
Thanks for the timely replay, @MartyG-RealSense. I did use a cable with a screw-lock connector similar to the one in the left figure when I was doing the tests. Considering your response, now I think the problem might be caused by the USB side connected to the PC. I am also wondering if it is necessary to use an official cable from Intel Realsense or a similar cable from a third-party manufacturer should also work.  Thanks!

MartyG-RealSense, 2022-07-29:
Third-party cables can be used with RealSense cameras.  If the cable is 2 meters or longer then it is important that the cable is high quality due to the high volume of data typically transmitted along the cable by the camera, otherwise degraded performance or disconnections may occur.

**Newnex** are a company that supply RealSense-validated industrial grade USB equipment.

https://www.newnex.com/realsense-3d-camera-connectivity.php

https://www.youtube.com/watch?v=GLQgR1jT04M

Another brand of cable that has been recommend by RealSense users in the past is **OCP**.

https://www.ocp.com/product/usb-3-0-3-1-cable-type-c-to-a-gen1-5g-black-intel-d435-realsense-camera-62-00214/

robotdream, 2022-07-29:
Thanks! The cable I used was about 4m or even longer. Will try another cable and see if it will work.

MartyG-RealSense, 2022-08-04:
Hi @robotdream  Have you been able to test with another cable yet, please?  Thanks!

MartyG-RealSense, 2022-08-09:
Hi @robotdream  Do you require further assistance with this case, please?  Thanks!

robotdream, 2022-08-11:
Hi @MartyG-RealSense, I changed to a shorter cable but the issue still happened occasionally. Have changed to the cable that came with the realsense cam and will do more tests and see if it was a cable issue. Thanks!

MartyG-RealSense, 2022-08-11:
Thanks very much for the update.  I look forward to your next test results.  Good luck!

closed 2022-09-04 by MartyG-RealSense