realsenseai/realsense-ros #2429: I want to know how to decode ir dot projector image of SR300.

ohsm1125, 2022-07-28:
I'm not sure if it's the right question to ask here.

As far as I know, the SR300 has an IR Dot Projector and one IR camera.

And I know that it generates depth map by decoding pattern using "soc(maybe primesense?)".

What I'm curious about here is, using only the dot projected image How to create a depth map.

Is decoding impossible with software alone??

If software decoding is available(dont care processing speed), 
can I get a sample code? (without soc's resource/only using cpu resource)

please leave comment for me... thanks

MartyG-RealSense, 2022-07-28:
Hi @ohsm1125  The RealSense SR300 coded light camera (similar to structured light) is a different camera technology to the 400 Series stereo depth cameras.  It does have only one infrared sensor, though it does not project IR dots.  Instead, it projects vertical bar patterns.

[image]

Page 9 of the official data sheet document for the SR300 provides the following explanation for how it calculates depth.

https://dev.intelrealsense.com/docs/intel-realsense-sr300-datasheet

*************

The Intel RealSense SR300 Series Depth Camera uses multiple components to generate a depth image.  To generate a depth frame, the IR projector illuminates the scene with a set of predefined, increasing spatial frequency coded IR vertical bar patterns.  These patterns are warped by the scene, reflected back and captured by the IR camera.  The IR camera pixel values are then processed by the vision processor to generate a depth frame.  Subsequent depth frames create a video stream that is transmitted to the host system.

*********

Structured light camera technology was used by the past PrimeSense and Kinect V1 cameras.

You should be able to obtain a depth map from an SR300 if you create a script in the RealSense SDK 2.0 (librealsense) software and set the resolution and frames per second of the depth stream to 640x480 and 30 frames per second (FPS).  This script will run on the CPU of the computer.

Alternatively, you could use the SR300 without the need for programming with the RealSense SDK's powerful pre-made **RealSense Viewer** tool, again setting resolution / FPS to 640x480 and 30 FPS.

https://github.com/IntelRealSense/librealsense

[image]

If you are using Windows then you can use an automated installer program to install the RealSense SDK and pre-built executable examples and tools (including the RealSense Viewer) by visiting the link below and downloading and running the **Intel.RealSense.SDK-WIN10** file from the 'Assets' file list at the bottom of the page.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.50.0

[image]

If you are using Linux then you can build the SDK from source code or install it from Debian packages.

MartyG-RealSense, 2022-08-04:
Hi @ohsm1125  Do you require further assistance with this case, please?  Thanks!

ohsm1125, 2022-08-09:
thank you for your help.

actually, i am aiming to create a depth map using a dot pattern (signle ir dot projector + single ir camera) no bar pattern.

so, i wanted to do a test to find a depth map with an image projected with a dot pattern (without asic hardware chip's help)

I'm looking for an algorithm to find a depth map with only a dot pattern, 
but i can't find it.

is there any open source code that i can refer to...?

Even if I don't get a proper answer, I really appreciate your reply.

MartyG-RealSense, 2022-08-09:
The RealSense SDK's C++ **software device** interface enables a 'synthetic camera' to be defined that can output depth without the need for a real camera to be attached as long as you have an image source to feed into software device.  The example program **rs-software-device** demonstrates this.

https://github.com/IntelRealSense/librealsense/tree/master/examples/software-device

MartyG-RealSense, 2022-08-15:
Hi @ohsm1125  Was the advice in the comment above helpful to you, please?  Thanks!

closed 2022-08-16 by ohsm1125