realsenseai/realsense-ros #2430: Realsense L515 Odometry topic

Cristian-wp, 2022-07-28:
Hi, I am running ROS MELODIC on UBUNTU 18, I have installed all the package and the camera works.
Sorry for the basic question, but I can not understand which is the ODOMETRY topic published from the package, and how can abilitate it.

As I understand this camera like T265 can provide an odometry topic, it is correct?

MartyG-RealSense, 2022-07-28:
Hi @Cristian-wp  Intel's RealSense ROS wrapper documentation page at the link below lists support for the odometry topic **/camera/odom/sample** for T265 only.

https://dev.intelrealsense.com/docs/ros-wrapper#t265-tracking-camera

This topic should be published by default when launching a T265 in the ROS wrapper.

MartyG-RealSense, 2022-08-03:
Hi @Cristian-wp  Do you require further assistance with this case, please?  Thanks!

Cristian-wp, 2022-08-03:
Yes, but I do not know if I have to open a new topic. I am trying to use the camera fused with its imu in order to get a visual odometry topic, but sometimes the realsense lost the frame... I am not using the pointcloud, I am using the image_raw topic and during my experiments my robot fails the navigation from this fact.
I have try to change the original cable with another one, change usb3 port on my pc (INTEL NUC i7 with ros melodic), change frame frequency but nothing change...actually the publish frequency is 30Hz.
@MartyG-RealSense  can you please suggest me a solution to this problem?
The strange thing is that I have even a rs435i, and I have the sam problem. Same PC, but different usb3 port and different cables.
(even the cables are all usb3)

MartyG-RealSense, 2022-08-03:
Intel's D435i SLAM guide for ROS provides the following warning about the IMU.

************
The built-in IMU can only keep track for a very short time.  Moving or turning too quickly will break the sequence of successful point cloud matches and will result in the system losing track.  It could happen that the system will recover immediately if stopped moving but if not, the longer the time passed since the break, the farther away it will drift from the correct position.  The odds for recovery get very slim, very quickly.  The parameters set in the launch file are most likely not ideal but this is a good starting point for calibrating.

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i

MartyG-RealSense, 2022-08-09:
Hi @Cristian-wp  Was the information in the comment above helpful to you, please?  Thanks!

Cristian-wp, 2022-08-10:
Hello @MartyG-RealSense , I am sorry but I was not in office.

**"Does stability improve if the robot moves or turns slower?"**
I do not use the internal IMU, I am using the camera like a normal global shutter camera.
Even if the cameras stay still, the situation does not change, I always lost frames after random time that can be 1min or 5s after the package start running.

MartyG-RealSense, 2022-08-10:
I conducted further research of your question but could not find any references about obtaining odometry data in ROS without using an IMU.

It sounds as though your camera is working in ROS but you are losing frames.  What roslaunch instruction are you using, please?

closed 2022-08-25 by MartyG-RealSense