realsenseai/realsense-ros #2431: Post processing depth to pointcloud using GPU(CUDA)

Camilochiang, 2022-07-28:
Is there any implementation from intel or others for posprocessing  rosbags recorded with ROS (not intelRealsense SDK) that transform depth frames to PCL using GPU (hopefully CUDA)?

MartyG-RealSense, 2022-07-28:
Hi @Camilochiang  Some suggestions for how to process RealSense ROS wrapper topics with PCL are provided in the link below.

https://answers.ros.org/question/316701/ros-intel-realsense-how-to-process-data-with-pcl/

If CUDA support is enabled in the librealsense SDK then it will automatically accelerate color conversion, pointcloud and depth-color alignment operations but would not assist PCL operations.  

CUDA support can be enabled by building librealsense from Debian packages or building it from source code with the CMake flag **-DBUILD_WITH_CUDA=true** included in the CMake build instruction.  CUDA is not included though when building from packages with the ROS wrapper's **Method 1** procedure.

Camilochiang, 2022-07-28:
@MartyG-RealSense are you a computer?

MartyG-RealSense, 2022-07-28:
No I am a real engineer called Marty.  :)

Camilochiang, 2022-07-28:
Interesting. Where are you based?

i will re ask my question: I’m looking for a way of process depth images outside of the ros wrapper using Cuda. Any suggestions ?

MartyG-RealSense, 2022-07-28:
I am not able to provide my location.

The link below may be helpful to you if you wish to translate a pointcloud from a rosbag file to PCL in ROS and not in the RealSense ROS wrapper.

https://stackoverflow.com/questions/39772424/how-to-effeciently-convert-ros-pointcloud2-to-pcl-point-cloud-and-visualize-it-i/67296769#67296769

I do not have advice that I can offer about applying CUDA outside of the CUDA implementation in the RealSense SDK, unfortunately.

MartyG-RealSense, 2022-08-09:
Case closed due to no further comments received.

closed 2022-08-09 by MartyG-RealSense

tugbakara, 2024-02-15:
> Hi @Camilochiang Some suggestions for how to process RealSense ROS wrapper topics with PCL are provided in the link below.
> 
> https://answers.ros.org/question/316701/ros-intel-realsense-how-to-process-data-with-pcl/
> 
> If CUDA support is enabled in the librealsense SDK then it will automatically accelerate color conversion, pointcloud and depth-color alignment operations in the ROS wrapper too but would not assist PCL operations.
> 
> CUDA support can be enabled by building librealsense from Debian packages or building it from source code with the CMake flag **-DBUILD_WITH_CUDA=true** included in the CMake build instruction. CUDA is not included though when building from packages with the ROS wrapper's **Method 1** procedure.

did you mean with method 1 installing from apt for ros wrapper ( asking for ROS2, I am not sure that installing from apt ros2 wrappers includes CUDA support or not)

MartyG-RealSense, 2024-02-15:
@tugbakara  CUDA support is enabled when installing the librealsense SDK from Jetson apt packages or from source code with the **-DBUILD_WITH_CUDA=ON** flag included in the CMake build instruction.

CUDA support will not be included when using Method 1 of the ROS wrapper.