realsenseai/realsense-ros #2432: Possibly deceptive extrinsics topic name?

john-maidbot, 2022-07-30:
Please correct me if I am wrong, but it appears that the .../extrinsics/depth_to_color is actually the extrinsics going from the color camera to the depth camera. If that is the case, I think the topic should be .../extrinsics/color_to_depth to clarify this point.

Here is where you lookup the extrinsic information:
https://github.com/IntelRealSense/realsense-ros/blob/442d6d9a19149cb9cb40ffb351593c3fa8b9eb9f/realsense2_camera/src/base_realsense_node.cpp#L2260

```
ex = getAProfile(stream).get_extrinsics_to(base_profile);
```

if,
base_profile == depth profile
getAProfile(stream) == color profile
then ex will be the extrinsic from the color to the depth camera, right?

If I am wrong, I am so sorry for wasting your time. But I think this is the case, since when I get the extrinsic from realsense-ros, I get:

```
ros2 topic echo /d435/extrinsics/depth_to_color
rotation:
- 0.9999857544898987
- 0.0021478114649653435
- -0.0048892986960709095
- -0.0021304551046341658
- 0.9999914169311523
- 0.003552317386493087
- 0.004896886646747589
- -0.0035418502520769835
- 0.9999817609786987
translation:
- -0.014618603512644768
- -4.5008506276644766e-05
- -0.0005783233791589737
---

```

Which I have confirmed is in fact the extrinsic from color to depth, not "depth_to_color":
```
rs-enumerate-devices -c
Extrinsic from "Color"	  To	  "Depth" :
 Rotation Matrix:
   0.999986        -0.00213046       0.00489689    
   0.00214781       0.999991        -0.00354185    
  -0.0048893        0.00355232       0.999982      

 Translation Vector: -0.0146186035126448  -4.50085062766448e-05  -0.000578323379158974  
```

MartyG-RealSense, 2022-07-30:
Hi @john-maidbot  I will enquire to the RealSense ROS developer team about your question.  Thanks for your patience!

SamerKhshiboun, 2022-08-08:
Hi @john-maidbot,
You are right, there is a bug when calculating the extrinsic, while the naming is correct.
The line you pointed on should be fixed (by swapping the base_profile and stream) to:
`ex = getAProfile(base_profile).get_extrinsics_to(stream);
`
We will fix this and update you.

MartyG-RealSense, 2022-08-14:
Thanks very much, @SamerKhshiboun  :)

I have added an **Enhancement** label to this case as a reminder to keep it open.

john-maidbot, 2022-10-11:
Looks like the fix was merged, so i will close the issue. Thanks @SamerKhshiboun !!! :heart:

closed 2022-10-11 by john-maidbot

schmidtp1, 2022-10-14:
@john-maidbot, please note that topic you are referring to is published in these lines i think: https://github.com/IntelRealSense/realsense-ros/blob/ddd6fe116859b960a59412538e91d085a368c871/realsense2_camera/src/base_realsense_node.cpp#L2097 (and not here with /tf_static: https://github.com/IntelRealSense/realsense-ros/blob/442d6d9a19149cb9cb40ffb351593c3fa8b9eb9f/realsense2_camera/src/base_realsense_node.cpp#L2260).

the code looks right to me but i'm still trying to understand the (unexpected) behavior that you are seeing.

did you try it after the merge? (i assume it doesn't change anything?)

SamerKhshiboun, 2023-04-03:
see comments inside the PR itself #2477