realsenseai/realsense-ros #2433: How to stream only image_raw ?

Janebek, 2022-08-01:
Hi ! 
I noticed that when I use "roslaunch realsense2_camera rs_camera.launch" it will stream not only image_raw , but also depth , rgbd camera , stereo_module. 
How do I turn those other things off ? I just want to use image_raw only ? How do I do it ? 
Really looking forward to you reply !!!

MartyG-RealSense, 2022-08-01:
Hi @Janebek  To disable publishing of depth and color in your launch, you can add the terms below to your launch instruction.

enable_depth:=false enable_color:=false

For example:

**ROS1**
roslaunch realsense2_camera rs_camera.launch enable_depth:=false enable_color:=false

**ROS2**
ros2 launch realsense2_camera rs_launch.py enable_depth:=false enable_color:=false

The infrared topics, and the IMU topics if your camera has an IMU, are disabled by default and so do not need to be disabled in the launch instruction.

Janebek, 2022-08-01:
I just tried it out .

The **/camera/rgb_camera & /camera/stereo_module** still exist , is this a actually stream ? I assume it is not .

MartyG-RealSense, 2022-08-01:
Depth and infrared are the Stereo Module, and infrared is already disabled by default so you would only need to disable depth.

I'm not aware of a command for disabling intrinsics and extrinsics.

stereo_module and rgb_camera are not things that you would disable like a stream.  They are a way of categorizing the type of instruction (e.g depth auto_exposure being under the stereo_module category and the RGB instruction auto_exposure_priority being under the rgb_camera category

[image]

Motion module topics (Accel and Gyro) are disabled by default.

So if you want to disable depth and keep color, you should only need:

**roslaunch realsense2_camera rs_camera.launch enable_depth:=false**

Janebek, 2022-08-01:
Thank you ! I got it !

Janebek, 2022-08-01:
Hi ! I am running **"roslaunch realsense2_camera rs_camera.launch enable_depth:=false"** on RaspberryPi .

But when run this command , the **CPU% comes to 37% ("roslaunch realsense2_camera rs_camera.launch" is 41%")** 

I find it very not worthy , becuase I want run other SLAM algorithm , the command is so heavy to run , do you have any methods to reduce the CPU% ?

I find when I run "roslaunch realsense2_camera rs_camera.launch" , **the first ros_info is Initializing nodelet with 4 worker threads ,**  maybe reducing the threads is a way to reduce CPU% ? I don't know .

Do you have any suggestion on this problem?

Thank you !

MartyG-RealSense, 2022-08-01:
The default launch configuration that is applied on D435i is likely using a resolution of 1280x720 for color at 30 FPS.  You could therefore test whether using a custom stream configuration with lower settings reduces CPU usage.  For example, setting color FPS to 15 and using resolution 640x480.

**roslaunch realsense2_camera rs_camera.launch color_width:=640 color_height:=480 color_fps:=15**

Did you install the librealsense SDK from source code with CMake?  If you did then including the build term **-dcmake_build_type=Release** in the CMake build instruction may help by building in Release mode to take advantage of optimisations if you are not using this build term already.

Janebek, 2022-08-01:
I test it after reducing the resolution and FPS , the CPU% is 36% , it means this does not help a lot .

I installed the librealsense SDK , but what do you meam by **"may help by building in Release mode to take advantage of optimizations"** ? I don't understand .

Can you please make it more specific ?

Thank you !

MartyG-RealSense, 2022-08-01:
If Release mode is not specified when building the SDK from source code then the SDK is built in **Debug** mode instead, which may not run as fast as a Release build.which supports speed optimizations.

Janebek, 2022-08-01:
where can I check this ? maybe in the build file ?

MartyG-RealSense, 2022-08-01:
The only way I know of to check whether the SDK is a Debug build once it has been built is to see whether the SDK's DLL file names have had a 'd added to the end of them, such as **realsense2d.dll** instead of realsense2.dll

MartyG-RealSense, 2022-08-01:
They should be in the "bin" folder.  

On Windows they should be at the location described at https://github.com/IntelRealSense/librealsense/issues/8035#issuecomment-748936135

For Linux please check in **usr/local/bin**

Janebek, 2022-08-01:
Thank you !

I find nothing with d in **usr/local/bin** , but I still re-make the whole SDK file with **-dcmake_build_type=Release**.

Howerver, the CPU% is still 36%. Do you any other suggestions to slove this problem?

I think if it can go under 20% , it would be really great.

MartyG-RealSense, 2022-08-02:
If you are using a Raspberry Pi model with a specification that is less than a Pi 4 then there may be a limit to how much further you can reduce the CPU utilisation percentage.

If a Pi model has a low amount of memory capacity and is using Raspian / Raspberry Pi OS then once an application has used up the real memory then it will start using the slower to access **swapfile**, which is storage space on a drive such as an SD card that is used as virtual memory.

Janebek, 2022-08-03:
Hi! Thanks for your reply.

I'm using a 4B/8G , with 256G SD card . 

I don't think it has a memory/storage problem . Do you think there maybe any chance reducing the CPU% even more ?

Maybe change some code to delete those redundant algorithm , to just stream color image only ?

MartyG-RealSense, 2022-08-03:
You could use **rostopic list** to get a full list of the topics that are currently being published after depth has been disabled and then **unsubscribe** from the topics that you do not want to save CPU processing.  

The link below provides an example of topics published automatically by the RealSense ROS wrapper, some of which you may not need (such as **/diagnostics** which provides camera information such as its current operating temperatures).

https://github.com/IntelRealSense/realsense-ros#published-topics

Janebek, 2022-08-03:
what is the unsubscribe command for example? Like for **/diagnostics**.

MartyG-RealSense, 2022-08-03:
There is a discussion about unsubscribing from topics at https://github.com/IntelRealSense/realsense-ros/issues/1909

Janebek, 2022-08-10:
Not for now, thanks for asking!

closed 2022-08-10 by Janebek