realsenseai/realsense-ros #2434: remote mapping

mwelponer, 2022-08-01:
I have followerd this tutorial https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i succesfully, using a D435i camera attached to a PC running ROS noetic. Everything seems to work properly, enabling MapCloud in RViz I can see the 3D mapCloud beeing generating. 
I am now trying to make it work with the camera on my rover robot running a raspberry Pi4. So I have configured .bashrc on both systems (ROS_MASTER_URI, ROS_HOSTNAME and ROS_IP) so that the master is on the robot.
If I start the camera on my robot using ``roslauch realsense2_camera rs_camera.launch`` and I run ``rostopic list`` on my PC I can see the new published camera topics, so the setup seems to be ok. 
Now if I launch ``roslauch realsense2_camera rs_camera.launch`` on my rover and ``roslauch realsense2_camera opensource_tracking.launch`` with argument ``offline=true`` on my PC everything should continue to work properly, but it doesn't. 
In rqt I don't see the map and the odom frames in the frames viewer. In RViz I have a Global status error: "Unknown frame map"
What am I doing wrong and how can I make it work?

MartyG-RealSense, 2022-08-02:
Hi @mwelponer  The D435i guide describes how live camera data can be saved to a rosbag file, and then that bag can be played back using the opensource_tracking launch file set to offline = true so that the bag data is used as the data source instead of a live RealSense camera.

May I ask if a rosbag file has already been recorded before the 'offline = true' launch is performed, please?  If the launch is not provided with a bag file in offline mode then RViz may have no data to play back if it is not accessing a live camera.

mwelponer, 2022-08-03:
Hello MartyG, thank you for your message.
Actually I am not using a rosbag, never recorded anything,.I am just trying to understand what I should change in my launch files when I move the camera from my PC (where I launch opensource_tracking) to my external robot. Isn't it correct to set offline=true if the camera is attached to the external robot?
Thank you for your help!

MartyG-RealSense, 2022-08-03:
The D435i SLAM guide was designed for use with a single computer and there are no references that I know of regarding how to convert it to a two-computer setup using ROS_MASTER_URI.

I believe that you are on the right track with using ROS_MASTER_URI though, like in the examples of two computer communication described at https://github.com/IntelRealSense/librealsense/issues/10730#issuecomment-1197918767

In Intel's guide to using the wrapper across 2 computers at the link below, they open multiple ROS terminals, with **roscore** on its own in the first terminal and the second computer with a camera attached **ssh'd** to from the first computer and with both computers using the same launch file.

https://github.com/IntelRealSense/realsense-ros/wiki/showcase-of-using-3-cameras-in-2-machines

Apologies that I could not be more helpful in advice about this particular project.

alvaropascuaI, 2022-08-04:
How can I launch the **opensource_tracking.launch** file but with ros2 ?

MartyG-RealSense, 2022-08-04:
Hi @alvaropascuaI  It would be impractical to try to convert the opensource_tracking ROS1 launch file to ROS2, especially as it is interconnected with the rs_camera.launch ROS1 file.

You could edit the rs_camera.py ROS2 launch file to replicate some of the settings of opensource_tracking, such as making enable_gyro and enable_accel = true and setting unite_imu_method to linear_interpolation

alvaropascuaI, 2022-08-04:
We use the opensource file in this tutorial  https://www.youtube.com/watch?v=mguXtybmZv4, but of course according to what you are telling me, I don't think it will open an rviz viewer, with the same command as him, right?

MartyG-RealSense, 2022-08-04:
You can open **RViz2** in ROS2 after launch outside of the launch file with the command **ros2 run rviz2 rviz2**

If the rviz=true command in the launch file does not open RViz2 then you could try changing it to **rviz2**

alvaropascuaI, 2022-08-04:
how i can change that setting unite_imu_method to linear_interpolation ?

MartyG-RealSense, 2022-08-04:
You could set it in the launch instruction.

**ros2 launch realsense2_camera rs_launch.py unite_imu_method:="linear_interpolation"**

mwelponer, 2022-08-04:
Thank you for your message Marty, 
maybe I am missing something but as far as I understood ROS was engineered as a peer-to-peer network in which ROS nodes may process data and interact through the use of network messages.
So it should be the most natural thing to imagine a scenario in which a node is created by computer A to publish e.g. camera topics and another nodes (e.g. rtabmap, imu_filter, localization) are created by computer B to process data coming from the subscribed topics. (see for example this tutorial that uses Kinect http://wiki.ros.org/rtabmap_ros/Tutorials/RemoteMapping)
I would do it alltogether on my rover, but the computation launched by opensource_tracking takes more than 100% of my raspberry cpu. That's why I am trying to understand how to move the mapping/localization processing on my PC and leave just the publication of the camera topics to the rover!

MartyG-RealSense, 2022-08-04:
ROS networking is an area of specialist ROS knowledge that I am less familiar with, so I appreciate your patience.  

Your goal to generate topics on the rover that has a camera attached to it and transmit the camera data to another computer to perform the heavy processing on that machine sounds similar to RealSense ROS users who have converted Intel's RealSense open-source **re-server** networking system to use ROS network protocols instead of RTSP protocol.  https://github.com/IntelRealSense/realsense-ros/issues/1808 is an example of such a conversion to ROS protocols.

MartyG-RealSense, 2022-08-10:
Hi @mwelponer  Do you require further assistance with this case, please?  Thanks!

mwelponer, 2022-08-10:
Hi @MartyG-RealSense thank you for your assistance, yes I think you can close the case, thanks!

MartyG-RealSense, 2022-08-10:
You are very welcome, @mwelponer - thanks very much for the update!  As you suggested, I will close the case.  Thanks again!

closed 2022-08-10 by MartyG-RealSense