realsenseai/realsense-ros #2435: color_fps problem.

Janebek, 2022-08-02:
Hi ! @MartyG-RealSense 

I'm trying to chang the color fps of my D435i using the command **" color_fps:= num "** , but I can not change it , it's default is 15 , and I can change it to 30 , but other numbers seem do not fit into this system . (I test them with rostopic hz)

I want to change it to 10fps , how can I do this ? 

Really looking forward to your reply! Much thanks!

MartyG-RealSense, 2022-08-02:
Hi @Janebek  The data sheet document for the 400 Series cameras states that the FPS speeds that are supported for RGB on D435i are 6, 15 and 30, and also 60 if using 960x540 resolution or less (such as 848x480).

https://dev.intelrealsense.com/docs/intel-realsense-d400-series-product-family-datasheet

If you require.a specific speed in ROS that is not officially supported then you could investigate the possibility of achieving it with a **rate**, as described in the link below.

https://roboticsbackend.com/ros-rate-roscpy-roscpp/

Janebek, 2022-08-03:
Thank you ! I got it.

closed 2022-08-03 by Janebek