realsenseai/realsense-ros #2439: pixelated 2d map with D435

JoaoVitorBranco, 2022-08-05:
I'm a beginner  in ROS and I'm trying to map a room with Realsense Depth Camera D435i, using SLAM. When I run:
```
roslaunch realsense2_camera opensource_tracking.launch
```
the Image appears like this in RViz:
[image]

The image of the camera apparently appears normal, but the map isn't good...

MartyG-RealSense, 2022-08-05:
Hi @JoaoVitorBranco   Originally the IMU topics accel and gyro were enabled by default in the RealSense ROS wrapper when the opensource_tracking.launch file was created.  From wrapper 2.2.15 onwards these IMU topics were disabled by default.  They can be enabled in the roslaunch instruction.

**roslaunch realsense2_camera opensource_tracking.launch enable_accel:=true enable_gyro:=true**

Does enabling the IMU topics improve your image, please?

Intel's D435i SLAM guide for ROS at the link below is an excellent reference for using the opensource_tracking launch file.

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i

This guide pre-dates the IMU topics being disabled by default, so remember to add the enable commands to the roslaunch when following its instructions.

JoaoVitorBranco, 2022-08-08:
Hi @MartyG-RealSense ! Thanks for the answer !
Before sending this comment, I try a lot of things, and after reinstalling a lot of times the realsense2 packages, one error appeard and then I reinstall my VM and all the programs again. I'm telling this because I tried to run this command and apparently the output in the terminal has changed:

JoaoVitorBranco, 2022-08-08:
Another question: it's necessary to have a robot with odom? I'm just using the realsense camera...

JoaoVitorBranco, 2022-08-08:
I miss click the button "Close issue", srry

MartyG-RealSense, 2022-08-09:
If you are following the D435i SLAM guide and installing the prerequisite modules, because of the age of that guide it provides instructions for installing ROS **Kinetic** packages.  As you are using ROS Noetic, I wonder whether installing Noetic versions of the modules may provide better results.  For example:

**sudo apt install ros-noetic-imu-filter-madgwick**

The D435i SLAM guide should only require an IMU-equipped RealSense camera such as D435i and not need a robot.

JoaoVitorBranco, 2022-08-09:
I did that, but it still makes those maps...

MartyG-RealSense, 2022-08-09:
It's excellent to hear that you made significant process after reinstallation.

In the above log, the most noticeable problem is the continuously generating **Failed to meet update rate!** warnings.  There is little documentation available on this error.  When ROS does generate this error though, it sometimes provides this advice:

**Try decreasing the rate, limiting sensor output frequency, or limiting the number of sensors.**

I believe that 'sensor output frequency' is referring to the IMU frequency.

JoaoVitorBranco, 2022-08-10:
Those were the errors:

1. [ERROR] [1660133849.523741694]: PluginlibFactory: The plugin for class 'octomap_rviz_plugin/ColorOccupancyGrid' failed to load.  Error: According to the loaded plugin descriptions the class octomap_rviz_plugin/ColorOccupancyGrid with base class type rviz::Display does not exist. Declared types are  rtabmap_ros/Info rtabmap_ros/MapCloud rtabmap_ros/MapGraph rviz/AccelStamped rviz/Axes rviz/Camera rviz/DepthCloud rviz/Effort rviz/FluidPressure rviz/Grid rviz/GridCells rviz/Illuminance rviz/Image rviz/InteractiveMarkers rviz/LaserScan rviz/Map rviz/Marker rviz/MarkerArray rviz/Odometry rviz/Path rviz/PointCloud rviz/PointCloud2 rviz/PointStamped rviz/Polygon rviz/Pose rviz/PoseArray rviz/PoseWithCovariance rviz/Range rviz/RelativeHumidity rviz/RobotModel rviz/TF rviz/Temperature rviz/TwistStamped rviz/WrenchStamped rviz_imu_plugin/Imu rviz_plugin_zed_od/ZedOdDisplay

2. [ERROR] (2022-08-10 05:19:32.269) Rtabmap.cpp:1277::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 3560 is ignored!

And a strange warning appeard to:

[ WARN] (2022-08-10 05:18:27.977) OdometryF2M.cpp:529::computeTransform() Registration failed: "Not enough inliers 14/20 (matches=181) between -1 and 432" (guess=xyz=0.012924,0.033032,-0.002298 rpy=0.008266,0.002370,0.065071)
[ WARN] (2022-08-10 05:18:27.982) OdometryF2M.cpp:290::computeTransform() Failed to find a transformation with the provided guess (xyz=0.012924,0.033032,-0.002298 rpy=0.008266,0.002370,0.065071), trying again without a guess.
[ WARN] (2022-08-10 05:18:28.363) OdometryF2M.cpp:539::computeTransform() Trial with no guess succeeded!

MartyG-RealSense, 2022-08-10:
A user of opensource_tracking at https://github.com/IntelRealSense/realsense-ros/issues/1080 was also experiencing the **Not enough inliers** and "resource temporarily unavailable" errors that you had in your log above.

Advice given to them from a RealSense ROS wrapper developer about the **Not enough inliers** warning is to provide the camera with an "interesting" scene to look at (i.e one that is not too empty of objects and surfaces) and to keep the camera looking at such a scene.

The developer also advised switching to rtabmap's pointcloud and using topic **/voxel_cloud** and frame_id **map**

JoaoVitorBranco, 2022-08-11:
Thanks again for the answer! I made a lot of progress after the last message. I got a great map!

[image]

The one problem is that I tried a lot to get it because many times the "camera_link" stopped following the real track of the camera and started flying randomly... That was the longest track that I got with the camera before the camera started to fly (I couldn't finish the middle at that moment the camera started to fly"

I will try what you said to correct these errors.

MartyG-RealSense, 2022-08-11:
It's great to hear that you made significant progress!

At the link below, a RealSense user created an adaptation of Intel's D435i SLAM guide that added localization with RTABMAP and obtaining the relative position of the camera from a recorded pointcloud map.

https://shinkansan.github.io/2019-UGRP-DPoom/SLAM

JoaoVitorBranco, 2022-08-11:
Okay, I will follow the guide and tell if it works, thanks for the help.

JoaoVitorBranco, 2022-08-11:
I stills flys... I run the launch file without and with "enable_accel:=true enable_gyro:=true" and the same....

I got that warning:
[ WARN] [1660218865.315585789]: Warning: Some linear velocity entries in parameter imu0_config are listed true, but an sensor_msgs/Imu contains no information about linear velocities
and I follow this to correct it:
https://answers.ros.org/question/209734/ekf_localization_node-type-error-for-topic-imu0/.

I changed that param from 300 to 100 and the camera still flys:
'<param name="/ukf_se/frequency" value="100"/>'

I tested with and without all the changes that I mentioned...

That's the output:

roslaunch realsense2_camera opensource_tracking_tk_online.launch enable_accel:=true enable_gyro:=true
... logging to /home/jooj/.ros/log/964fb1a0-196a-11ed-a560-b984e8afea74/roslaunch-ubuntu-7402.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:44689/

SUMMARY
========

CLEAR PARAMETERS
 * /points_xyzrgb/
 * /rtabmap/rgbd_odometry/
 * /rtabmap/rtabmap/
 * /ukf_se/

PARAMETERS
 * /ImuFilter/_publish_tf: False
 * /ImuFilter/_use_ma: False
 * /ImuFilter/_world_frame: enu
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
[... 649 lines cut ...]
[ INFO] [1660219168.746787506]: Odom: quality=441, std dev=0.222841m|0.038960rad, update time=0.203218s
[ INFO] [1660219169.020480302]: Odom: quality=419, std dev=0.246797m|0.043116rad, update time=0.267880s
[ INFO] [1660219169.199740244]: rtabmap (29): Rate=1.00s, Limit=0.000s, Conversion=0.0021s, RTAB-Map=0.6606s, Maps update=0.0288s pub=0.0000s (local map=1, WM=1)
[ INFO] [1660219169.534672709]: Odom: quality=452, std dev=0.224225m|0.036256rad, update time=0.355911s
[ INFO] [1660219169.744190496]: Odom: quality=443, std dev=0.233642m|0.036256rad, update time=0.196731s
[ INFO] [1660219169.967389774]: Odom: quality=445, std dev=0.221636m|0.036256rad, update time=0.193543s
[ INFO] [1660219170.516994179]: Odom: quality=436, std dev=0.234842m|0.032761rad, update time=0.398526s
[ INFO] [1660219170.752480660]: rtabmap (30): Rate=1.00s, Limit=0.000s, Conversion=0.0024s, RTAB-Map=0.7381s, Maps update=0.0442s pub=0.0000s (local map=1, WM=1)
[ INFO] [1660219170.808682987]: Odom: quality=426, std dev=0.220480m|0.036256rad, update time=0.288316s
[ INFO] [1660219171.068796136]: Odom: quality=435, std dev=0.232195m|0.038960rad, update time=0.228267s
[ INFO] [1660219171.307948408]: Odom: quality=403, std dev=0.261552m|0.048989rad, update time=0.213620s
[ INFO] [1660219171.670598373]: Odom: quality=430, std dev=0.240037m|0.036256rad, update time=0.319826s
[ INFO] [1660219172.001880201]: rtabmap (31): Rate=1.00s, Limit=0.000s, Conversion=0.0058s, RTAB-Map=0.6670s, Maps update=0.0269s pub=0.0000s (local map=1, WM=1)
[ INFO] [1660219172.211648919]: Odom: quality=435, std dev=0.251734m|0.038960rad, update time=0.215859s
[ INFO] [1660219172.417402883]: Odom: quality=433, std dev=0.221857m|0.038960rad, update time=0.202906s
[ INFO] [1660219172.751835506]: Odom: quality=427, std dev=0.201509m|0.041195rad, update time=0.305350s
[ INFO] [1660219173.073568125]: Odom: quality=425, std dev=0.242313m|0.041195rad, update time=0.268525s
 11/08 04:59:33,211 ERROR [139721240192768] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130
[... cut ...]
[ WARN] [1660219174.275122898]: Still waiting for data on topics /imu/data_raw and /imu/mag...
[ WARN] [1660219184.275947954]: Still waiting for data on topics /imu/data_raw and /imu/mag...
[ WARN] [1660219194.275075344]: Still waiting for data on topics /imu/data_raw and /imu/mag...
[ WARN] [1660219204.275385845]: Still waiting for data on topics /imu/data_raw and /imu/mag...
[ WARN] [1660219214.275167371]: Still waiting for data on topics /imu/data_raw and /imu/mag...
[ WARN] [1660219224.275240243]: Still waiting for data on topics /imu/data_raw and /imu/mag...

MartyG-RealSense, 2022-08-11:
Do you see any improvement if **initial_reset:=true** is added to the opensource_tracking roslaunch instruction to reset the camera at launch?

JoaoVitorBranco, 2022-08-12:
It didn't work. But I saw something. All the params that I'm changing in the command line weren't changing the params. In all the last terminal outputs, the "enable_gyro" and "enable_accel" are "False". Now I changed the values in the "includes/nodelet.launch.xml" file, and now they appear as "True". I made this change:

'<param name="enable_gyro"              type="bool" value="true"/> '
   ' <param name="enable_accel"             type="bool" value="true"/> '

The output is different: 

roslaunch realsense2_camera opensource_tracking.launch
... logging to /home/jooj/.ros/log/db14eae4-1a37-11ed-9604-f93abcdc0d98/roslaunch-ubuntu-17456.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:43919/

SUMMARY
========

CLEAR PARAMETERS
 * /points_xyzrgb/
 * /rtabmap/rgbd_odometry/
 * /rtabmap/rtabmap/
 * /ukf_se/

PARAMETERS
 * /ImuFilter/_publish_tf: False
 * /ImuFilter/_world_frame: enu
 * /ImuFilter/use_mag: False
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
[... 680 lines cut ...]
[ WARN] (2022-08-12 05:31:20.554) OdometryF2M.cpp:525::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=125) between -1 and 195"
[ INFO] [1660307480.555860752]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.409648s
[ WARN] (2022-08-12 05:31:21.193) OdometryF2M.cpp:525::computeTransform() Registration failed: "Not enough inliers 16/20 (matches=137) between -1 and 196"
[ INFO] [1660307481.194536368]: Odom: quality=16, std dev=0.000000m|0.000000rad, update time=0.428181s
[ WARN] (2022-08-12 05:31:21.656) OdometryF2M.cpp:525::computeTransform() Registration failed: "Not enough inliers 19/20 (matches=126) between -1 and 197"
[ INFO] [1660307481.662038395]: Odom: quality=19, std dev=0.000000m|0.000000rad, update time=0.463789s
[ WARN] (2022-08-12 05:31:22.115) OdometryF2M.cpp:525::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=145) between -1 and 198"
[ INFO] [1660307482.118648146]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.448170s
[ERROR] (2022-08-12 05:31:22.121) Rtabmap.cpp:1277::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 3134 is ignored!
[ INFO] [1660307482.121930010]: rtabmap (57): Rate=1.00s, Limit=0.000s, Conversion=0.0031s, RTAB-Map=0.0001s, Maps update=0.0000s pub=0.0000s (local map=41, WM=41)
[ INFO] [1660307482.736840429]: Odom: quality=69, std dev=0.099413m|0.055939rad, update time=0.614914s
[ INFO] [1660307483.168307855]: Odom: quality=325, std dev=0.022557m|0.032761rad, update time=0.421968s
[ INFO] [1660307483.511671270]: Odom: quality=330, std dev=0.023452m|0.032761rad, update time=0.220963s
[ INFO] [1660307483.873888075]: Odom: quality=325, std dev=0.021415m|0.050171rad, update time=0.340619s
[ INFO] [1660307484.356416645]: rtabmap (58): Rate=1.00s, Limit=0.000s, Conversion=0.0022s, RTAB-Map=0.8363s, Maps update=0.0032s pub=0.0019s (local map=42, WM=42)
[... cut ...]
[ INFO] [1660307485.994782988]: Odom: quality=248, std dev=0.039508m|0.043116rad, update time=0.822405s
[ WARN] (2022-08-12 05:31:26.572) OdometryF2M.cpp:529::computeTransform() Registration failed: "Not enough inliers 11/20 (matches=188) between -1 and 205" (guess=xyz=-0.016283,0.017912,-0.005712 rpy=0.000174,-0.004470,0.050659)
[ WARN] (2022-08-12 05:31:26.589) OdometryF2M.cpp:290::computeTransform() Failed to find a transformation with the provided guess (xyz=-0.016283,0.017912,-0.005712 rpy=0.000174,-0.004470,0.050659), trying again without a guess.

JoaoVitorBranco, 2022-08-12:
I also changed the "initial_reset" in this file, but the RViz didn't read anything....

JoaoVitorBranco, 2022-08-12:
I think now it's another problem, not the related to the title "pixeled 2d map with D435". I will close this issue and create another one

closed 2022-08-12 by JoaoVitorBranco