realsenseai/realsense-ros #2440: How to convert depth image to laser scan?

pannatorn6395, 2022-08-05:


MartyG-RealSense, 2022-08-05:
Hi @pannatorn6395  Have you investigated using the ROS2 Foxy branch of **depthimage_to_laserscan** to convert a depth image to laser scan, please?

https://github.com/ros-perception/depthimage_to_laserscan/tree/foxy-devel

pannatorn6395, 2022-08-05:
yes, @MartyG-RealSense I used it already but I don't know why depth image can't publish to laser scan and I can't change topic /depth.would you recommend, please
[image]

MartyG-RealSense, 2022-08-05:
I have not personally used depthimage_to_laserscan, though in a past case about it I provided the following information regarding guidance for getting it working.

*********

If you have not been able to launch depthimage_to_laserscan, a guide in the link below for doing so with D435 and the RealSense ROS wrapper may be helpful.

https://programmersought.com/article/62234275154/

If you are able to launch it successfully, the RealSense ROS forum link at https://github.com/IntelRealSense/realsense-ros/issues/510 discusses use of depthimage_to_laserscan for mapping with D435.

***********

The original discussions about depthimage_to_laserscan that the above information originated from are at https://github.com/IntelRealSense/librealsense/issues/9272 and https://github.com/IntelRealSense/librealsense/issues/9294

closed 2022-08-25 by MartyG-RealSense