realsenseai/realsense-ros #2441: Can't See\Access nearly any camera topics with ROS

DanielZamanski, 2022-08-06:
I have installed both the SDK and the ROS-wrapper according to your support manuals. While I haven't gotten any error messages during the installation process and can view the camera's Input fine using the realsense-viewer software
[image]

, I'm running into trouble trying to run it via ROS.
When running the command 'roslaunch realsense2_camera rs_camera.launch' I get a very limited amount of topics when using 'rostopic list', with no camera information and no image.
[image]

Neither Rviz nor rqt_image_view detect them as well.
[image]
[image]

 I've tried launching through different .launch files within the folder and while the results vary slightly I still get no image and little to no information. I've tried updating the drivers,re-installing and installing in a clean workspace and the problems persist. Thank you for your help.

MartyG-RealSense, 2022-08-06:
Hi @DanielZamanski   Please try launching with the instruction below.

**roslaunch realsense2_camera rs_camera.launch enable_gyro:=true enable_accel:=true enable_fisheye1:=true enable_fisheye2:=true enable_pose:=true**

DanielZamanski, 2022-08-08:
> Hi @DanielZamanski Please try launching with the instruction below.
> 
> **roslaunch realsense2_camera rs_camera.launch enable_gyro:=true enable_accel:=true enable_fisheye1:=true enable_fisheye2:=true enable_pose:=true**
Thank you for the fast response!
I've tried it, I now get the following warnings when running:
[image]

and while I do get more topics now, it still doesnt include most of the topics mentioned on the camera's github page and there's still no camera and image topic
[image]

Also, when adding the new topics to RVIZ i get "Status:Error":
[image]

MartyG-RealSense, 2022-08-08:
There are some topics in the ROS wrapper documentation that you will not be able to obtain because a T265 does not have built-in depth sensing like the RealSense 400 Series models do.  It is intended that the T265 is used primarily for tracking and paired with a 400 Series camera near it (such as both on a mount together) if depth sensing is required.  

The topics that a T265 on its own can publish in the ROS wrapper are listed at the link below.

https://dev.intelrealsense.com/docs/ros-wrapper#t265-tracking-camera

Outside of the ROS wrapper, depth sensing can be simulated on T265 with the fisheye sensors using the Python example script at the link below but the results are not as accurate as a true depth-sensing 400 Series camera

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/t265_stereo.py

DanielZamanski, 2022-08-08:
> There are some topics in the ROS wrapper documentation that you will not be able to obtain because a T265 does not have built-in depth sensing like the RealSense 400 Series models do. It is intended [...]

I'm still not getting most of the topics mentioned in that link. No Odometry, no camera info and no Image for example

MartyG-RealSense, 2022-08-08:
What happens if you use either of the T265-specific roslaunch instructions provided by the wrapper?

roslaunch realsense2_camera rs_t265.launch

roslaunch realsense2_camera demo_t265.launch

DanielZamanski, 2022-08-08:
> What happens if you use either of the T265-specific roslaunch instructions provided by the wrapper?
> 
> roslaunch realsense2_camera rs_t265.launch
> 
> roslaunch realsense2_camera demo_t265.launch

I get camera IMU and odometry, but still no camera info and no image. (Also no point cloud but I assume it's a matter of editing the launch input once everything else works)

MartyG-RealSense, 2022-08-08:
Though your camera shows as being on a USB 3 connection in the RealSense Viewer image at the start of this issue, if the ROS wrapper's launch was detecting it as a USB 2.1 connection then the pose stream data would be available but fisheye images would not as they require USB 3.

https://github.com/IntelRealSense/librealsense/blob/master/doc/t265.md#what-are-the-system-requirements

Bearing in mind that some topics including fisheye are disabled by default, let's try the enabling instructions with the T265-specific launches.

roslaunch realsense2_camera rs_t265.launch enable_gyro:=true enable_accel:=true enable_fisheye1:=true enable_fisheye2:=true enable_pose:=true

roslaunch realsense2_camera demo_t265.launch enable_gyro:=true enable_accel:=true enable_fisheye1:=true enable_fisheye2:=true enable_pose:=true

DanielZamanski, 2022-08-08:
> Though your camera shows as being on a USB 3 connection in the RealSense Viewer image at the start of this issue, if the ROS wrapper's launch was detecting it as a USB 2.1 connection then the pose [...]

The same issue persists, unfortunately. 
I'm adding screenshots of the terminal output when I run the roslaunch command, maybe there's a settings issue or an error that I'm missing - 

[image]
[image]
[image]

MartyG-RealSense, 2022-08-08:
The 'image' in RViz usually refers to the RGB color image but the T265 does not have an RGB color sensor, just monochrome fisheye.

DanielZamanski, 2022-08-08:
> The 'image' in RViz usually refers to the RGB color image but the T265 does not have an RGB color sensor, just monochrome fisheye.
> 
> On the technical specification document for the T265 at the link below, it states **RGB Sensor: No**
> 
> https://ark.intel.com/content/www/us/en/ark/products/192742/intel-realsense-tracking-camera-t265.html

I should still be able to get Image, camera, pose graph and pointcloud topics. I've even tried switching for the D435 camera but I still get the same issue of a very limited amount of topics published

MartyG-RealSense, 2022-08-08:
Okay, let's look at your SDK installation.  You are using librealsense SDK 2.50.0 and wrapper 2.3.2 which is the correct match and you are using ROS1 Noetic.

How did you install the SDK and wrapper on Ubuntu?  Did you install them separately (installing librealsense first and the ROS wrapper secondly after that) or did you install both librealsense and the wrapper at the same time from packages with the wrapper's **Method 1** instructions?

DanielZamanski, 2022-08-08:
Method 2

MartyG-RealSense, 2022-08-08:
Ok, so you installed the wrapper from source code after having first installed librealsense from packages or from source and had no problems with librealsense or the RealSense Viewer?

DanielZamanski, 2022-08-09:
> 

The realsense viewer seems to be working fine, it shows images from both cameras and tracks the camera correctly

MartyG-RealSense, 2022-08-09:
Thanks very much for your continued patience!

When you are using the Viewer, do you also have a launch active in ROS or is realsense-viewer the only RealSense program that is using the camera?  If ROS is active, does running only realsense-viewer make a difference?

**Failed to set power state** typically means that the camera was unable to be accessed, perhaps because it is already busy because a stream has been started or because it could not be detected.

MartyG-RealSense, 2022-08-21:
Case closed due to no further comments received.

closed 2022-08-21 by MartyG-RealSense

andreasbrecl, 2023-04-05:
@DanielZamanski were you able to solve this issue? I am currently experiencing the exact same problem with my D435 camera.

Landau33, 2023-06-22:
same problem with neotic ros, t265, and no camera topics

DanielZamanski, 2023-06-29:
> @DanielZamanski were you able to solve this issue? I am currently experiencing the exact same problem with my D435 camera.

Not really. When we switched from ROS1 to ROS2 it started working so maybe it's a ROS installation issue but I weren't able to figure it out.