realsenseai/realsense-ros #2442: How to create local costmap2D from intel realsense D435i

tharittapol, 2022-08-09:
Hello, I'm beginner for ROS, could anyone recommend me please.

Now, I can already open the camera and public depth image topic. But i don't know next step to convert depth to local costmap.

Thank a lot.

tharittapol, 2022-08-09:
In the meantime, I tried depthimage_to_laserscan pkg, which received a laser scan. But I don't know how to use that laser to make a local costmap.

MartyG-RealSense, 2022-08-10:
Hi @tharittapol   A comon method of generating costmap2d is to use a ROS package called **costmap_2d**

http://wiki.ros.org/costmap_2d

https://roboticsknowledgebase.com/wiki/common-platforms/ros/ros-cost-maps/

********

The only guide that I could find for using this package with RealSense cameras was for the RealSense R200 stereo depth camera from 2015 but you may be able to convert the knowledge for use with D435i (the modern descendant of R200).

https://blog.csdn.net/Jaryblueky/article/details/54407998

As the guide is not in English language, I have posted a translation below.

*********

Using realsense R200 can directly get /camera/color/image_raw, /camera/depth/image_raw and /camera/depth/points.. To convert to a 2D costmap, use the costmap_2d package.

The costmap_2d package only supports data conversion in laserScan, pointCloud and pointCloud2 formats.  Fortunately, /camera/depth/points of Realsense R200 is in pointCloud2 format and can be used directly with costmap_2d.  However, the data in pointCloud2 format is too large, and the operation may freeze.

In this case, you can use the packge named depthimage_to_laserscan-indigo-devel to convert the depth data in sensor_msg/Image format of /camera/depth/image_raw into laserScan.  And then use costmap_2d.

Both methods are available, use pointClouds2 directly, or convert the depth data to laserScan and use costmap_2d. If other binocular cameras do not have the output in pointCloud format, they can only be converted to laserScan first.

Create a new yaml file in the costmap_2d/launch folder with an arbitrary name, such as example.yaml, and write the following code: The focus is on the two lines of code marked in red, the data type is PointCloud2, and the topic is the name.

Both methods are available, use pointClouds2 directly, or convert the depth data to laserScan and use costmap_2d. If other binocular cameras do not have the output in pointCloud format, they can only be converted to laserScan first.

Create a new yaml file in the costmap_2d/launch folder with an arbitrary name, such as example.yaml, and write the following code: The focus is on the two lines of code marked in red, the data type is PointCloud2, and the topic is the name.

**Note: the yaml code in the article could not be copy-pasted, so please refer to the code in the article link.**

Then, roslaunch realsense_camera r200_default.launch, run r200 first, in roslaunch costmap_2d example.launch.
At this point, it is found that costmap_2d is not running normally. Open rviz to watch, and costmap_2d is not converted normally.

After observing the launch file of costmap_2d, it is found that when running r200, there is a parameter /user_sim_time in the default example.launch file of costmap_2d to set it to true.  But when running realsense and costmap_2d, no node provides the value of /user_sim_time. Then set it to false.  Run it again and find that costmap_2d can be converted normally.

************""

As an alternative to the costmap approach described above, Intel have published a modern robotics guide for D435i and the ROS wrapper at the link below that states that it can get updates via costmap, though it is a more complex project.

https://www.intel.com/content/www/us/en/develop/documentation/edge-insights-amr-2022-1-doc/top/tutorials/tutorials-amr/wandering-application-with-realsense-camera.html

tharittapol, 2022-08-10:
thank you for the advice  I will try to follow your advice.

MartyG-RealSense, 2022-08-26:
Hi @tharittapol  Do you require further assistance with this case, please?  Thanks!

tharittapol, 2022-08-29:
> Hi @tharittapol Do you require further assistance with this case, please? Thanks!

Hi, @MartyG-RealSense . Thank you for following up.

Now I can apply laser scan to costmap_2d and render costmap on rviz, but the rendered costmap is not erased when the object or laser scan is no longer there. (It's like it updates the marking, but it doesn't update when clearing)

What should I do in order for costmap to detect objects from a realtime laser scan?

I have attached a test video below.

https://user-images.githubusercontent.com/57582928/187142783-77ad1f7b-b791-49f2-87fc-ced924a60867.mp4

tharittapol, 2022-08-29:
This is the costmap parameters.
[image]

This is the launch file for costmap.
[image]

This is the depth_image to laser_scan launch file.
[image]

MartyG-RealSense, 2022-08-29:
There is a case at https://github.com/IntelRealSense/librealsense/issues/9294 that sounds similar to yours where the RealSense ROS user in that case was using depthimage_to_laserscan but finding that obstacles were not being cleared in time, causing problems for the robot navigation.  Their eventual solution was described at https://github.com/IntelRealSense/librealsense/issues/9294#issuecomment-879579215

closed 2022-09-10 by MartyG-RealSense