realsenseai/realsense-ros #2443: Problem:rtabmap_ros/Tutorials/HandHeldMapping use RealSense D435i

heidytsai, 2022-08-09:
I run:
roslaunch realsense2_camera rs_camera.launch \
    align_depth:=true \
    unite_imu_method:="linear_interpolation" \
    enable_gyro:=true \
     enable_accel:=true
Terminal show:
RLException: [rs_camera. launch] is neither a launch file in package [realsense2_ camera] nor is [realsense2_camera] a launch file name The traceback for the_exception was written to the log file
How can I solve?

heidytsai, 2022-08-10:
Hi @MartyG-RealSense 
Excuse me. I have executed the following code.
git clone https://github.com/jetsonhacksnano/installLibrealsense
cd installLibrealsense
./installLibrealsense.sh
./buildLibrealsense.sh
sudo apt-get install libcanberra-gtk-module libcanberra-gtk3-module
sudo reboot
realsense-viewer
It should be SDK you said right? So I try to install wrapper for my ROS Melodic, but it goes wrong. Where am I doing wrong?

mkdir -p ~/catkin_ws/src
cd ~/catkin_ws/src/
git clone https://github.com/IntelRealSense/realsense-ros.git
cd realsense-ros/
git checkout `git tag | sort -V | grep -P "^2.\d+\.\d+" | tail -1`
cd ..
catkin_init_workspace
cd ..
catkin_make clean
catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release
catkin_make install
echo "source ~/catkin_ws/devel/setup.bash" >> ~/.bashrc
source ~/.bashrc
roslaunch realsense2_camera rs_camera.launc

MartyG-RealSense, 2022-08-10:
https://github.com/jetsonhacksnano/installLibrealsense would be a fine method for installing the RealSense SDK if your computer is an Nvidia Jetson, which the installLibrealsense script. 

If you have a desktop or laptop PC with Ubuntu then installing the SDK and wrapper together at the same time with **Method 1** may work well for you, so you would not have to install the wrapper afterwards because Method 1 has already done it for you.

If you have a desktop or laptop PC (not a Jetson) and want to try this, please follow the steps below.

1.  Uninstall the SDK installation that you did earlier by using the command below to remove all packages with 'realsense' in their name.

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#uninstalling-the-packages

2.  Install the SDK and wrapper together with Method 1 using apt-get install

https://github.com/IntelRealSense/realsense-ros#method-1-the-ros-distribution

*******

I note that the roslaunch instruction that you listed above was **roslaunch realsense2_camera rs_camera.launc** without an h on the end of launch.  Can you confirm that you entered the roslaunch instruction correctly please?

heidytsai, 2022-08-10:
thanks you. I use Nvidia Jetson nano. Can you teach me how to install wrapper? 

oh! I didn't post h here. But didn't miss when executed.

MartyG-RealSense, 2022-08-10:
JetsonHacks also have a Jetson Nano installation script for the RealSense ROS wrapper that you could try on your Nano.

https://jetsonhacks.com/2019/10/25/realsense-ros-wrapper-jetson-nano/

heidytsai, 2022-08-11:
In your link "$ git checkout <version number>" Where can I find my version number? I can't find.

Thanks you very much.

MartyG-RealSense, 2022-08-11:
The instructions seem to be referring to the JetPack **L4T** version number (33.2.1, for example).  If you know which JetPack version you are using (such as 4.5.1 or 4.6) then you can find its equivalent L4T number to use in the git checkout instruction by looking up your version on the JetPack Archive page at the link below.

https://developer.nvidia.com/embedded/jetpack-archive

heidytsai, 2022-08-11:
Do I have to install JetPack first?!??

MartyG-RealSense, 2022-08-11:
My feeling about this section of the instructions is that if you already have JetPack installed (which you very likely do) then skip this section of the instructions and just use these commands:

**$ git clone https://github.com/JetsonHacksNano/installRealSenseROS
$ cd installRealSenseROS
$ ./installRealSenseROS <catkin workplace name>**

heidytsai, 2022-08-11:
I executed "$sudo apt-cache show nvidia-jetpack", and terminal show:
Package: nvidia-jetpack
Version: 4.6-b199 
I think I should have Jetpcak

I did "
$ git clone https://github.com/JetsonHacksNano/installRealSenseROS
$ cd installRealSenseROS
$ ./installRealSenseROS <catkin workplace name>
" yesterday.
Do you mean that I should skip "$ git checkout <version number>", and execute "$ roslaunch realsense2_camera rs_camera.launch" ?

MartyG-RealSense, 2022-08-11:
Yes, skip checkout and try inputting **roslaunch realsense2_camera rs_camera.launch** into the ROS terminal.

heidytsai, 2022-08-11:
Terminal show:
[ERROR] [1660213530.506900347]: Failed to load nodelet [/camera/realsense2_camera] of type [realsense2_camera/RealSenseNodeFactory] even after refreshing the cache: Could not find library corresponding to plugin realsense2_camera/RealSenseNodeFactory. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1660213530.507060194]: The error before refreshing the cache was: Could not find library corresponding to plugin realsense2_camera/RealSenseNodeFactory. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[FATAL] [1660213530.507454420]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[camera/realsense2_camera-2] process has died [pid 21476, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/abcd/.ros/log/9923c1c0-195e-11ed-9dda-1418c3dc730a/camera-realsense2_camera-2.log].
log file: /home/abcd/.ros/log/9923c1c0-195e-11ed-9dda-1418c3dc730a/camera-realsense2_camera-2*.log
How can I solve? Thanks.

MartyG-RealSense, 2022-08-11:
Another person who experienced this set of errors said:

********

I solved it by doing **catkin build** in the catkin workspace and then **source devel/setup.bash**

heidytsai, 2022-08-12:
Sorry. I have a stupid question. Is ORB_SLAM and RTABMAP an equal system or is there a hierarchical relationship? 
Do I have to install ORB_SLAM before I use RTABMAP?

MartyG-RealSense, 2022-08-12:
ORB_SLAM and RTABMAP are two different software tools for performing SLAM navigation.  ORB_SLAM2 can be integrated with rtabmap_ros though.

A PDF guide at the link below about using them together installs rtabmap_ros first and then ORB_SLAM2 afterwards.

https://webdocs.cs.ualberta.ca/~zhang/c631/lectures/Demo_RTAB-Map_ORB-SLAM2.pdf

*******

Alternatively, you can compile RTABMAP with ORB_SLAM2 included in it.

http://official-rtab-map-forum.206.s1.nabble.com/How-to-use-ORB-SLAM2-library-in-rtabmap-td4230.html

heidytsai, 2022-08-17:
Hi, I got a new SD card and started over,
I use: Jetson nano (melody), Ubuntu18.04, intel D435i
Goal: Use rtabmap for indoor positioning and mapping
let me clear my mind
(1) RealSense SDK = librealsense?
(2) RealSense ROS wrapper = realsense-ros ?
(3) As far as I know, I need the packages included in realsense-ros: librealsense2, realsense2_camera? So do I need to have both librealsense and librealsense2?
This information confuses me, how do I start my steps? thank you for your help

MartyG-RealSense, 2022-08-18:
1 and 2.  RealSense SDK = librealsense2 and RealSense ROS Wrapper = realsense-ros

3.  'librealsense' refers to an older generation of the RealSense SDK that does not support the 400 Series cameras.  **librealsense2** is the correct SDK to use.  librealsense2 should be installed first and then the realsense-ros wrapper secondly.

4.  No, the ROS wrapper is not included in the SDK and has to be installed separately.

*********

Instead of using the JetsonHacks install scripts, you could try the official Intel installation instructions instead.

1.  Install librealsense2 from Jetson-compatible packages at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#4-install-with-debian-packages

2.  Once you have confirmed that the librealsense2 SDK is working correctly, install realsense-ros by building it from source code using the wrapper's **Method 2** instructions.

https://github.com/IntelRealSense/realsense-ros#step-2-install-intel-realsense-ros-from-sources

heidytsai, 2022-08-19:
I can't add the server to the list of repositories.
s**udo add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo bionic main" -u**
Terminal show:
E: Failed to fetch http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/bionic/InRelease  403  Forbidden [IP: 52.218.109.66 80]
E: The repository 'http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

MartyG-RealSense, 2022-08-19:
If you are using the Jetson package install instructions, what happens if you use the command as written in the instructions with **lsb_release -cs** instead of using apt-repo bionic, please?

**sudo add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo $(lsb_release -cs) main" -u**

heidytsai, 2022-08-19:
I found the problem
I executed the following command incorrectly earlier
**sudo add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic main" -u**
This will cause the new URL to be wrong as well, right? But I don't know how to rescue, please help me. Thank you very much!

MartyG-RealSense, 2022-08-19:
That's great news - thanks very much for the update  :)

heidytsai, 2022-08-22:
Hi, I found that the teaching website that installs ros will install the ddynamic_reconfigure package, but the official website does not mention it. Is this necessary?

MartyG-RealSense, 2022-08-22:
In the ROS1 instructions for building the wrapper from source code, it advises to clone ddynamic_reconfigure into the catkin workspace.

[image]

You can alternatively install ddynamic_reconfigure with apt install.

**Kinetic**
sudo apt install ros-kinetic-ddynamic-reconfigure

**Melodic**
sudo apt install ros-melodic-ddynamic-reconfigure

**Noetic**
sudo apt install ros-noetic-ddynamic-reconfigure

The 'Releases' link that you quoted lists all RealSense ROS wrapper versions for ROS1 and ROS2 in the date order that they were released.  Versions start start with '4' are the current **ros2_beta** wrapper, versions that start with 3 are the older **ros2** wrapper and versions that start with '2' are the ROS1 wrapper.

For SDK 2.50.0 the wrapper versions compatible with that librealsense SDK version at the time of writing this are:

ros2_beta: 4.0.4
ros2: 3.2.3
ROS1: 2.3.2

heidytsai, 2022-08-22:
Can I specify version 2.3.2 as instructed in Method 2: The RealSense™ distribution? I can't find the relevant code.

MartyG-RealSense, 2022-08-22:
The source code for ROS1 wrapper **2.3.2** is available from the link below, in the **Assets** file list at the bottom of that page.

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.3.2

If you build from packages with Method 1 using **sudo apt-get install ros-$ROS_DISTRO-realsense2-camera** though then librealsense 2.50.0 and wrapper 2.3.2 should be installed together at the same time.

If you have already installed librealsense then you should remove that SDK version from your computer first before running the Method 1 apt-get installation.

https://github.com/IntelRealSense/realsense-ros#method-1-the-ros-distribution

MartyG-RealSense, 2022-08-22:
Yes, build SDK and wrapper separately if using Jetson instead of PC.

heidytsai, 2022-08-22:
I just need to change the git clone https://github.com/IntelRealSense/realsense-ros.git of [method 2](https://github.com/IntelRealSense/realsense-ros#step-2-install-intel-realsense-ros-from-sources) to https://github.com/IntelRealSense/realsense-ros/releases/tag/2.3.2, and then execute it in sequence?

MartyG-RealSense, 2022-08-22:
If you are using the ROS1 wrapper's git to build then you should not need to change the git clone instruction as it will build the **latest** wrapper version.  For the ROS1 wrapper the latest is 2.3.2 as there were no more ROS1 wrapper versions published after 2.3.2.

heidytsai, 2022-08-22:
Oh! git commands are exclusive to ROS1?!

MartyG-RealSense, 2022-08-22:
The ROS2 wrappers have their own GitHub addresses.

**ros2**
https://github.com/IntelRealSense/realsense-ros/tree/ros2

**ros2_beta**
https://github.com/IntelRealSense/realsense-ros/tree/ros2-beta

MartyG-RealSense, 2022-08-22:
However, I believe that they use the same **git clone** address but use the git **tag** commands to include or exclude particular wrapper version numbers (for example, if searching for the tag that contains '2', ignore the ROS2 versions that start with 3 and 4).

heidytsai, 2022-08-22:
Thanks you. How can I use the git tag commands?

MartyG-RealSense, 2022-08-22:
To install the RealSense ROS1 wrapper in the Ubuntu terminal, first create a catkin workspace if you have not done so already.

```
mkdir -p ~/catkin_ws/src
cd ~/catkin_ws/src/
```

You should have now changed directory to **/catkin_/ws/src**

Next, git clone the RealSense ROS1 wrapper by inputting thr commands below into the Ubuntu terminal.

```
git clone https://github.com/IntelRealSense/realsense-ros.git
cd realsense-ros/
git checkout `git tag | sort -V | grep -P "^2.\d+\.\d+" | tail -1`
cd ..
```

*************************

If you wish to insteaad git clone the **ros2** wrapper, follow these instructions instead.

First create a ROS2 catkin workspace in the Ubuntu terminal.

```
mkdir -p ~/ros2_ws/src

cd ~/ros2_ws/src/
```

You should have now changed directory to **/ros2_ws/src/**

Next, git clone the RealSense ROS2 wrapper by inputting thr command below into the Ubuntu terminal.

```
git clone --depth 1 --branch `git ls-remote --tags https://github.com/IntelRealSense/realsense-ros.git | grep -Po "(?<=tags/)3.\d+\.\d+" | sort -V | tail -1` https://github.com/IntelRealSense/realsense-ros.git

cd ~/ros2_ws
```

heidytsai, 2022-08-23:
Ok, thanks you very much.

How to install ddynamic_reconfigure using APT?

MartyG-RealSense, 2022-08-23:
**Kinetic**
sudo apt install ros-kinetic-ddynamic-reconfigure

**Melodic**
sudo apt install ros-melodic-ddynamic-reconfigure

**Noetic**
sudo apt install ros-noetic-ddynamic-reconfigure

***********

ROS2 does not have dynamic_reconfigure.  In the **ros2** wrapper you can though still change parameters during runtine with **ros2 run rqt_reconfigure rqt_reconfigure**

heidytsai, 2022-08-23:
oh! Sorry!! Thanks you!

heidytsai, 2022-08-23:
git clone https://github.com/IntelRealSense/realsense-ros.git
cd realsense-ros/
git checkout `git tag | sort -V | grep -P "^2.\d+\.\d+" | tail -1`
cd ..

After executing the above command, I need to execute the following command, right?

catkin_init_workspace
cd ..
catkin_make clean
catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release
catkin_make install
echo "source ~/catkin_ws/devel/setup.bash" >> ~/.bashrc
source ~/.bashrc

MartyG-RealSense, 2022-08-23:
If you are building the ROS1 wrapper for Kinetic, Melodic or Noetic then yes, execute that list of instructions.

heidytsai, 2022-08-23:
Thank you for answering.

heidytsai, 2022-08-23:
May I ask what does this command mean?
**git checkout `git tag | sort -V | grep -P "^2.\d+\.\d+" | tail -1`**
Should I do **git tag | sort -V | grep -P "^2.\d+\.\d+" | tail -1** first?

MartyG-RealSense, 2022-08-23:
It is basically saying to install the latest version of the ROS wrapper, and only install the version number that begins with the number '2' (so that the latest version of the **ros2** wrapper that starts with the version number '3' is not installed.

Instead of using a git tag instruction, a simpler method is to specify the version number that you want to install, with a 'v' in front of the number.  For example:  **git checkout v2.3.2**

heidytsai, 2022-08-23:
Oh! This is the git tag command you mentioned earlier! ! Thanks you. Forgive my stupidity.

heidytsai, 2022-08-23:
Sorry to bother you. Could you please tell me the meaning of the following two commands?

catkin_init_workspace
cd ..
catkin_make clean
catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release
catkin_make install

echo "source ~/catkin_ws/devel/setup.bash" >> ~/.bashrc
source ~/.bashrc

MartyG-RealSense, 2022-08-23:
DCMAKE_BUILD_TYPE - build in **Release** mode instead of the developer 'Debug' mode.  The 'Release' mode should run better due to being optimized.

DCATKIN_ENABLE_TESTING - I am not certain what this does, other than run some tests as the name suggests.  Usually a ROS wrapper build will break if it is set to True though.

heidytsai, 2022-08-24:
Hi. Thanks you. 
I get a CMake Error when I reach the **catkin_make clean** command, How can I solve?

[image]

MartyG-RealSense, 2022-08-24:
Earlier in this discussion you you stated that you were using Noetic, but the error messages have Melodic in them.  Have you had ROS Melodic installed on your computer before you began using Noetic, please?

heidytsai, 2022-08-25:
I've been using melody.

heidytsai, 2022-08-25:
I use [this way](https://its304.com/article/weixin_43994752/123780278) and solved the Error. thank you.
But can you tell me what is [this commands](https://github.com/IntelRealSense/realsense-ros/issues/2443#issuecomment-1223824646) to do? Is it Installing the ros package ddynamic_reconfigure?? But you have provided me with the installation instructions for APT.
Now I have executed the **catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release** command, and Jetson nano crashes.

MartyG-RealSense, 2022-08-25:
The commands at https://github.com/IntelRealSense/realsense-ros/issues/2443#issuecomment-1223824646 are not related to installing ddynamic_reconfigure.  

In regard to the commands at that link: aside from the ones already discussed,  **catkin_init_workspace** creates the catkin workspace folder and **catkin_make clean** cleans up the catkin workspace and can be used to remove a build from the workspace.

**catkin_make install** builds the wrapper with the configuration that you specified in the previous **catkin_make** instruction

Have you sourced the bashrc file after performing catkin_make install

```
echo "source ~/catkin_ws/devel/setup.bash" >> ~/.bashrc
source ~/.bashrc
```

heidytsai, 2022-08-25:
After my **catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release** command ran to 96%, Jetson nano crashed. Do I need to execute it again after restarting?

MartyG-RealSense, 2022-08-25:
If the instructions ask to source bashrc then please do it even if it has been done in other projects.

If the build failed to reach 100% then execute it again after restarting.

heidytsai, 2022-08-26:
I'm very grateful. Thank you.

closed 2022-09-07 by MartyG-RealSense