realsenseai/realsense-ros #2445: Setting visual presets for D435i in ROS 2

hello-chintan, 2022-08-11:
Hello,

I am using a D435i camera with ROS 2 Galactic and looking to launch the realsense ROS 2 driver with the HighAccuracyPreset visual preset. How do I apply this preset while launching the driver? I also want to know if there is a way to change the visual preset to its default values while the camera is running without having to relaunch the driver.

Thank you!

MartyG-RealSense, 2022-08-11:
Hi @hello-chintan   You could first obtain the **json** preset file for HighAccuracyPreset from the link below if you do not have it already.

https://dev.intelrealsense.com/docs/d400-series-visual-presets#preset-table

You could then load and apply the json file during launch by defining a path to it in the ROS2 rs_launch.py launch file's **json_file_path** parameter.

https://github.com/IntelRealSense/realsense-ros/blob/ros2-beta/realsense2_camera/launch/rs_launch.py#L20

I am not aware of a method for changing the preset in ROS2 without stopping the session and relaunching.

MartyG-RealSense, 2022-08-18:
Hi @hello-chintan  Do you require further assistance with this case, please?  Thanks!

hello-chintan, 2022-08-19:
Hey @MartyG-RealSense. Thank you for the response! I got the camera to use the HighAccuracyPreset through the .json file. I have just one follow-up question on this. I understand that in ROS 1 we can switch the visual preset parameter using the dynamic_reconfigure package while the camera is active. Since dynamic_reconfigure has not been ported to ROS 2, is there an alternative to achieving this through other means, possibly by using a similar ROS 2 package? I think I am just confused that if it was possible in ROS 1, is it just the absence of the right ROS 2 package that is the hurdle?

MartyG-RealSense, 2022-08-19:
In the **ros2** wrapper branch (the previous generation of ROS2 wrapper before the current **ros2_beta** one that is currently at version 4.0.4) you can access the parameters during runtime with **ros2 run rqt_reconfigure rqt_reconfigure**

https://github.com/IntelRealSense/realsense-ros/tree/ros2#view-and-modify-camera-controls-params-in-runtime

hello-chintan, 2022-08-29:
Hey, @MartyG-RealSense. Thank you for pointing me to the resources. I'm closing the issue.

closed 2022-08-29 by hello-chintan