realsenseai/realsense-ros #2446: D435i and Release 4.0.4, Could not set param: pointcloud.stream_index_filter with -1

madgrizzle, 2022-08-11:
Using the latest release (installed via apt), I receive this error message and cannot get a pointcloud to be produced.  I found this issue (that was closed) was "solved" from building from source and was going to try that out tonight until I saw this comment a few minutes ago.

_Originally posted by @youngday in https://github.com/IntelRealSense/realsense-ros/issues/2347#issuecomment-1211997078

I can try to build from source, but I would think doing so shouldn't be the "solution".

MartyG-RealSense, 2022-08-11:
Hi @madgrizzle  I believe that the meaning of the comment was that building from source worked for that RealSense user but it would have been their preference to build from packages with apt-get if that install method had worked for them.

madgrizzle, 2022-08-11:
Hmm.. I took the comment from @youngday the completely other way... I don't think either works for @youngday.

I, personally, haven't tried building from source.  All I know is the release has this issue.

MartyG-RealSense, 2022-08-11:
Please do try a source code build if you have the opportunity to do so, as you mentioned in your opening comment that the issue **was "solved" from building from source** and so it shobe worth trying yourself.

madgrizzle, 2022-08-11:
I will try it, though I am curious as to why their would be a difference.

youngday, 2022-08-12:
I compiled librealsensense by version 2.51,point cloud and image could get from rviz2, but the warn still print out ,I can use it first,waiting for the reason of this warn ,thanks .

MartyG-RealSense, 2022-08-18:
Hi @madgrizzle  Do you require further assistance with this case, please?  Thanks!

madgrizzle, 2022-08-18:
I still get the error message after building from source, but it still outputs point clouds.  Can this error message be ignored?

MartyG-RealSense, 2022-08-18:
If the pointcloud appears to be performing well with no obvious negative effects despite frequently appearing log warnings then yes, the warnings can likely be safely ignored.

MartyG-RealSense, 2022-08-24:
Hi @madgrizzle  Do you have an update about this case that you can provide, please?  Thanks!

madgrizzle, 2022-08-24:
It "seems" to work even though I get the error message, though I would like to know what the error represents and why I get it, but I'll close nevertheless.

closed 2022-08-24 by madgrizzle