realsenseai/realsense-ros #2447: Flying D435i

JoaoVitorBranco, 2022-08-12:
Coming from issue [pixelated 2d map with D435 #2439](https://github.com/IntelRealSense/realsense-ros/issues/2439):

JoaoVitorBranco, 2022-08-12:
It didn't work. But I saw something. All the params that I'm changing in the command line weren't changing the params. In all the last terminal outputs, the "enable_gyro" and "enable_accel" are "False". Now I changed the values in the "includes/nodelet.launch.xml" file, and now they appear as "True". I made this change:

'param name="enable_gyro" type="bool" value="true"/ !--$(arg enable_gyro)--'
' param name="enable_accel" type="bool" value="true"/ !--$(arg enable_accel)--'

The output is different:

roslaunch realsense2_camera opensource_tracking.launch
... logging to /home/jooj/.ros/log/db14eae4-1a37-11ed-9604-f93abcdc0d98/roslaunch-ubuntu-17456.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:43919/

SUMMARY
CLEAR PARAMETERS

/points_xyzrgb/
/rtabmap/rgbd_odometry/
/rtabmap/rtabmap/
/ukf_se/
PARAMETERS

/ImuFilter/_publish_tf: False
/ImuFilter/_world_frame: enu
/ImuFilter/use_mag: False
/camera/realsense2_camera/accel_fps: -1
/camera/realsense2_camera/accel_frame_id: camera_accel_frame
/camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
/camera/realsense2_camera/align_depth: True
/camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
/camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
/camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
/camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
/camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
/camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
/camera/realsense2_camera/allow_no_texture_points: False
/camera/realsense2_camera/base_frame_id: camera_link
/camera/realsense2_camera/calib_odom_file:
/camera/realsense2_camera/clip_distance: -2.0
/camera/realsense2_camera/color_fps: -1
/camera/realsense2_camera/color_frame_id: camera_color_frame
/camera/realsense2_camera/color_height: -1
/camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
/camera/realsense2_camera/color_width: -1
/camera/realsense2_camera/confidence_fps: -1
/camera/realsense2_camera/confidence_height: -1
/camera/realsense2_camera/confidence_width: -1
/camera/realsense2_camera/depth_fps: -1
/camera/realsense2_camera/depth_frame_id: camera_depth_frame
/camera/realsense2_camera/depth_height: -1
/camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
/camera/realsense2_camera/depth_width: -1
/camera/realsense2_camera/device_type:
/camera/realsense2_camera/enable_accel: True
/camera/realsense2_camera/enable_color: True
/camera/realsense2_camera/enable_confidence: True
/camera/realsense2_camera/enable_depth: True
/camera/realsense2_camera/enable_fisheye1: False
/camera/realsense2_camera/enable_fisheye2: False
/camera/realsense2_camera/enable_fisheye: False
/camera/realsense2_camera/enable_gyro: True
/camera/realsense2_camera/enable_infra1: False
/camera/realsense2_camera/enable_infra2: False
/camera/realsense2_camera/enable_infra: False
/camera/realsense2_camera/enable_pointcloud: False
/camera/realsense2_camera/enable_pose: False
/camera/realsense2_camera/enable_sync: False
/camera/realsense2_camera/filters:
[... 678 lines cut ...]
[ WARN] (2022-08-12 05:31:20.554) OdometryF2M.cpp:525::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=125) between -1 and 195"
[ INFO] [1660307480.555860752]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.409648s
[ WARN] (2022-08-12 05:31:21.193) OdometryF2M.cpp:525::computeTransform() Registration failed: "Not enough inliers 16/20 (matches=137) between -1 and 196"
[ INFO] [1660307481.194536368]: Odom: quality=16, std dev=0.000000m|0.000000rad, update time=0.428181s
[ WARN] (2022-08-12 05:31:21.656) OdometryF2M.cpp:525::computeTransform() Registration failed: "Not enough inliers 19/20 (matches=126) between -1 and 197"
[ INFO] [1660307481.662038395]: Odom: quality=19, std dev=0.000000m|0.000000rad, update time=0.463789s
[ WARN] (2022-08-12 05:31:22.115) OdometryF2M.cpp:525::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=145) between -1 and 198"
[ INFO] [1660307482.118648146]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.448170s
[ERROR] (2022-08-12 05:31:22.121) Rtabmap.cpp:1277::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 3134 is ignored!
[ INFO] [1660307482.121930010]: rtabmap (57): Rate=1.00s, Limit=0.000s, Conversion=0.0031s, RTAB-Map=0.0001s, Maps update=0.0000s pub=0.0000s (local map=41, WM=41)
[ INFO] [1660307482.736840429]: Odom: quality=69, std dev=0.099413m|0.055939rad, update time=0.614914s
[ INFO] [1660307483.168307855]: Odom: quality=325, std dev=0.022557m|0.032761rad, update time=0.421968s
[ INFO] [1660307483.511671270]: Odom: quality=330, std dev=0.023452m|0.032761rad, update time=0.220963s
[ INFO] [1660307483.873888075]: Odom: quality=325, std dev=0.021415m|0.050171rad, update time=0.340619s
[ INFO] [1660307484.356416645]: rtabmap (58): Rate=1.00s, Limit=0.000s, Conversion=0.0022s, RTAB-Map=0.8363s, Maps update=0.0032s pub=0.0019s (local map=42, WM=42)
[ WARN] (2022-08-12 05:31:24.652) OdometryF2M.cpp:529::computeTransform() Registration failed: "Not enough inliers 6/20 (matches=155) between -1 and 203" (guess=xyz=-0.053079,0.009716,0.010459 rpy=0.036400,-0.034479 [... cut ...]
[ INFO] [1660307485.994782988]: Odom: quality=248, std dev=0.039508m|0.043116rad, update time=0.822405s
[ WARN] (2022-08-12 05:31:26.572) OdometryF2M.cpp:529::computeTransform() Registration failed: "Not enough inliers 11/20 (matches=188) between -1 and 205" (guess=xyz=-0.016283,0.017912,-0.005712 rpy=0.000174,-0.004470,0.050659)
[ WARN] (2022-08-12 05:31:26.589) OdometryF2M.cpp:290::computeTransform() Failed to find a transformation with the provided guess (xyz=-0.016283,0.017912,-0.005712 rpy=0.000174,-0.004470,0.050659), trying again without a guess.

JoaoVitorBranco, 2022-08-15:
Hi @MartyG-RealSense ! I will try to correct the not enought inliers warning. About the "kidnapped robot problem", do you know some reference that tells how to make the relocalization that can help me?

MartyG-RealSense, 2022-08-15:
If you are using rtabmap_ros then it may be most efficient to relocalize via rtabmap_ros (or 'relocate' as the reference below calls it).

http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping#Localization_mode

JoaoVitorBranco, 2022-08-15:
Okay, I will try that. Thanks for the help!

closed 2022-08-18 by JoaoVitorBranco