realsenseai/realsense-ros #2449: Roslaunch error

Userpc1010, 2022-08-16:
Platform | Desktop
-- | --
OS | Ubuntu 20.04.4 LTS
Kernel Version | 5.15.0-46-generic
Camera Model | D455
SDK Version | 2.50.0
Firmware | 05.13.00.50
ROS Version | Noetic

<p dir="auto">I have installed the librealsense2 packages and realsense-ros. Then, I try to run <code class="notranslate">roslaunch realsense2_camera rs_camera.launch filters:=pointcloud</code>. I have here is my output.</p>

> sencis@sencis-desktop:~$ roslaunch realsense2_camera rs_camera.launch
> ... logging to /home/sencis/.ros/log/98aa1290-1d95-11ed-8548-23b1e1c3ce80/roslaunch-sencis-desktop-31191.log
> Checking log directory for disk usage. This may take a while.
> Press Ctrl-C to interrupt
> Done checking log file disk usage. Usage is <1GB.
> 
> started roslaunch server http://sencis-desktop:39683/
> 
> SUMMARY
> ========
> 
> PARAMETERS
>  * /camera/realsense2_camera/accel_fps: -1
>  * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
>  * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
>  * /camera/realsense2_camera/align_depth: False
>  * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
>  * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
>  * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
>  * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
>  * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
>  * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
>  * /camera/realsense2_camera/allow_no_texture_points: False
>  * /camera/realsense2_camera/base_frame_id: camera_link
>  * /camera/realsense2_camera/calib_odom_file: 
>  * /camera/realsense2_camera/clip_distance: -2.0
>  * /camera/realsense2_camera/color_fps: -1
>  * /camera/realsense2_camera/color_frame_id: camera_color_frame
>  * /camera/realsense2_camera/color_height: -1
>  * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
>  * /camera/realsense2_camera/color_width: -1
>  * /camera/realsense2_camera/confidence_fps: -1
>  * /camera/realsense2_camera/confidence_height: -1
>  * /camera/realsense2_camera/confidence_width: -1
>  * /camera/realsense2_camera/depth_fps: -1
>  * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
>  * /camera/realsense2_camera/depth_height: -1
>  * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
>  * /camera/realsense2_camera/depth_width: -1
>  * /camera/realsense2_camera/device_type: 
>  * /camera/realsense2_camera/enable_accel: False
>  * /camera/realsense2_camera/enable_color: True
>  * /camera/realsense2_camera/enable_confidence: True
>  * /camera/realsense2_camera/enable_depth: True
>  * /camera/realsense2_camera/enable_fisheye1: False
>  * /camera/realsense2_camera/enable_fisheye2: False
>  * /camera/realsense2_camera/enable_fisheye: False
>  * /camera/realsense2_camera/enable_gyro: False
>  * /camera/realsense2_camera/enable_infra1: False
>  * /camera/realsense2_camera/enable_infra2: False
>  * /camera/realsense2_camera/enable_infra: False
>  * /camera/realsense2_camera/enable_pointcloud: False
>  * /camera/realsense2_camera/enable_pose: False
>  * /camera/realsense2_camera/enable_sync: False
>  * /camera/realsense2_camera/filters: 
>  * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
>  * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
>  * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
>  * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
[... 83 lines cut ...]
> [ INFO] [1660676388.312788827]: RGB Camera was found.
> [ INFO] [1660676388.313120472]: Motion Module was found.
> [ INFO] [1660676388.313170185]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
> [ INFO] [1660676388.313216953]: num_filters: 0
> [ INFO] [1660676388.313249451]: Setting Dynamic reconfig parameters.
> [ WARN] [1660676388.473539672]: Param '/camera/rgb_camera/power_line_frequency' has value 3 that is not in the enum { {50Hz: 1} {60Hz: 2} {Disabled: 0} }. Removing this parameter from dynamic reconfigure options.
> [ INFO] [1660676388.480632705]: Done Setting Dynamic reconfig parameters.
> /opt/ros/noetic/lib/nodelet/nodelet: symbol lookup error: /home/sencis/catkin_ws/devel/lib//librealsense2_camera.so: undefined symbol: _ZN2cv3MatC1EiiiRKNS_7Scalar_IdEE
> [camera/realsense2_camera_manager-2] process has died [pid 31245, exit code 127, cmd /opt/ros/noetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/sencis/.ros/log/98aa1290-1d95-11ed-8548-23b1e1c3ce80/camera-realsense2_camera_manager-2.log].
> log file: /home/sencis/.ros/log/98aa1290-1d95-11ed-8548-23b1e1c3ce80/camera-realsense2_camera_manager-2*.log
> [camera/realsense2_camera-3] process has finished cleanly
> log file: /home/sencis/.ros/log/98aa1290-1d95-11ed-8548-23b1e1c3ce80/camera-realsense2_camera-3*.log

logs

`[0m[ INFO] [1660675524.714065073]: Loading nodelet /camera/realsense2_camera of type realsense2_camera/RealSenseNodeFactory to manager realsense2_camera_manager with the following remappings:[0m

[0m[ INFO] [1660675524.720699057]: waitForService: Service [/camera/realsense2_camera_manager/load_nodelet] has not been advertised, waiting...[0m

[0m[ INFO] [1660675524.764486542]: waitForService: Service [/camera/realsense2_camera_manager/load_nodelet] is now available.[0m

[0m[ INFO] [1660675553.507910827]: Bond broken, exiting[0m`

[image]
[image]

MartyG-RealSense, 2022-08-17:
Hi @Userpc1010  I note that you are using kernel version 5.15.  When installing librealsense from packages, the most recent kernel version supported is 5.4.

With a source code build of librealsense, the most recent officially supported kernels are 5.8 and 5.11.  An unofficial test patch that adds support for 5.13 and 5.15 kernels is available though at https://github.com/IntelRealSense/librealsense/issues/10439#issuecomment-1171965106

Alternatively, if you install librealsense and the ROS wrapper together at the same time from packages using the ROS wrapper's **Method 1** instructions then the build should be based on the **RSUSB backend**, which bypasses the kernel.  It is therefore not dependent on Linux versions or kernel versions and does not require patching.

https://github.com/IntelRealSense/realsense-ros#method-1-the-ros-distribution

*************

Before trying Method 1, you should first remove the previous librealsense package build and ROS wrapper source code build.

1.  Remove all packages on your computer with 'realsense' in their name using the instruction below to remove the librealsense package build.

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

2.  Remove the ROS wrapper source code build by deleting the whole **catkin_ws/src** catkin workspace folder.

Userpc1010, 2022-08-18:
i run sudo apt-get install ros-noetic-realsense2-camera and i have:

E: Unable to locate package ros-noetic-realsense2-camera
& E: Unable to locate package ros-noetic-realsense-camera
& E: Unable to locate package ros-$ROS_DISTRO-realsense2-camera
& E: Unable to locate package ros-noetic-realsense2-description

MartyG-RealSense, 2022-08-19:
Are you experiencing the same errors today if you repeat the installation commands?

Userpc1010, 2022-08-19:
Sorry I added this by mistake. Now I'm installing the library on Ubuntu 18.04.06 lts karnel 5.4 everything seems fine.

MartyG-RealSense, 2022-08-19:
That's great to hear that progress seems to be successful for you!

MartyG-RealSense, 2022-08-25:
Hi @Userpc1010  Do you require further assistance with this case, please?  Thanks!

Userpc1010, 2022-08-25:
No thanks it's ok

MartyG-RealSense, 2022-08-25:
Thanks very much @Userpc1010 for the update!  As you do not require further assistance, I will close this case.  Thanks again!

closed 2022-08-25 by MartyG-RealSense