realsenseai/realsense-ros #2450: Device 1/1 failed with exception: failed to set power state

jwson97, 2022-08-17:
I installed ROS2 foxy on jetson orin (jetpack 5.0.1)

and installed ros realsens2-camera

`sudo apt-get install ros-$ROS_DISTRO-realsense2-camera`

When I tried to run as below

`ros2 launch realsense2_camera rs_launch.py`

I got error 

[realsense2_camera_node-1] [WARN] [1660699467.825580159] [camera.camera]: Device 1/1 failed with exception: failed to set power state
[realsense2_camera_node-1] [ERROR] [1660699467.825682049] [camera.camera]: The requested device with  is NOT found. Will Try again.

when I check camera is detected

$ lsusb
**Bus 002 Device 012: ID 8086:0b3a Intel Corp.** 
Bus 002 Device 003: ID 0781:5595 SanDisk Corp. 
Bus 002 Device 002: ID 0bda:0420 Realtek Semiconductor Corp. 4-Port USB 3.0 Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 046d:c534 Logitech, Inc. Unifying Receiver
Bus 001 Device 003: ID 0bda:5420 Realtek Semiconductor Corp. 4-Port USB 2.0 Hub
Bus 001 Device 002: ID 13d3:3549 IMC Networks Bluetooth Radio
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

What is the problem?

MartyG-RealSense, 2022-08-17:
Hi @jwson97  It looks as though you are using the old **ros2** branch of the RealSense ROS2 wrapper from the link below and not the current **ros2_beta** wrapper, as you are performing a `sudo apt-get install ros-$ROS_DISTRO-realsense2-camera` installation from packages (the 'Method 1' installation instructions of the ROS wrapper).   Is that correct, please?

https://github.com/IntelRealSense/realsense-ros/tree/ros2

My understanding is that a Method 1 package installation does not include CUDA support and so would not be the best choice for a Jetson board.  I would recommend instead installing the librealsense SDK and ROS wrapper separately if you wish to benefit from CUDA acceleration of color conversion, depth-color alignment and pointclouds.

May I also ask whether you first installed the librealsense SDK before using Method 1?  Because Method 1 installs both librealsense and the ROS wrapper from packages together at the same time, it would be installing librealsense a second time if it had already been installed, which could create conflicts between the two SDK installations.

MartyG-RealSense, 2022-08-24:
Hi @jwson97  Do you require further assistance with this case, please?  Thanks!

jwson97, 2022-08-24:
Sorry for late response. After several attempts, I was able to solve this problem.  Thank you. 
No further help is needed on this matter.

MartyG-RealSense, 2022-08-24:
Thanks very much @jwson97 for the update!  As you achieved a solution, I will close this case.  Thanks again!

closed 2022-08-24 by MartyG-RealSense

kkats-mech, 2023-02-03:
Could you please share the solution, as I am facing the same issue?

MartyG-RealSense, 2023-09-20:
Hi @HowieHepburn  A RealSense ROS user with Docker and  ROS2 Humble on a Jetson Orin was able to resolve the 'Failed to set power state' error by running an **udevadm** command outside of Docker, as described at https://github.com/IntelRealSense/librealsense/issues/10693#issuecomment-1191070560

HowieHepburn, 2023-09-20:
邮件已收到

102hit, 2023-09-28:
hi, I met same question!could you tell me the solution?thanks a lot

MartyG-RealSense, 2023-09-28:
Hi @102hit  Which librealsense version and firmware driver version are you using, please?  Thanks!

AlexKlimaj, 2023-12-15:
I am having this same problem. However, it appears that the realsense is functioning correctly. I have output from all three cameras and the IMU.

```
[component_container-1] [WARN] [1702661716.085183775] [visual_slam_node]: Delta between current and previous frame [33.440000] is above threshold [22.000000]
[component_container-1] [WARN] [1702661716.301112533] [visual_slam_node]: Delta between current and previous frame [33.374976] is above threshold [22.000000]
[realsense2_camera_node-2] [WARN] [1702661716.373325188] [camera.camera]: Device 1/1 failed with exception: failed to set power state
[realsense2_camera_node-2] [ERROR] [1702661716.373445384] [camera.camera]: The requested device with  is NOT found. Will Try again.
[component_container-1] [WARN] [1702661716.451613211] [visual_slam_node]: Delta between current and previous frame [33.374464] is above threshold [22.000000]
[component_container-1] [WARN] [1702661716.561114873] [visual_slam_node]: Delta between current and previous frame [33.374464] is above threshold [22.000000]
[component_container-1] [WARN] [1702661716.732069250] [visual_slam_node]: Delta between current and previous frame [33.374208] is above threshold [22.000000]
[component_container-1] [WARN] [1702661716.773638942] [visual_slam_node]: Delta between current and previous frame [33.373952] is above threshold [22.000000]
[component_container-1] [WARN] [1702661716.864696763] [visual_slam_node]: Delta between current and previous frame [33.373696] is above threshold [22.000000]

```

MartyG-RealSense, 2023-12-17:
Hi @AlexKlimaj  The only other case with that error that I could find is at https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_slam/issues/91 and involved a RealSense D455 camera.  That case did not have a clear solution, however.  But if your camera is performing normally despite the messages then it is likely to be safe to ignore them.

Stark-Bear, 2024-01-26:
> Hi @jwson97 It looks as though you are using the old **ros2** branch of the RealSense ROS2 wrapper from the link below and not the current **ros2_beta** wrapper, as you are performing a `sudo [...]

I did what you noted . Firstly I installed the librealsense SDK --version 2.5 , and I could see the camera picture (color and depth) through  realsense-viewer ; Then I installed the realsense_ros , type " sudo apt-get install ros-melodic-realsense2-camera" . But finally , " roslaunch realsense2_camera rs_camera.launch  "  did not work .  What should I do now?

yurii-drevnytskyi-dianomix, 2024-02-14:
Had same issue, installing librealsense on top of ros noetic docker, that's how I fixed it:

Installed librealsense and realsense-ros with:
```shell
# LibRealsense2
# realsense ros:    https://github.com/IntelRealSense/realsense-ros
# realsense git:    https://github.com/IntelRealSense/librealsense/blob/development/doc/distribution_linux.md
# realsense docker: https://github.com/IntelRealSense/librealsense/tree/master/scripts/Docker
RUN mkdir -p /etc/apt/keyrings
RUN curl -sSf https://librealsense.intel.com/Debian/librealsense.pgp | tee /etc/apt/keyrings/librealsense.pgp > /dev/null
RUN echo "deb [signed-by=/etc/apt/keyrings/librealsense.pgp] https://librealsense.intel.com/Debian/apt-repo `lsb_release -cs` main" | tee /etc/apt/sources.list.d/librealsense.list
RUN apt-get update -y
RUN apt-get install -y --no-install-recommends librealsense2-dkms librealsense2-utils librealsense2-dev librealsense2-udev-rules
RUN apt install -y --no-install-recommends ros-noetic-realsense2-* 
```

After installation your udev rules should be here:
**ll /lib/udev/rules.d/*realsense***
or here
**ll /etc/udev/rules.d/*realsense***

Alternatively you can copy it manually from librealsense master branch:
(**99-realsense-libusb.rules** and **99-realsense-d4xx-mipi-dfu.rules**)
https://github.com/IntelRealSense/librealsense/tree/master/config

They do this during docker installation:
https://github.com/IntelRealSense/librealsense/blob/master/scripts/Docker/Dockerfile

Then reload udev rules - run inside docker (on each host reboot, or on each container start for simplicity)
```shell
sudo apt install -y --no-install-recommends udev
sudo /lib/systemd/systemd-udevd --daemon
sudo udevadm control --reload-rules && sudo udevadm trigger
```