realsenseai/realsense-ros #2451: rosdep install error

jwson97, 2022-08-17:
I try to install ROS2 Wrapper for Intel® RealSense™ Devices on ubuntu 20.04 computer.

When I tried to 

`rosdep install -i --from-path src --rosdistro $ROS_DISTRO --skip-keys=librealsense2 -y`

I got error 

```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
realsense2_description: Cannot locate rosdep definition for [rclcpp_components]
realsense2_camera_msgs: Cannot locate rosdep definition for [builtin_interfaces]
realsense2_camera: Cannot locate rosdep definition for [rclcpp_components]

```
How Can I solve it?

jwson97, 2022-08-17:
I solved this issue. I make it sure the ros distribution is 'foxy' with source /opt/ros/foxy/setup.bash

But I got another issuse when I run the node
`
$ ros2 launch realsense2_camera rs_launch.py`

```
[realsense2_camera_node-1] /opt/ros/foxy/lib/realsense2_camera/realsense2_camera_node: symbol lookup error: /opt/ros/foxy/lib/realsense2_camera/realsense2_camera_node: undefined symbol: _ZNK12class_loader11ClassLoader14getLibraryPathB5cxx11Ev
[ERROR] [realsense2_camera_node-1]: process has died [pid 16297, exit code 127, cmd '/opt/ros/foxy/lib/realsense2_camera/realsense2_camera_node --ros-args --log-level info --ros-args -r __node:=camera -r __ns:=/camera --params-file /tmp/launch_params_o7zaesct'].

```

jwson97, 2022-08-17:
I installed [Intel® RealSense™ SDK 2.0 (v2.50.0)] and realsense-viewer is working. 

And I tried wrapper ROS2 Wrapper in Ubuntu 20.04 (foxy)

install dependencites

```
sudo apt-get install ros-foxy-cv-bridge ros-foxy-librealsense2 ros-foxy-message-filters ros-foxy-image-transport
sudo apt-get install -y libssl-dev libusb-1.0-0-dev pkg-config libgtk-3-dev
sudo apt-get install -y libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev
```

and downloaded wrapper from here [](https://github.com/IntelRealSense/realsense-ros/releases)

`colcon build --base-paths src/realsense-ros-4.0.4/`

all build was done successfully. 

But when I try to run 

```
ros2 launch realsense2_camera rs_launch.py

[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [30811]
[realsense2_camera_node-1] /home/jwson/ros2_ws/install/realsense2_camera/lib/realsense2_camera/realsense2_camera_node: symbol lookup error: /home/jwson/ros2_ws/install/realsense2_camera/lib/realsense2_camera/realsense2_camera_node: undefined symbol: _ZNK12class_loader11ClassLoader14getLibraryPathB5cxx11Ev
[ERROR] [realsense2_camera_node-1]: process has died [pid 30811, exit code 127, cmd '/home/jwson/ros2_ws/install/realsense2_camera/lib/realsense2_camera/realsense2_camera_node --ros-args --log-level info --ros-args -r __node:=camera -r __ns:=/camera --params-file /tmp/launch_params_6084uwun'].

```
Please let me know the solution

MartyG-RealSense, 2022-08-17:
Hi @jwson97  It looks as though your dependency installation instructions are based on ones in a very old ROS2 wrapper that is obsolete.

https://github.com/intel/ros2_intel_realsense

The current official RealSense ROS2 wrapper is the 4.0.4 one that you used.

If you installed the librealsense SDK wrapper first and realsense-viewer worked, and then you used the syudo apt-get dependency installation instructions **after** that, then you may have installed librealsense twice, resulting in a conflict between the two installations.

jwson97, 2022-08-18:
As you said, that's how it was installed. I installed twice and there seems to be a conflict. If so, should I uninstall librealsense? Can you tell how to do that?

MartyG-RealSense, 2022-08-18:
`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge` should uninstall all packages with 'realsense' in their name, so that should take care of removing any package installations that you performed.

As you apparently built the ROS1 wrapper for use with Noetic by installing it from packages with `apt-get install`, running the above purge command should also uninstall the ROS1 wrapper.

From your descriptions above it does not sound as though you installed the librealsense SDK only from packages (apt-get install) and not from source code, so we do not need to perform a source code uninstall of librealsense.

In regard to the RealSense ROS2 wrapper that you are using with Foxy, as you did a **colcon build** source code build of ROS2 wrapper 4.04, you should be able to remove that by deleting the whole **/ros2_ws/src** catkin workspace folder.

**************

If your particular computer prefers package builds to source code builds, you could try building ROS2 wrapper version **3.2.3** for Foxy and librealsense SDK 2.50.0 from packages using the command below.  This will install librealsense 2.50.0 and ROS2 wrapper 3.2.3 from packages together at the same time, so you will not need to separately install librealsense before using the command.

`sudo apt-get install ros-foxy-realsense2-camera`

jwson97, 2022-08-23:
Thanks for your replay. I tried to uninstall and install several times. But it show another error.

sudo apt-get install ros-foxy-realsense2-camera

reos2 launch realsense2_camera rs_launch.py

I got error.

```
[realsense2_camera_node-1] [INFO] [1661244725.684940864] [camera.camera]: RealSense ROS v4.0.4
[realsense2_camera_node-1] [INFO] [1661244725.685059037] [camera.camera]: Built with LibRealSense v2.50.0
[realsense2_camera_node-1] [INFO] [1661244725.685081386] [camera.camera]: Running with LibRealSense v2.50.0
[realsense2_camera_node-1] [WARN] [1661244725.710052020] [camera.camera]: Device 1/1 failed with exception: failed to set power state
[realsense2_camera_node-1] [ERROR] [1661244725.710099620] [camera.camera]: The requested device with  is NOT found. Will Try again.
[realsense2_camera_node-1]  23/08 17:52:05,709 ERROR [140386290882304] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
[realsense2_camera_node-1]  23/08 17:52:05,709 ERROR [140386792191744] (sensor.cpp:572) acquire_power failed: failed to set power state
[realsense2_camera_node-1]  23/08 17:52:05,710 WARNING [140386792191744] (rs.cpp:310) null pointer passed for argument "device"

```

In noetic, camera is detected and working. But in foxy, why is not camera detected?

MartyG-RealSense, 2022-08-23:
I am not certain how you have RealSense ROS 4.0.4 as your listed ROS version in the log if you used the instruction **sudo apt-get install ros-foxy-realsense2-camera** as the ros2_beta wrapper (4.0.4) does not have an apt-get installation process.  Instead, the ros2_beta wrapper is built from source code using the instructions at the link below.

https://github.com/IntelRealSense/realsense-ros/tree/ros2-beta#step-3-install-intel-realsense-ros2-wrapper-from-sources

Using **sudo apt-get install ros-foxy-realsense2-camera** should cause the **ros2** wrapper 3.2.3 and librealsense SDK 2.50.0 to be installed together at the same time from packages.

jwson97, 2022-08-24:
Hi, I tried to install ros2_beta_wrapper from source code and using sudo-apt-get install ros-foxy-realsense2-camera. After several times failure, Now it's working. Thanks so much.

MartyG-RealSense, 2022-08-24:
You are very welcome, @jwson97 - it's excellent news that you succeeded  :)

closed 2022-09-05 by MartyG-RealSense