realsenseai/realsense-ros #2452: How to publish Gyro and Accel topics under same topic?

adamanov, 2022-08-18:
Hello, 
I have d435i realsense and want to get IMU data from it.

By setting below two flags I can see 2 topics  `/camera/gyro/sample` and `camera/accel/sample ` being published, they both have a msg-type: `sensor_msgs/Imu`. However, I want acceleration and angular velocity to be published on the same topic under, for example, `/camera/imu_data`. Are there any parameters to this? I could not find it by myself. Or should I write an extra ros node to fuse these two topics into one?
```  
<arg name="enable_gyro" default="true" />
<arg name="enable_accel" default="true" />
```
All best

MartyG-RealSense, 2022-08-18:
Hi @adamanov  Accel and Gyro can be published under a single **imu** topic instead of two separate gyro and accel topics by setting the launch file option **unite_imu_method** to "linear_interpolation"

You can also enable it from the roslaunch instruction by adding the term **unite_imu_method:=linear_interpolation**

The ROS wrapper's documentation provides the following information about this function.

*********

The D435i and T265 cameras have built in IMU components which produce 2 unrelated streams: gyro - which shows angular velocity and accel which shows linear acceleration. Each with it's own frequency.  By default, 2 corresponding topics are available, each with only the relevant fields of the message sensor_msgs::Imu are filled out.  Setting unite_imu_method creates a new topic, imu, that replaces the default gyro and accel topics.  The imu topic is published at the rate of the gyro.  All the fields of the IMU message under the imu topic are filled out.

**linear_interpolation:** Every gyro message is attached by the an accel message interpolated to the gyro's timestamp.

**copy:** Every gyro message is attached by the last accel message.

adamanov, 2022-08-18:
> Hi @adamanov  Accel and Gyro can be published under a single **imu** topic instead of two separate gyro and accel topics by setting the launch file option **unite_imu_method** to [...]

Thanks for very quick reply @MartyG-RealSense that I was looking for :)

closed 2022-08-18 by adamanov