realsenseai/realsense-ros #2453: Recommended processing power for D435i open source tracking

jzstarpath, 2022-08-18:
Hi, I'm trying to use a d435i camera to build map of an indoor area and then localize in the pre-built map. I am currently running the open source tracking program on an Intel NUC with i3 processor and the performance is horrible. Before I go out and purchase a more powerful computer, can anyone please let me know the kind of processing power I should look for to accomplish my task? Thanks!

MartyG-RealSense, 2022-08-19:
Hi @jzstarpath  Whilst there is not a recommended specification for this particular task, in general if you are using a single camera then my expectation would be that an i5 processor would be sufficient for a RealSense application with high processing demands.

The **generation** number of the processor will be more important than the 'i' number though.  A recent generation of modern, powerful processor that you could search for is **intel core i5 11th generation**

If you have the budget for an i7 11th generation CPU for additional processing power then it is definitely worth considering.

jzstarpath, 2022-08-19:
Hi @MartyG-RealSense , thanks for the reply!
Another additional question I have is that whether d435 is capable of robustly tracking for 12+ hours? I am currently working on a project that requires relatively precise indoor positioning: it requires the robot to localize itself in an approximately 15m*15m room for 12+ hours each round, and the whole task includes multiple rounds. It also requires the robot to localize itself in the same world frame on every startup (so localizing relative to the start up position does not really work). 
I am aware that d435i with rtabmap provides the functionality of recording a map and navigate in the pre-recorded map. But before I purchase the processing power needed for d435i tracking, can you let me know if the requirement of robust tracking for 12+ hours can be generally achieved by the camera? If not, can you give me any hardware/software modification advices to achieve the performance? Thanks!

jzstarpath, 2022-08-19:
And does adding a T265 help with the quality of map building/tracking? Does it require slightly less computation and offer better tracking result compare to using only the d435i?

MartyG-RealSense, 2022-08-19:
RealSense 400 Series cameras have been tested by Intel for long-running and are able to operate indefinitely so long as they remain within recommended operating temperature tolerances.  The official recommended maximums are an outer casing temperature of 50 degrees C and an internal temperature of 35 degrees C.

Problems may still occur during a long-run session that are not directly related to the RealSense camera hardware.  These could include glitches in the computer hardware, its operating system software or USB equipment.  If such problems do occur though, they are typically after at least a day of continuous activity and so 12 hours should be achievable.  Use of industrial grade computer or USB hardware may reduce the risk of session-disrupting issues.

I recommend creating a separate issue on this support forum for T265 questions or studying Intel's 'Better Together' resources about using T265 and 400 Series cameras together.  In the YouTube presentation, information about D435 begins at the 17.25 point.

https://www.intelrealsense.com/depth-and-tracking-combined-get-started/

https://youtube.com/watch?v=62vm0_RZ1nU

closed 2022-09-02 by MartyG-RealSense