realsenseai/realsense-ros #2454: depth-color align from remote device using ROS

RichardPlatz, 2022-08-19:
Hello, I'm using Realsense d435i connected to Raspberry Pi 4 4GB with ROS.
When I launch
`roslaunch realsense2_camera rs_camera.launch align_depth:=true`
the topic `/camera/aligned_depth_to_color/image_raw` is published with really low frequency, around 3-4 fps.
However I noticed that when I launch (without align_depth:=true)
`roslaunch realsense2_camera rs_camera.launch`
the topics `/camera/color/image_raw` and `/camera/depth/image_rect_raw` are published at 30 fps. I imagine that the aligning process requires too much resources for the Raspberry Pi 4.
In my project I would need to connect the Realsense camera on the Raspberry placed on a robot and send the data to a PC over the network using ROS and performing SLAM using rtabmap, based on the tutorials http://wiki.ros.org/rtabmap_ros/Tutorials/RemoteMapping and http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping, where the topic `/camera/aligned_depth_to_color/image_raw` is needed. I tried using two PCs and it works as espected, but I can't with Raspberry since the topic `/camera/aligned_depth_to_color/image_raw` is published with too low frequency. Therefore I was wondering if it might be possible to perform the aligning process not on Raspberry but on the PC using depth and rgb topics published by the Raspberry. I'm new in using ROS and Realsense SDK.
Thank you very much.

MartyG-RealSense, 2022-08-19:
Hi @RichardPlatz  Slow-down such as the rate that you described has been reported by other RealSense ROS users to occur when using align or pointcloud on Raspberry Pi 4.

Does performance improve if you define a custom stream configuration for 15 FPS instead of 30 at 640x480 resolution using the roslaunch instruction below, please?

**roslaunch realsense2_camera rs_camera.launch align_depth:=true depth_width:=640 depth_height:=480 depth_fps:=15 color_width:=640 color_height:=480 color_fps:=15**

RichardPlatz, 2022-08-19:
Thank you very much for your answer.
In the previous question I forgot to say that using
**roslaunch realsense2_camera rs_camera.launch align_depth:=true depth_width:=640 depth_height:=480 depth_fps:=6 color_width:=640 color_height:=480 color_fps:=6**
the topic `/camera/aligned_depth_to_color/image_raw` is published at around 5.8 FPS.
I also ran what you suggested, i.e. **roslaunch realsense2_camera rs_camera.launch align_depth:=true depth_width:=640 depth_height:=480 depth_fps:=15 color_width:=640 color_height:=480 color_fps:=15** obtaining around 5 FPS.

Beyond the problem of the network, my questions are:
1) Is there a way so that Raspberry Pi 4 using ROS can publish `/camera/aligned_depth_to_color/image_raw` with an higher frequency, for example 30 FPS?
2) If the previous is not possible, can the depth-color alignment be performed not on Raspberry but on a second machine? I mean the Rapsberry acquires depth and color data (run the rs_camera.launch file with align_depth:=false) and then the PC  connected over the network performs the alignment.
I noticed that Raspberry can publish `/camera/color/image_raw` and `/camera/depth/image_rect_raw` at 30 FPS and using ROS over the network the topics `/camera/color/image_raw/compressed` and `/camera/depth/image_rect_raw/compressed` are received by the PC still at around 30 FPS using an ethernet cable. Therefore, I was asking if using these topics it would be possible to perform the alignment on the PC.
Thank you very much.

MartyG-RealSense, 2022-08-19:
1.  Does FPS improve if you input **rosrun rqt_reconfigure rqt_reconfigure** after launch has completed to open the **dynamic reconfigure** interface, go to the **rgb_camera** section of the options and un-tick a setting called **auto_exposure_priority**

If auto_exposure is enabled and auto_exposure_priority is disabled then the RealSense SDK should try to force the FPS to be maintained at a constant rate instead of being permitted to vary.  It may not make a significant difference if the PI's CPU is under heavy processing demand though.

[image]

2.  An ethernet cable's data transmission speed is faster than USB 2 but slower than USB 3, so a remote ethernet connection is not likely to perform as fast as a camera that is directly connected to a computer's USB 3 port.

You could look into using ROS to connect to the Pi from a more powerful computer via an SSH connection and **ROS_MASTER_URI** like in Intel's ROS guide at the link below for accessing a camera that is on a second computer.

https://github.com/IntelRealSense/realsense-ros/wiki/showcase-of-using-3-cameras-in-2-machines

RichardPlatz, 2022-08-19:
Thank you for the answer.
1) I tried to do **rosrun rqt_reconfigure rqt_reconfigure** after launch has completed and un-tick the **auto_exposure_priority**, but the FPS remain at 5.9 even if now it is mantained at a constant rate as you said.
2) I already realized the network connection using **ROS_MASTER_URI** like in Intel's ROS guide. Using two PCs everything works fine, but not with the Raspberry because it publishes the required topic `/camera/aligned_depth_to_color/image_raw` with too low frequency.

Is it possible to perform the alignment on the PC with the camera connected to the Raspberry, given that the network is not the problem?
Thank you.

MartyG-RealSense, 2022-08-20:
I did not find a way to perform the alignment on PC using the Pi-attached camera unfortunately.  However, as you are using the ROS1 wrapper, it would be worth investigating whether the **rs_rgbd** launch method instead of rs_camera could provide faster performance from your Pi.

1.  Install RGBD launch support for the particular ROS1 version you are using (Kinetic, Melodic or Noetic) using the instructions at https://github.com/IntelRealSense/librealsense/issues/10223#issuecomment-1152864261

2.  Change the roslaunch instruction to:

**roslaunch realsense2_camera rs_rgbd.launch**

It should generate an RGBD point cloud without having to use the align_depth command and the cloud should be published at **depth_image_proc**

RichardPlatz, 2022-08-23:
Thank you for the answer and sorry for the delay.
I installed the rgbd_launch package and tried
**roslaunch realsense2_camera rs_rgbd.launch align_depth:=false depth_width:=640 depth_height:=480 depth_fps:=60 color_width:=640 color_height:=480 color_fps:=60**
I used align_depth:=false since I noticed that in the rs_rgbd.launch file align_depth is set to true by default. In this way, the registered depth topic `/camera/depth_registered/sw_registered/image_rect_raw` is published by the rgbd_launch surprisingly at 60 FPS on the Raspberry! Still I have some other problems, one regarding the network transmission (I need to transfer the compressed version of the topics but compression probably requires too much resources for the Raspberry causing FPS drop) and the other problem is that the imu topic stops publishing, but in case I am not able to solve this I will open another ticket.
Thank you very much for the help.

MartyG-RealSense, 2022-08-23:
You are very welcome and I'm pleased that I could be of help  :)

MartyG-RealSense, 2022-09-04:
Case closed due to no further comments received.

closed 2022-09-04 by MartyG-RealSense

RojinZandi, 2023-05-18:
Hi!

We have a 640*480 frame from the color camera of D435 and we are getting this frames through ROS2 and we do not have access to the Realsense hardware (we only get the frames from a ROS2 topic). we tried different methods to align the depth and color stream, but it is not working. Can you help us with this matter? Thank you in advance.