realsenseai/realsense-ros #2456: camera calibration parameters in d435i

adamanov, 2022-08-23:
Hello,

I was wondering why  `rs-enumerate-devices -c` and `realsense-viewer` provides different camera calibration parameters, should not they be same? 

`rs-enumerate-devices -c`
```
 Intrinsic of "Color" / 1280x720 / {YUYV/RGB8/BGR8/RGBA8/BGRA8/Y16}
  Width:      	1280
  Height:     	720
  PPX:        	645.505981445312
  PPY:        	353.784606933594
  Fx:         	908.771240234375
  Fy:         	908.71630859375
  Distortion: 	Inverse Brown Conrady
  Coeffs:     	0  	0  	0  	0  	0  
  FOV (deg):  	70.31 x 43.22
  ```
[image]

all best

MartyG-RealSense, 2022-08-24:
Hi @adamanov  Your rs-enumerate-devices listing is for the **Color** intrinsics.  The Camera Calibration window's data is for the **Depth** intrinsics.  Please check the Depth intrinsics for 1280x720 resolution to see whether they match with the Camera Calibration window's values.

As an example, below is a comparison of the values from my D455 at 1280x720 depth resolution.

[image]

[image]

adamanov, 2022-08-25:
> Hi @adamanov Your rs-enumerate-devices listing is for the **Color** intrinsics. The Camera Calibration window's data is for the **Depth** intrinsics. Please check the Depth intrinsics for 1280x720 [...]

Hey @MartyG-RealSense  thanks for you reply, Yeah they match perfectly :) Thanks

closed 2022-08-25 by adamanov