realsenseai/realsense-ros #2458: D455 CameraInfo

oceanusxiv, 2022-08-24:
Hello, I'm currently using the Realsense D455 camera, and I noticed that for the D455, the ROS driver reports a distortion_model of `plumb_bob` for the CameraInfo topic of /camera/color/camera_info.

However, reading issues like https://github.com/IntelRealSense/librealsense/issues/8810 would seem to suggest that the actual distortion model used by the color camera of the D455 is in fact the _inverse_ brown conrady, e.g. the inverse of plumb bob, rather than plumb bob itself. Indeed looking through the code in https://github.com/IntelRealSense/realsense-ros/blob/e4938bbbfd7b285fe71f81fe69cebcc76dbf72fc/realsense2_camera/src/base_realsense_node.cpp#L1936 it appears to me that no check was ever added to see if the distortion model is inverse brown conrady, rather it defaults to plumb bob.

As such, I believe that the ROS wrapper is actually outputting incorrect calibration information when it comes to the Realsense D455 color stream. Reporting the distortion model to be plumb_bob instead of its inverse.

oceanusxiv, 2022-08-24:
[image]

MartyG-RealSense, 2022-08-24:
Hi @oceanusxiv  There are previous discussions of the D455's color distortion model at https://github.com/IntelRealSense/librealsense/issues/7734 and https://github.com/IntelRealSense/librealsense/issues/9551

On most RealSense 400 Series models the coefficients are all set to zero for reasons provided at https://github.com/IntelRealSense/librealsense/issues/1430#issuecomment-375945916

However, on the D455 camera model the **color** coefficients are non-zero.

[image]

Despite the unusual nature of the D455's color coefficients compared to other 400 Series models, they do not appear to cause problems.

oceanusxiv, 2022-08-24:
@MartyG-RealSense hi, yes I have read those previous discussions, but what I don't think has been clarified in those previous discussions, exactly, is whether these coefficients are useable by external undistort programs such as opencv. Since these distortion models are nonstandard, e.g. not simple Brown Conrady.

The documentation on projection would seem to indicate that inverse Brown Conrady requires special undistort functions in order to make it work, librealsense seems to have the necessary functions within its API, but that is of course no accessible by programs outside the driver. Im simply concerned that these camera coefficients represent a nonstandard distortion model while presenting itself as a standard one to external ROS consumers.

MartyG-RealSense, 2022-08-25:
As far as I am aware, it does not matter to the **undistort()** function of OpenCV whether the distortion model is Brown Conrady or a different one.

https://docs.opencv.org/4.x/d9/d0c/group__calib3d.html#ga69f2545a8b62a6b0fc2ee060dc30559d

An exception to this is **fisheye** images like those produced by the fisheye lenses of the RealSense T265 tracking camera, which have a dedicated **undistortImage** OpenCV function for fisheye undistortion.

https://docs.opencv.org/3.4/db/d58/group__calib3d__fisheye.html#ga167df4b00a6fd55287ba829fbf9913b9

MartyG-RealSense, 2022-08-25:
The **KANNALA_BRANDT4** reference in the code that you quoted at the beginning of this case is apparently a check for whether the Kannala Brandt distortion model that is used for the fisheye type lenses of the **RealSense T265 Tracking Camera** model is the one that is being applied.

https://github.com/IntelRealSense/librealsense/issues/3880#issuecomment-489049908

oceanusxiv, 2022-08-26:
@MartyG-RealSense The distortion parameters at play here are all somewhat low, so I do not think the naked eye would be able to really easily tell whether an image is correctly undistorted or not. It does still matter though since I am using it as input to various computer vision and SLAM algorithms, where accuracy of calibration parameters is key. I would highly appreciate some official clarification on whether it is in fact valid to use the inverse brown conrady coefficients with opencv's plumb bob undistort function.

MartyG-RealSense, 2022-08-26:
One of my Intel RealSense colleagues is going to be checking the correctness of the Brown-Conrady distortion model on D455 as part of the case at https://github.com/IntelRealSense/librealsense/issues/10811 so it would be worth following that discussion to see what emerges from my colleague's checks.

closed 2022-09-07 by MartyG-RealSense