realsenseai/realsense-ros #2459: frame_id of /camera/depth/color/points wrong?

theolitl, 2022-08-26:
When using only depth and infra1 stream with the options `align_depth:=true` `enable_pointcloud:=true`, the pointcloud is published to `/camera/depth/color/points`. 
However the frame_id of this topic is set to "camera_color_optical_frame" but no tf data is available for this frame_id. Thus the pointcloud cannot be showed in Rviz with following error:

```
Transform [sender=unknown_publisher]
For frame [camera_color_optical_frame]: Frame [camera_color_optical_frame] does not exist
```

Is there a way to change the frame_id for `/camera/depth/color/points` from "camera_color_optical_frame" to "camera_infra1_optical_frame" which is indeed available in the tf-tree? Or is there something else I am missing?

MartyG-RealSense, 2022-08-26:
Hi @theolitl  If your camera supports it, you may be able to set **infra_rgb** to true in order to stream color from the infrared sensor instead of an RGB sensor.  

The ROS wrapper documentation states about this feature: "When set to True (default: False), it configures the infrared camera to stream in RGB (color) mode, thus enabling the use of a RGB image in the same frame as the depth image, **potentially avoiding frame transformation related errors**.  When this feature is required, you are additionally required to also enable enable_infra:=true for the infrared stream to be enabled".

theolitl, 2022-08-29:
Thank you for your answer. It seems that my camera doesn't support this feature. 
I am using Intel Realsense D435 (FW version 05.13.00.50)

I get following message after launching with `enable_infra:=true` and `infra_rgb:=true`

```
(Infrared, 0) sensor isn't supported by current device! -- Skipping...
```

MartyG-RealSense, 2022-08-29:
What happens if you roslaunch with **pointcloud_texture_stream** set to RS2_STREAM_ANY instead of RS2_STREAM_COLOR?

**roslaunch realsense2_camera rs_camera.launch align_depth:=true filters:=pointcloud pointcloud_texture_stream:=RS2_STREAM_ANY**

theolitl, 2022-08-29:
This was already set at RS2_STREAM_ANY. Here is my full launch file:

```
<launch>

  <arg name="serial_no"           default=""/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="camera"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>
  <arg name="output"              default="screen"/>
  <arg name="respawn"              default="false"/>

  <arg name="fisheye_width"       default="-1"/>
  <arg name="fisheye_height"      default="-1"/>
  <arg name="enable_fisheye"      default="false"/>

  <arg name="depth_width"         default="1280"/>
  <arg name="depth_height"        default="720"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="confidence_width"    default="-1"/>
  <arg name="confidence_height"   default="-1"/>
  <arg name="enable_confidence"   default="false"/>
  <arg name="confidence_fps"      default="-1"/>

  <arg name="infra_width"         default="1280"/>
  <arg name="infra_height"        default="720"/>
  <arg name="enable_infra"        default="true"/>
  <arg name="enable_infra1"       default="false"/>
  <arg name="enable_infra2"       default="false"/>
  <arg name="infra_rgb"           default="true"/>

  <arg name="color_width"         default="320"/>
  <arg name="color_height"        default="180"/>
  <arg name="enable_color"        default="false"/>

  <arg name="fisheye_fps"         default="-1"/>
  <arg name="depth_fps"           default="30"/>
  <arg name="infra_fps"           default="6"/>
  <arg name="color_fps"           default="6"/>
  <arg name="gyro_fps"            default="-1"/>
  <arg name="accel_fps"           default="-1"/>
  <arg name="enable_gyro"         default="false"/>
  <arg name="enable_accel"        default="false"/>

  <arg name="enable_pointcloud"         default="true"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_ANY"/>
  <arg name="pointcloud_texture_index"  default="0"/>
  <arg name="allow_no_texture_points"   default="true"/>
  <arg name="ordered_pc"                default="true"/>

  <arg name="enable_sync"               default="false"/>
  <arg name="align_depth"               default="true"/>

  <arg name="publish_tf"                default="true"/>
  <arg name="tf_publish_rate"           default="0"/>

  <arg name="filters"                   default="disparity,spatial,temporal,decimation"/>
  <arg name="clip_distance"             default="-2"/>
  <arg name="linear_accel_cov"          default="0.01"/>
  <arg name="initial_reset"             default="false"/>
  <arg name="reconnect_timeout"         default="6.0"/>
  <arg name="wait_for_device_timeout"   default="-1.0"/>
  <arg name="unite_imu_method"          default=""/>
  <arg name="topic_odom_in"             default="odom_in"/>
  <arg name="calib_odom_file"           default=""/>
[... 59 lines cut ...]
      <arg name="tf_publish_rate"          value="$(arg tf_publish_rate)"/>

      <arg name="filters"                  value="$(arg filters)"/>
      <arg name="clip_distance"            value="$(arg clip_distance)"/>
      <arg name="linear_accel_cov"         value="$(arg linear_accel_cov)"/>
      <arg name="initial_reset"            value="$(arg initial_reset)"/>
      <arg name="reconnect_timeout"        value="$(arg reconnect_timeout)"/>
      <arg name="wait_for_device_timeout"  value="$(arg wait_for_device_timeout)"/>
      <arg name="unite_imu_method"         value="$(arg unite_imu_method)"/>
      <arg name="topic_odom_in"            value="$(arg topic_odom_in)"/>
      <arg name="calib_odom_file"          value="$(arg calib_odom_file)"/>
      <arg name="publish_odom_tf"          value="$(arg publish_odom_tf)"/>
      <arg name="stereo_module/exposure/1" value="$(arg stereo_module/exposure/1)"/>
      <arg name="stereo_module/gain/1"     value="$(arg stereo_module/gain/1)"/>
      <arg name="stereo_module/exposure/2" value="$(arg stereo_module/exposure/2)"/>
      <arg name="stereo_module/gain/2"     value="$(arg stereo_module/gain/2)"/>

      <arg name="allow_no_texture_points"  value="$(arg allow_no_texture_points)"/>
      <arg name="ordered_pc"               value="$(arg ordered_pc)"/>
      
    </include>
  </group>
<launch>

```

MartyG-RealSense, 2022-08-29:
Returning to your original idea of changing the frame_id of /camera/depth/color/points from "camera_color_optical_frame" to "camera_infra1_optical_frame", I researched this subject deeply but unfortunately did not find a case other than this one where somebody else had attempted this.  I do apologize.

theolitl, 2022-08-30:
Thank you for your effort. 
As a workaround I just enable the color stream at lowest resolution and fps (320x180@6fps). Although I don't need a color stream, this way the "camera_color_optical_frame" is published and the pointcloud can be displayed in Rviz. 
My intent for disabling the color stream was for performance purposes, but is seems this setup is also workable for me.

MartyG-RealSense, 2022-08-30:
It's great to hear that you found a workaround solution, @theolitl  :)

closed 2022-09-11 by MartyG-RealSense