realsenseai/realsense-ros #2463: T265 V-slam not working on ros2-beta

adityakamath, 2022-08-28:
I just updated my realsense-ros installation, and looks like a lot has changed between the ros2 and ros2-beta branches. When I try it with my T265, I do not see any kind of odom or pose transform or message being published. Only IMU values. 

Is there any particular parameter that I need to enable? I dont see one. This is the command that I use to run:

```
ros2 run realsense2_camera realsense2_camera_node --ros-args -p enable_pose:=true -p device_type:=t265 -p enable_fisheye1:=false -p enable_fisheye2:=false -p initial_reset:=true
```

IanBurwell, 2022-10-09:
I am having the same issue and thought I would add my finding thus far. I am able to get the ros2-beta driver to produce the static transforms via tf2 with the following command: 
```ros2 launch realsense2_camera rs_launch.py tf_publish_rate:=10 pose_enabled:=true pose_fps:=10 publish_odom_tf:=true publish_tf:=true base_frame_id:=link```
However, the odometry frame is not published via topic nor tf2 even with the additional parameters I specified. I will also note that the pose stream from the T265 is not enabled and all I get when launching is the following:
```
[realsense2_camera_node-1] [INFO][start():219]: Open profile: stream_type: Fisheye(1), Format: Y8, Width: 848, Height: 800, FPS: 30
[realsense2_camera_node-1] [INFO][start():219]: Open profile: stream_type: Fisheye(2), Format: Y8, Width: 848, Height: 800, FPS: 30
[realsense2_camera_node-1] [INFO][publishTopics():148]: RealSense Node Is Up!
```

I have gotten it to work once, however. This was when I first launched the realsense2_camera node in ros1 noetic, killed it, then launched the above command in ros2. My guess here is that there is some issue with requesting the pose stream from the realsense initially.

BrayanPallares, 2022-12-02:
Hi, have you found a solution to the problem?

I'm having the same behavior with the T265 and the latest version of realsense-ros in ROS-Galactic. No odom topic is published, there is no odom transform. The odom_frame_id parameter no longer exists.

IanBurwell, 2022-12-07:
I was able to get it working by setting `pose_fps` to the default 200 (it seems anything else will fail). I ran it successfully with the following parameters:
```yaml
/camera/T265:
  camera:
    ros__parameters:
      device_type: "t265"
      # camera_name: "T265"

      enable_fisheye1: false
      enable_fisheye2: false

      enable_gyro: false
      enable_accel: false

      tf_publish_rate: 50.0 
      pose_fps: 200 
      pose_enabled: true 
      publish_odom_tf: true 
      publish_tf: true 
```
Note here that I have the node named `T265` running under the `/drone` namespace. The above parameters should work in the launch command line aswell, but I chose to put them in a yaml file.

WayenVan, 2023-05-11:
For Ros2, if you look at the source code, you will find the odom_publisher now named as 'camera/pose/sample'

```cpp
data_topic_name << stream_name << "/sample";
_odom_publisher = _node.create_publisher<nav_msgs::msg::Odometry>(data_topic_name.str(), rclcpp::QoS(rclcpp::QoSInitialization::from_rmw(qos), qos));
```
if you run `ros2 topic type /camera/pose/metadata`, you will get:
`nav_msgs/msg/Odometry`

so just switch to the `/camera/pose/sample`

closed 2026-10-23 by MartyG-RealSense