realsenseai/realsense-ros #2471: Problem with topic TF

alvaropascuaI, 2022-09-06:
Hi, i try to use RTABMAP - Ros2, but i have the next error, i try to launch topic tf since module realsense2 but i can't
What i need to do for publishing data from topic /tf
[image]

MartyG-RealSense, 2022-09-06:
Hi @alvaropascuaI  There is a case of this error with a RealSense D435i and rtabmap_ros on the Nvidia Jetson support forum.

https://forums.developer.nvidia.com/t/jetson-xavier-nx-intel-realsense-d435i-camera-stop-transmit-data-after-a-time/216318

The RealSense user in that case found that the error was resolved if they reduced depth and color FPS to 15, though they still had other warnings being generated.

alvaropascuaI, 2022-09-06:
I try, but i have the same problem
[image]

MartyG-RealSense, 2022-09-06:
Which launch file of the RealSense ROS2 wrapper are you using, please?  And have you set the IMU gyro and accel topics to True, as they are disabled by default.

alvaropascuaI, 2022-09-06:
I launch with next command **ros2 run realsense2_camera realsense2_camera_node --ros-args -p initial_reset:=true -p enable_infra1:=true -p depth_width:=640 -p depth_height:=480 -p depth_fps:=15.0 -p infra1_width:=640 -p infra1_height:=480 -p infra1_fps:=30.0 -p color_width:=640 -p color_height:=480 -p color_fps:=15.0 -p enable_accel:=true -p enable_gyro:=true -p gyro_fps:=200.0 -p accel_fps:=63.0 -p enable_pointcloud:=true -p unite_imu_method:="linear_interpolation"**

MartyG-RealSense, 2022-09-06:
I note that your earlier image showing the streams that are being published does not include the gyro and accel streams.  Can you confirm please whether the IMU topics are being published?

[image]

alvaropascuaI, 2022-09-06:
I have next topics, if i launch with command that i told u before
[image]

MartyG-RealSense, 2022-09-06:
What happens if you launch with rs_camera.launch instead of ros2 run?

`ros2 launch realsense2_camera rs_launch.py depth_module.profile:=640x480x30 initial_reset:=true  pointcloud.enable:=true enable_accel:=true enable_gyro:=true gyro_fps:=200.0 accel_fps:=63.0 unite_imu_method:="linear_interpolation"`

**********

If you are using the **ros2_beta** wrapper then it is worth highlighting that in that wrapper, enable_pointcloud:=true was changed to **pointcloud.enable:=true**

You also do not need to define a long list of width, height and FPS values in the ro2_beta wrapper as you instead use a **profile** instruction such as depth_module.profile:=640x480x30 which is the same as depth_width:=640 depth_height:=480 depth_fps:=30.0 infra1_width:=640 infra1_height:=480 infra1_fps:=30.0

The full list of changes is here:

https://github.com/IntelRealSense/realsense-ros/tree/ros2-beta#please-notice-if-you-are-moving-from-realsense-ros2-branch-to-ros2-beta

alvaropascuaI, 2022-09-07:
I launch with your command and i have the next topics, and about ros2_beta i don't use 
[image]

MartyG-RealSense, 2022-09-07:
Thanks for your confirmation that you are using **ros2** instead of **ros2_beta**.

So you have tried launching with both ros2 run and ros2 launch and both methods resulted in the same error?

The Nvidia forum case that I quoted earlier is the only other occurrence of this exact error message that I could find.  In a case at https://github.com/introlab/rtabmap_ros/issues/653#issuecomment-939555927 it was suggested that the error might result from timestamps not being synchronized.

Does the error disappear if you add **global_time_enabled:=true** to your launch instruction, please?

alvaropascuaI, 2022-09-07:
Thank you, very much, I added that global_time_enabled:=true and i pass the error, i want to ask u all question in me life

MartyG-RealSense, 2022-09-07:
You are very welcome.  It's great to hear that you were successful  :)

closed 2022-09-07 by alvaropascuaI