realsenseai/realsense-ros #2472: I got a warning message after update realsense-ros version

RI-SMJ, 2022-09-07:
Dear IntelRealsense ROS team.
I use 2 l515 cameras in my robot system and I got a warning message "WARNING [547029492096] (backend-v4l2.cpp:1127) Frames didn't arrived within 5 seconds" after update realsense-ros version from 2.2.4 to 2.3.2
Can I delete or solve this problem?

PC : Jetson Xavier
OS : Ubuntu 18.04
librealsense version : 2.50.0
realsense-ros version : 2.3.2
USB type : 3.2
There are no case of pressed or twisted cable.

[ INFO] [1662515468.410336237]: Initializing nodelet with 8 worker threads.
[ INFO] [1662515469.128682758]: PlaneEstimationNodelet Constructor
[ INFO] [1662515469.129407054]: PlaneEstimationNodelet Constructor
[ INFO] [1662515469.374174203]: PlaneEstimationNodelet - onInit
[ INFO] [1662515469.375070739]: PlaneEstimationNodelet - onInit
[ INFO] [1662515469.408070798]: Initializing IMU Filter Nodelet
[ INFO] [1662515469.408324775]: Initializing IMU Filter Nodelet
[ INFO] [1662515469.408803959]: Starting ImuFilter
[ INFO] [1662515469.408965959]: Starting ImuFilter
[ INFO] [1662515469.423452381]: Using dt computed from message headers
[ INFO] [1662515469.424777344]: Using dt computed from message headers
[ INFO] [1662515469.466832473]: Imu filter gain set to 0.100000
[ INFO] [1662515469.466973959]: Gyro drift bias set to 0.000000
[ INFO] [1662515469.467040782]: Magnetometer bias values: 0.000000 0.000000 0.000000
[ INFO] [1662515469.468128569]: Imu filter gain set to 0.100000
[ INFO] [1662515469.468272232]: Gyro drift bias set to 0.000000
[ INFO] [1662515469.468334638]: Magnetometer bias values: 0.000000 0.000000 0.000000
[ INFO] [1662515469.930320615]: RealSense ROS v2.3.2
[ INFO] [1662515469.930434578]: Built with LibRealSense v2.50.0
[ INFO] [1662515469.930492376]: Running with LibRealSense v2.50.0
[ INFO] [1662515469.932047538]: RealSense ROS v2.3.2
[ INFO] [1662515469.932208001]: Built with LibRealSense v2.50.0
[ INFO] [1662515469.932312780]: Running with LibRealSense v2.50.0
[ INFO] [1662515469.985452748]:  
[ INFO] [1662515469.985679810]:  
[ INFO] [1662515470.353212875]: Device with serial number f0264493 was found.

[ INFO] [1662515470.353323478]: Device with physical ID /sys/devices/3610000.xhci/usb2/2-1/2-1:1.0/video4linux/video4 was found.
[ INFO] [1662515470.353369178]: Device with name Intel RealSense L515 was found.
[ INFO] [1662515470.356471401]: Device with port number 2-1 was found.
[ INFO] [1662515470.356651834]: Device USB type: 3.2
[ INFO] [1662515470.356815914]: Resetting device...
[ INFO] [1662515470.461817863]: Device with serial number f0264493 was found.

[ INFO] [1662515470.462011897]: Device with physical ID /sys/devices/3610000.xhci/usb2/2-1/2-1:1.0/video4linux/video4 was found.
[ INFO] [1662515470.462089665]: Device with name Intel RealSense L515 was found.
[ INFO] [1662515470.464204975]: Device with port number 2-1 was found.
[ INFO] [1662515470.722757188]: Device with serial number f0264703 was found.

[ INFO] [1662515470.723033119]: Device with physical ID /sys/devices/3610000.xhci/usb2/2-3/2-3:1.0/video4linux/video0 was found.
[ INFO] [1662515470.723182606]: Device with name Intel RealSense L515 was found.
[ INFO] [1662515470.724661022]: Device with port number 2-3 was found.
[ INFO] [1662515470.724911446]: Device USB type: 3.2
[ INFO] [1662515470.725089064]: Resetting device...
init
[ INFO] [1662515476.428789182]:  
[ INFO] [1662515476.692612291]: Device with serial number f0264493 was found.

[ INFO] [1662515476.693064108]: Device with physical ID /sys/devices/3610000.xhci/usb2/2-1/2-1:1.0/video4linux/video0 was found.
[ INFO] [1662515476.693150740]: Device with name Intel RealSense L515 was found.
[ INFO] [1662515476.694766149]: Device with port number 2-1 was found.
[ INFO] [1662515476.694888337]: Device USB type: 3.2
[ INFO] [1662515476.699618940]: getParameters...
[ INFO] [1662515476.752524547]:  
[ INFO] [1662515476.969347638]: setupDevice...
[ INFO] [1662515476.969548744]: JSON file is not provided
[ INFO] [1662515476.969735321]: ROS Node Namespace: service_visioncamera_right
[ INFO] [1662515476.969837314]: Device Name: Intel RealSense L515
[ INFO] [1662515476.969969614]: Device Serial No: f0264493
[... 48 lines cut ...]
[ INFO] [1662515477.564902577]: RGB Camera was found.
[ INFO] [1662515477.565535338]: Motion Module was found.
[ INFO] [1662515477.565693624]: Add Filter: pointcloud
[ INFO] [1662515477.567599618]: num_filters: 1
[ INFO] [1662515477.567708235]: Setting Dynamic reconfig parameters.
[ INFO] [1662515477.751355707]: Done Setting Dynamic reconfig parameters.
[ INFO] [1662515477.753652104]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1662515477.756883848]: color stream is enabled - width: 960, height: 540, fps: 30, Format: RGB8
[ INFO] [1662515477.757142143]: gyro stream is enabled - fps: 400
[ INFO] [1662515477.757594792]: accel stream is enabled - fps: 400
[ INFO] [1662515477.757762806]: setupPublishers...
[ INFO] [1662515477.766450205]: Expected frequency for depth = 30.00000
[ INFO] [1662515477.853145388]: Expected frequency for color = 30.00000
[ INFO] [1662515477.932284633]: Start publisher IMU
[ INFO] [1662515477.936875314]: setupStreams...
 07/09 10:51:18,174 WARNING [547247604096] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
[ WARN] [1662515478.183018840]: 
[ WARN] [1662515478.183198279]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.
[ INFO] [1662515478.188927872]: SELECTED BASE:Depth, 0
 07/09 10:51:18,217 WARNING [546853343616] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
[ INFO] [1662515478.231422107]: SELECTED BASE:Depth, 0
[ INFO] [1662515478.243775931]: First IMU message received.
[ INFO] [1662515478.253255837]: RealSense Node Is Up!
[ WARN] [1662515478.285877620]: 
[ WARN] [1662515478.286074661]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.

MartyG-RealSense, 2022-09-14:
Hi @RI-SMJ  Do you require further assistance with this case, please?  Thanks!

RI-SMJ, 2022-09-14:
@MartyG-RealSense Thank you so much!! I will test again with the version you recommended as soon as possible.

closed 2022-09-14 by RI-SMJ