realsenseai/realsense-ros #2474: Missing frames in the aligned_depth_to_color/image_raw topic

ricber, 2022-09-09:
Hi, I report a problem we are facing with a D435i camera. As you can see from the following screenshot of rqt_bag, we regularly miss frames for some seconds when we record the depth data with the d435i camera.
[image]

What can be the cause of the problem? 

The D435i camera is connected directly to a USB 3.0 port with the official Intel cable. The Ubuntu version is 18.04, and the ROS distro is melodic. RealSense ROS version is v2.3.2 and LibRealSense is v2.50.0. The device FW version is 05.12.14.50. Everything is running on a Shuttle PC with an Intel Core i7-4790S CPU.

When the camera is started, we also get some WARNINGS on the terminal. Look at the screenshot below. 
[image]

MartyG-RealSense, 2022-09-09:
Hi @ricber  Does the number of warnings reduce if you add **global_time_enabled:=true** to your roslaunch instruction, please?

ricber, 2022-09-09:
No, the parameter **global_time_enabled** did not reduce the number of warnings nor the missing frames. Moreover, I am not sure the warnings are related to the missing frames problem.

MartyG-RealSense, 2022-09-09:
The warning **Resource temporarily unavailable**, when generating continuously like in your provided image, can indicate that there is a serious communication problem between the camera and computer, 

Do the warnings reduce if you add **initial_reset:=true** to the roslaunch instruction to reset the camera at launch if you are not using this instruction already?

MartyG-RealSense, 2022-09-09:
If a **control_transfer returned** warning only occurs occasionally after launch (such as once every one or two minutes) then it can safely be ignored.

 I note that you are using a computer with an ** i7-4790S** processor.  Although it is an i7 processor, the first number after the 'i' number can be more important as it indicates the age of the processor technology in terms of its architecture **generation**.  The '4' in 4790S shows that it is 4th Generation (codename Haswell), which dates to around 2014-2015.  

https://www.intel.com/content/www/us/en/products/sku/80808/intel-core-i74790s-processor-8m-cache-up-to-4-00-ghz/specifications.html

although RealSense 400 Series cameras can work with any Intel processor, the processor in this particular computer may therefore be having difficulty processing the volume of camera data being generated by your launch due to the age of the processor's technology.

closed 2022-09-21 by MartyG-RealSense