realsenseai/realsense-ros #2475: ROS2 Humble: Install error about "rosdep"

mat-macnica, 2022-09-12:
Hello,

I tried to install realsense-ros2-wrapper on Jetson AGX Xavier NX based on below guide link.
https://github.com/IntelRealSense/realsense-ros/tree/ros2-beta#step-3-install-intel-realsense-ros2-wrapper-from-sources

Current Configuration:
  Jetson Jetson AGX Xavier NX with Jetpack 5.0.2 which is L4T 35.1
  Docker running Ubuntu 20.04 with ROS2 Humble

I've succeed to run realsense-viewer already by librealsense2 build process on JetsonNX.
But I found Error like below in rosdep command; 
[image]

I also found "Support ROS2 Humble" comment below link;
I expect proper behavior with "ros2-beta branch" on Humble Environment.
https://github.com/IntelRealSense/realsense-ros/compare/development...ros2-beta
[image]

So, We would appreciate if you tell us how to solve this issue.
–
Best regards,

MartyG-RealSense, 2022-09-12:
Hi @mat-macnica   There is information about installing for ROS2 Humble at https://github.com/IntelRealSense/librealsense/issues/10439#issuecomment-1186588624 and the comments below it.

https://github.com/IntelRealSense/realsense-ros/issues/2309#issuecomment-1085877405 suggests a method for resolving the **No definition of [xacro]** warning on Ubuntu 20.04 Focal.

mat-macnica, 2022-09-13:
Hi,

Thank you very much for your kindness.
I really appreciate your help.

I'd like to check the behavior with your guide.

Best regards,

closed 2022-09-13 by mat-macnica