realsenseai/realsense-ros #2476: Pointcloud filter type meesed d435i

tugbakara, 2022-09-12:
Hi @MartyG-RealSense  ! :)
GPU: Nvidia Jetson AGX
2 D435i 

I have that type of launch: 
```
<launch>
  <arg name="serial_no"           default=""/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="camera"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>
  <arg name="output"              default="screen"/>
  <arg name="respawn"              default="false"/>

  <arg name="fisheye_width"       default="-1"/>
  <arg name="fisheye_height"      default="-1"/>
  <arg name="enable_fisheye"      default="false"/>

  <arg name="depth_width"         default="848"/>
  <arg name="depth_height"        default="480"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="confidence_width"    default="-1"/>
  <arg name="confidence_height"   default="-1"/>
  <arg name="enable_confidence"   default="true"/>
  <arg name="confidence_fps"      default="-1"/>

  <arg name="infra_width"         default="848"/>
  <arg name="infra_height"        default="480"/>
  <arg name="enable_infra"        default="false"/>
  <arg name="enable_infra1"       default="false"/>
  <arg name="enable_infra2"       default="false"/>
  <arg name="infra_rgb"           default="false"/>

  <arg name="color_width"         default="1280"/>
  <arg name="color_height"        default="720"/>
  <arg name="enable_color"        default="true"/>

  <arg name="fisheye_fps"         default="-1"/>
  <arg name="depth_fps"           default="30"/>
  <arg name="infra_fps"           default="30"/>
  <arg name="color_fps"           default="60"/>
  <arg name="gyro_fps"            default="-1"/>
  <arg name="accel_fps"           default="-1"/>
  <arg name="enable_gyro"         default="true"/>
  <arg name="enable_accel"        default="true"/>

  <arg name="enable_pointcloud"         default="false"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_COLOR"/>
  <arg name="pointcloud_texture_index"  default="0"/>
  <arg name="allow_no_texture_points"   default="false"/>
  <arg name="ordered_pc"                default="true"/>

  <arg name="enable_sync"               default="true"/>
  <arg name="align_depth"               default="true"/>

  <arg name="publish_tf"                default="true"/>
  <arg name="tf_publish_rate"           default="0"/>

  <arg name="filters"                   default=""/>
  <arg name="clip_distance"             default="2"/>
  <arg name="linear_accel_cov"          default="0.01"/>
  <arg name="initial_reset"             default="false"/>
  <arg name="reconnect_timeout"         default="6.0"/>
  <arg name="wait_for_device_timeout"   default="-1.0"/>
  <arg name="unite_imu_method"          default="copy"/>
[... 156 lines cut ...]
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_rgb: False
```
I am using rtabmap with Realsense to create map BUT when we select filter as pointcloud in realsense launch side, Realsense gets crazy and freezes in RVIZ and didn't give any pointcloud data in topic, I don't know what problem can cause this?

MartyG-RealSense, 2022-09-12:
Hi @tugbakara  There is a Jetson-specific known issue with pointcloud generation in the RealSense ROS wrapper that does not occur on desktop / laptop PCs.  If the pointcloud filter is not enabled then the issue does not occur on Jetson.  

Further information about it can be found at https://github.com/IntelRealSense/realsense-ros/issues/1967

I note that you are using the ROS1 wrapper as your launch file is rs_camera.launch.  The best known solution on ROS1 for the Jetson pointcloud problem is to use librealsense 2.43.0 and ROS wrapper 2.2.23, as described at https://github.com/IntelRealSense/realsense-ros/issues/1967#issuecomment-954814317

An alternative on ROS1 is to use the **rs_rgbd** launch file to generate the pointcloud in a different way that is not affected by the Jetson-specific issue.  Instructions for an RGBD launch are at https://github.com/IntelRealSense/realsense-ros/issues/2454#issuecomment-1221254065

hsynklceng, 2022-09-12:
Hi Marty,
I have one question! why did not work pointcloud with 2.50 but its work with 2.43 ?  it must be more compatible newest version. isn't it?

MartyG-RealSense, 2022-09-12:
Hi @hsynklceng  There is not currently a known reason why the pointcloud problem on Jetson manifested after librealsense 2.43.0 / wrapper  2.2.23.  There is an official internal bug report for the issue.  However, the focus of RealSense ROS wrapper development is now on the **ros2_beta** ROS2 wrapper.

tugbakara, 2022-09-12:
Thanks @MartyG-RealSense ! :) 
We will try them, have a nice day.

closed 2022-09-12 by tugbakara

tugbakara, 2022-09-15:
Hi @MartyG-RealSense , I edited closed issue that's why I think you didn't see my edit and I add another question in it. Can you look at ?

MartyG-RealSense, 2022-09-15:
My sincere apologies for the oversight.  

rs_rgbd publishes an **ordered** point cloud by default, whilst rs_camera publishes an **unordered** pointcloud by default.  You can though publish an ordered cloud with rs_camera by adding the term **ordered_pc:=true** to the roslaunch instruction.