realsenseai/realsense-ros #2479: No RGB frames with realsense2 D435 inside jetson container

harel314, 2022-09-15:
Hi,
I am trying to run 2 realsense cameras D435 and D435i with ros2 foxy (in order to implement shared memory with zero copy), and 
I am using the ```ros:foxy-ros-base-l4t-r35.1.0``` container inside a Jetson, with:

ubuntu version 20.04.5 LTS
L4T version Linux vision-server 4.9.253-tegra
Build cuda_11.4.r11.4

I builded from source the realsense2 2.48.0 since it's the same version i use outside the container. 

I modified the docker_run.sh so it may detect the cameras (and it works)
and i modified  the ```/etc/docker/daemon.json``` as mentioned in the guide.

The only problem is that when I run the camera's realsense-viewer or ros2 launch file, I don't get any RGB frames, only depth frames, with the following error:

[image]

[image]

I haven't modified the camera settings and it works fine outside the container.

both realsense-viewer and launch file retrieve same results, hence it's probably not the wrapper fault.

any other suggestions?

I may mention i tried before using 2.51 as well, with same results.

thanks in advance

MartyG-RealSense, 2022-09-16:
Hi @harel314  There was a previous case at https://github.com/IntelRealSense/librealsense/issues/7366 where a Docker user who built librealsense from source code found that the RealSense Viewer could display depth but not RGB.  The solution that worked for them in their particular case was to leave out CUDA support for the build by removing **-DBUILD_WITH_CUDA=true** from the CMake instruction for building the SDK from source.

harel314, 2022-09-18:
It worked in my case too! Thanks !

closed 2022-09-18 by harel314

MartyG-RealSense, 2022-09-18:
That's excellent news, @harel314 - thanks very much for the update  :)