realsenseai/realsense-ros #2481: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'

HripsimeS, 2022-09-16:
Hello @doronhi 

I use Intel® RealSense™ LiDAR Camera L515, installed ROS Noetic for Ubuntu, all python and ROS python dependencies. 
Then installed realsense2_camera  with the following command  sudo apt-get install ros-noetic-realsense2-camera

I used the command below to launch the camera but received saying failed to load nodelet (please find in attached)
roslaunch realsense2_camera rs_camera.launch

Can you please let me know how to fix this error, thanks in advance for your help!

[image]

MartyG-RealSense, 2022-09-16:
Hi @HripsimeS  The command **sudo apt-get install ros-noetic-realsense2-camera** should have installed the RealSense SDK (librealsense) and the RealSense ROS wrapper together at the same time from packages.

May I ask the following questions, please

1.  Before you performed **sudo apt-get install ros-noetic-realsense2-camera**, have you previously installed the librealsense SDK on your computer?  Because the apt-get instruction installs both librealsense and the wrapper, if librealsense had already been installed then a second librealsense version would be installed, creating potential conflicts.

2.  Does the error disappear if you unplug the micro-sized end of the USB cable from the base of the L515 camera, reverse the connector's orientation and plug it back into the camera, then perform roslaunch?

HripsimeS, 2022-09-16:
@MartyG-RealSense thanks for your quick reply. In order to make sure that I don't have a double installation for librealsense, I deleted everything and installed things from the scratch. Now when I run the commond to launch camera, I get the error that "No realsense devices were found", please find in attached.

Can you please tell me what is the issue and how to fix it. Thanks in advance!

[image]

MartyG-RealSense, 2022-09-16:
Is the camera detected if you add **initial_reset:=true** to the end of the roslaunch instruction to reset the camera at launch?

If the camera still is not detected, what information is provided if you input **lsusb** into the Ubuntu terminal?

HripsimeS, 2022-09-16:
@MartyG-RealSense I have tried the command you mentioned and camera is still not detected. Also while typing lsusb into the Ubuntu terminal, it does not show any results just going next line of terminal.
I use Intel Realsense Viewer and by connected my camera with a USB cable I can see RGB camera, L500 Depth Sensor and Montion module. So I don't think the issue with USB as it works Intel Realsense Viewer. 

Any ideas what is going on and how to fix this? Thanks a lot in advance!

MartyG-RealSense, 2022-09-16:
After the roslaunch fails, does the problem disappear if you unplug the camera, wait a couple of seconds and plug the camera back in, then repeat the roslaunch command?

Also, which Ubuntu version and kernel version are you using, please?

HripsimeS, 2022-09-17:
@MartyG-RealSense I followed your instruction to unplug the camera wait few seconds and plug the camera back in then repeat the roslaunch command. Unfortunatly the problem did not disappear...
I have a Windows OS on my laptop, but I installed Ubuntu on WSL on Windows 10. Ubuntu 20.04.5 LTS 
Kernel version: 5.10.16.3-microsoft-standard-WSL2

MartyG-RealSense, 2022-09-17:
Kernel 5.10 is not officially supported by the RealSense SDK.  Unsupported kernels can be used, but there may be unpredictable consequences in regards to stability.  

When librealsense is built from packages with apt-get, the newest kernel version officially supported is **5.4**.  When the SDK is built from source code, the most recent supported kernels are 5.8 and 5.11.  An unofficial test patch at https://github.com/IntelRealSense/librealsense/issues/10439#issuecomment-1171965106 adds support for 5.13 and 5.15.

It is more likely to be a WSL2-related issue than a kernel one though.  Another RealSense user at https://github.com/IntelRealSense/librealsense/issues/10506#issuecomment-1125860744 had camera detection problems when using WSL2.  They managed to detect the camera with Windows 11 after adjusting some settings, but did not succeed with Windows 10.

HripsimeS, 2022-09-17:
@MartyG-RealSense thanks for the links and tips. As you mentioned my issue is more likely to be a WSL2-related issue than a kernel, I tried the tips @tomstix that he proposed. I did completely unplugging/replugging the camera, also restarting WSL by restarting a LxssManager. But it unfortunatly did not fix the issue. So the only solution now is to install Windows 11 ? 
It might not also fix the solution because he had a D455 camera and I use L515. 

Did you have more submited issue related to L515 camera and WSL2 ?  Thanks in advance!

MartyG-RealSense, 2022-09-17:
WSL2 is discussed on the RealSense support forums very rarely, unfortunately.  This used to be mainly because it did not support USB devices and so RealSense cameras would not work with it.   As mentioned by tomstix, Microsoft have added USB support to WSL2 since then.  

I note that in the WSL2 instructions for connecting USB devices that tomstix linked to, Microsoft recommend Windows 11 for WSL2, though they indicate that Windows 10 support is possible by "building your own USBIP enabled WSL 2 kernel following the instructions in the USBIPD-WIN project repo": https://github.com/dorssel/usbipd-win/wiki/WSL-support

At https://github.com/IntelRealSense/librealsense/issues/8217#issuecomment-1003316783 an L515 user stated that they created their own USBIP as suggested by Microsoft and WSL2 could then access the camera but it would not show in the RealSense Viewer.

As you interest is in ROS rather than the Viewer, it may be worth trying to create a USBIP for Windows 10 if you have not done so already.

HripsimeS, 2022-09-19:
@MartyG-RealSense thank you very much for your reply! I did upgrade for my Windows to 11, later following the instruction for attach a USB device and running the command roslaunch realsense2_camera rs_camera.launch gave other type of errors. 

Device 1/1 failed with exception: failed to set power state, The requested device with is NOT found. Will Try again.
(sensor.cpp:572) acquire_power failed: failed to set power state
(rs.cpp:310) null pointer passed for argument "device"
(handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS

Do these errors have seen previously with other users and what could be the solutions? Look forward to hearing from you!

[image]

MartyG-RealSense, 2022-09-19:
**Failed to set power state** can indicate that the camera was unable to be accessed for some reason, such as not being detected or not having permission to access the camera.

Please check in the Windows privacy settings whether permission to access the camera hardware and permission for applications to use the camera are both set to On.  Instructions for doing so can be found at https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

HripsimeS, 2022-09-19:
@MartyG-RealSense thanks for your quick reply. That part for camera has been done already. 

I think the issue is probably is related to not having permissions without using sudo. So I followed the instruction 
https://stackoverflow.com/questions/62134563/how-to-give-permission-to-intel-realsense-camera-on-ubuntu/62369143

1) git clone https://github.com/IntelRealSense/librealsense.git (saved in \\wsl.localhost\Ubuntu-20.04\home\user\librealsense) 
2) cd librealsense
3) sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/
4)  sudo udevadm control --reload-rules
Failed to send reload request: No such file or directory

So why I got this error on 4th line, am I in wrong directory to run that command ?

MartyG-RealSense, 2022-09-19:
In Ubuntu, device handling permissions are handled by the **udev** rules.  In the RealSense Viewer in Ubuntu, do you see a warning message **Multiple realsense udev-rules were found!** ?  This would typically appear in the 'Dismiss' message boxes in the top corner of the Viewer window.

HripsimeS, 2022-09-20:
@MartyG-RealSense thanks for your reply! I did instalation following this instruction https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

When I realsense-viewer command on Ubuntu to verify the instalation, it is opening a Intel RealSense Viewer and I did not really have that error you mentioned above for "Multiple realsense udev-rules were found". But by opening a Collapse Debug Console Window I notice something like "backend-v4l2.cpp:508 - Cannot access /sys/class/video4linux", please find in attached.

Later when I run the command "sudo udevadm control --reload-rules" on Ubuntu terminal, I received this message "Failed to send reload request: No such file or directory".

Any ideas what is going on and how to fix this udev issue?

[image]

MartyG-RealSense, 2022-09-20:
Does it make any difference if you use the command below to install the udev rules?

**sudo apt install librealsense2-udev-rules**

If that does not work then I would next try installing librealsense separately from packages on its own instead of using the ROS wrapper Method 1 instructions to see whether the Viewer can then work correctly.

1.  Remove all RealSense related packages with this instruction in the Ubuntu terminal:

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

2.  Install librealsense from packages using the instructions here:

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

Once librealsense is confirmed as working then you could then install the ROS wrapper from source code.

HripsimeS, 2022-09-21:
@MartyG-RealSense I found this instruction to install Intel RealSense SDK and ROS Wrapper for Intel RealSense Devices.
https://eu.mouser.com/applications/getting-started-with-realsense-d455/

1) For Intel RealSense SDK Installation on the 6th step to run the permissions script (./scripts/setup_udev_rules.sh) I had results:
**Setting-up permissions for RealSense devices
Failed to send reload request: No such file or directory
udev-rules successfully installed**

**-->** Why it failed to send reload request by saying no such file or directory? I can see I do have a setup_udev_rules.sh file in my \librealsense\scripts directory. How is it possible to fix?

2) For the 7th step build and apply patched kernel modules (./scripts/patch-realsense-ubuntu-lts.sh) I had this result:
**Unsupported kernel version 5.15.57.1-microsoft-standard-WSL2 . 
The Focal patches are maintained for Ubuntu LTS with kernel 5.4, 5.8, 5.11 only**

**-->** If  5.15.57.1-microsoft-standard-WSL2 version is unsupported kernel version, how can I downgrade my kernel version to 5.4, 5.8 or 5.11 for Ubuntu 20.04 LTS ?

Look forward to hearing from you!

HripsimeS, 2022-09-21:
@MartyG-RealSense thanks for your reply and tips! I followed the instruction for the LibUVC-backend installation, it took some time instalation but finially it sucsesfully ended with message "Librealsense script completed". Next I run the command **rs-enumerate-devices** to verify the installation and the result was **No device detected. Is it plugged in?**

Then I followed this instruction to connect USB device https://learn.microsoft.com/en-us/windows/wsl/connect-usb
After re-running the command **rs-enumerate-devices** I received the folloing outcome 
**21/09 13:28:57,314 ERROR [140364689733376] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
21/09 13:28:57,319 ERROR [140364723331264] (sensor.cpp:576) acquire_power failed: failed to set power state
Could not create device - failed to set power state . Check SDK logs for details**

I assume the simple libuvc backend build script did not work in my case, I will try a second more complex procedure for performing an RSUSB build from source code with CMake. Hope this will work for me.

MartyG-RealSense, 2022-09-21:
You are very welcome.  I look forward to your next update.   Good luck!

HripsimeS, 2022-09-21:
@MartyG-RealSense I downloaded the source code, then I go to the librealsense folder where all my files are located and from there I run the command to create a sub-directory called 'build' and automatically navigate to it. I received the following:

**user@LAPTOP-C513COD3:~/librealsense$ mkdir build && cd build
mkdir: cannot create directory ‘build’: Permission denied**

Am I doing something wrong or this method did not work also in my case?

MartyG-RealSense, 2022-09-21:
In the librealsense folder, the 'root' folder that the **mkdr build && cd build** command should be used in is the one that contains files and folders.  Did you input the command whilst in this directory, please?

[image]

HripsimeS, 2022-09-21:
@MartyG-RealSense yes I run from that folder, but my mistake was probably to copy librealsense folder to **\\wsl.localhost\Ubuntu-20.04\home\hripsime** and then run from there on ubuntu terminal. 

I saved the librealsense folder on my Desktop, from Command Promt started wsl and with cd went to librealsense folder 
1)  /mnt/c/Users/Hripsime/Desktop/librealsense$ **mkdir build && cd build**
2) /mnt/c/Users/Hripsime/Desktop/librealsense/build$ **cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true**

After second command line I received the following error:

**CMake Error at /usr/share/cmake-3.16/Modules/CMakeDetermineSystem.cmake:185 (configure_file):
configure_file Problem configuring file
Call Stack (most recent call first):
 CMakeLists.txt:3 (project)**

**-- Configuring incomplete, errors occurred!
See also "/mnt/c/Users/Hripsime/Desktop/librealsense/build/CMakeFiles/CMakeOutput.log".
See also "/mnt/c/Users/Hripsime/Desktop/librealsense/build/CMakeFiles/CMakeError.log".**

MartyG-RealSense, 2022-09-21:
Do you experience the same problem if you strip the build instruction back to the minimum?

**cmake ../ -DFORCE_RSUSB_BACKEND=true**

HripsimeS, 2022-09-21:
@MartyG-RealSense please find below the output of command cmake ../ -DFORCE_RSUSB_BACKEND=true

[image]

MartyG-RealSense, 2022-09-21:
It looks as though it is trying to install graphics support and failing.  So please add -DBUILD_EXAMPLES=false -DBUILD_GRAPHICAL_EXAMPLES=false to prevent it from trying to install graphics support.

**cmake ../ -DFORCE_RSUSB_BACKEND=true -DBUILD_EXAMPLES=false -DBUILD_GRAPHICAL_EXAMPLES=false**

Thanks very much for your continued patience!

HripsimeS, 2022-09-21:
@MartyG-RealSense your advice helped to succeed that command, you can find in attached. Thank you!
Now I need to run this **command sudo make uninstall && make clean && make -j8 && sudo make install**  ?

[image]

MartyG-RealSense, 2022-09-21:
It's great to hear that you made progress!

Yes, please input that command.

HripsimeS, 2022-09-21:
Please find in attached the middle part of the installations where I noticed some errors and the last few lines are these:

**[ 19%] Linking CXX static library librealsense-file.a
[ 19%] Built target realsense-file
make: *** [Makefile:130: all] Error 2**

I assume something went wrong with the last command. How is possible to fix it?

[image]

MartyG-RealSense, 2022-09-21:
The advice at https://github.com/IntelRealSense/librealsense/issues/6436#issuecomment-653942842 may be relevant to the **config.h:  No such file or directory** error.

HripsimeS, 2022-09-21:
@MartyG-RealSense thank you very much! I did the following and seems the process ended without errors

$ sudo apt-get install libusb-1.0-0-dev
$ git clone https://github.com/IntelRealSense/librealsense.git
$ cd librealsense/
$ mkdir build && cd build
$ cmake ../ -DFORCE_RSUSB_BACKEND=true -DBUILD_EXAMPLES=false -DBUILD_GRAPHICAL_EXAMPLES=false
$ sudo make uninstall && make clean && make && sudo make install

Please let me know how I can check if this installation is done correctly and it can detect my camera?

MartyG-RealSense, 2022-09-21:
You can check the camera to see whether it can be detected by inputting **lsusb** into the Ubuntu terminal (that's LSUSB).

If the camera is detected then you could try building again with -DBUILD_EXAMPLES=true instead of false so that graphicless text tools and examples are built that can be used to test the camera, such as **rs-hello-realsense** that prints the distance of an observed object in the center of the camera's view as a text value in meters.

HripsimeS, 2022-09-21:
@MartyG-RealSense with a command **lsusb** into the Ubuntu terminal it could detect my camera with this output:
**Bus 002 Device 002: ID 8086:0b64 Intel Corp. Intel(R) RealSense(TM) 515
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub**

But when I try again to build with -DBUILD_EXAMPLES=true it ended with an error 
CMake Error at /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:146 (message):
Could NOT find X11 (missing: **X11_X11_INCLUDE_PATH X11_X11_LIB**)

Is this possible to fix somehow to not a get error on CMakeOutput.log ?

[image]

MartyG-RealSense, 2022-09-21:
The L515 camera has been successfully detected for the first time, which is excellent.

It should not need the X11 library if building with build_examples = true and build_graphical_examples = false.  

Let's try adding **-DCMAKE_BUILD_TYPE=Release** to the CMake build instruction to build librealsense in **Release** mode with optimizations included (it builds in Debug mode by default otherwise).

**cmake ../ -DCMAKE_BUILD_TYPE=Release -DFORCE_RSUSB_BACKEND=true -DBUILD_EXAMPLES=false -DBUILD_GRAPHICAL_EXAMPLES=false**

HripsimeS, 2022-09-21:
@MartyG-RealSense the buidling with command **cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true** and then with **sudo make uninstall && make clean && make && sudo make install** went good and the process ended withour error. 

But when I run the command  **rs-hello-realsense** I received No device connected (in attached)

Do I need to install also a ROS Wrapper for Intel RealSense Devices or that command should work even without wrapper?

[image]

MartyG-RealSense, 2022-09-21:
rs-hello-realsense is a librealsense program and so does not require the ROS wrapper.

HripsimeS, 2022-09-21:
@MartyG-RealSense when I rund a command **rs-enumerate-devices** the outpu is following
**ERROR [139698232760064] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
ERROR [139698266357952] (sensor.cpp:576) acquire_power failed: failed to set power state
Could not create device - failed to set power state . Check SDK logs for details**

MartyG-RealSense, 2022-09-21:
As you are using an L515, is the camera detected if you try unplugging the micro-sized end of the USB cable from the base of the camera, reversing the connector's orientation and plugging it back into the L515 camera (USB-C cables are two-way insertion at the micro-sized end).

HripsimeS, 2022-09-21:
@MartyG-RealSense I have tried it reverse side also,  **lsusb**  command can detect a device but **rs-enumerate-devices** gives the same error as before. For me it seems the error is connected to RS2_USB_STATUS_ACCESS. 
Maybe librealsense does not work on L515 cameras? Any ideas the reason of this error? Thanks in advance!

MartyG-RealSense, 2022-09-21:
The librealsense SDK is fully compatible with L515 cameras from version 2.36.0 onwards.  

Are you using the official 1 metre cable supplied with the camera or a longer cable of your own choice, please?

As you have a Windows computer, have you been able to test the camera successfully with the Windows version of librealsense?

HripsimeS, 2022-09-21:
@MartyG-RealSense thanks for your reply and help! Yes I am using a 1 metre cable supplied with camera.
Yes I tested the Windows version of Intel Realsense Viewer and it works perfectly. 
For the update of L515's firmware driver I don't know, as the camera is not my personal camera.

I saw other people also had exactly the same issue and there is a solution in comments to check if there is a **99-realsense-libusb.rules** in this directory  **/etc/udev/rules.d/**  
https://github.com/IntelRealSense/realsense-ros/issues/1408

I tried to run the command for udev rules with **sudo udevadm control --reload-rules** and received **Failed to send reload request: No such file or directory** Where I can find /etc/udev/rules.d/ directory, where is it installed usually ?

MartyG-RealSense, 2022-09-21:
As the firmware driver is stored within the camera hardware instead of on the computer, it can be updated in the RealSense Viewer's **More** menu in Windows and then be used in Linux.  If it is not your personal camera though then you may wish to get approval for this update first though.

Does the **/etc** rule install if you go to the root directory of the source code folder and input the command below?

`sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/`

HripsimeS, 2022-09-21:
@MartyG-RealSense I have tried the command **sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/** and the outcome is **cp: cannot stat 'config/99-realsense-libusb.rules': No such file or directory**
I have also tried to run the command **realsense-viewer** to see if I have some errors there and the outcome is in attached.

Can you please send a bit more details how update of L515's firmware driver. Thanks a lot in advance!

[image]

MartyG-RealSense, 2022-09-21:
The L515 firmware can be downloaded as a .bin file from the link below.  The most recent firmware is **1.5.8.1**

https://dev.intelrealsense.com/docs/firmware-releases-l500#l500-series-firmware-downloads

Once the file is downloaded to your computer, you can launch the Viewer and left-click on the **More** option to drop down a menu.  Select **Update Firmware** from the menu to open a file selection box.  Navigate to the downloaded .bin file on your computer and select it to automatically start the firmware update process.

[image]

HripsimeS, 2022-09-22:
@MartyG-RealSense Hello. Thanks a lot, I have done the **update of L515's firmware** driver with a firmware 1.5.8.1. I also checked 99-realsense-libusb.rules file is really located in udev with command **ls /etc/udev/rules.d/99-realsense-libusb.rules**

I would like to give the details of my installation steps on Ubuntu 20.04.5 LTS
1) **Install  ROS Noetic** http://wiki.ros.org/noetic/Installation/Ubuntu
2) **LibUVC-backend installation** https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md
3) **Connect USB device** https://learn.microsoft.com/en-us/windows/wsl/connect-usb

**lsusb** command is detecting Intel Corp. Intel(R) RealSense(TM) 515, then running **rs-enumerate-devices**command
**ERROR [140626576078592] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
ERROR [140626609676480] (sensor.cpp:576) acquire_power failed: failed to set power state
Could not create device - failed to set power state . Check SDK logs for details**

I have tried also unplugging the micro-sized end of the USB cable from the base of the camera, reversing the connector's orientation and plugging it back into the L515 camera and the same error after **rs-enumerate-devices**command.

Let me know if I missed some step in my installations or I can modify something to make the  **rs-enumerate-devices**command finially find the camera without any errors. Look forward to hearing from you!

MartyG-RealSense, 2022-09-22:
The link below has a case with similar circumstances to yours (lsusb worked but rs-enumerate-devices didn't), except that they were not using WSL2 and the cameras used were D435 and D405 instead of L515.  This would suggest that it may not be an issue related to use of WSL2.

https://support.intelrealsense.com/hc/en-us/community/posts/5172990520339/comments/5199160685715

Though there was not a definitive solution achieved at the end of that particular case, I did suggest trying a mains-powered USB hub to confirm whether the problem was due to insufficient power being supplied to the camera by the computer when a program tries to query the camera's details (different from using lsusb, which just queries details of a USB bus and the device attached to it).

HripsimeS, 2022-09-25:
@MartyG-RealSense Can you please tell me what is the intrinsic camera parameters for L515 camera?
It should be something like this but with different numbers  
np.array([[572.4114, 0., 325.2611], [0., 573.57043, 242.04899], [0., 0., 1.]])

Look forward to hearing from you and thanks in advance!

MartyG-RealSense, 2022-09-25:
Each individual camera will have unique intrinsics and extrinsics.  I have provided below the intrinsics from my own L515 as a rough guide though.

[image]

[image]

[image]

[image]

[image]

MartyG-RealSense, 2022-10-01:
Hi @HripsimeS  Do you require further assistance with this case, please?  Thanks!

closed 2022-10-02 by HripsimeS

sym-lening, 2023-09-30:
I did the following
[image]

[image]

[image]

leelening, 2023-09-30:
I did the following:
[image]

[image]

[image]

It seems that I have to build from the source, right?

> If it is possible to use librealsense on WSL2 now that it has USB support then I would think that the best chance of it working would be to build it from source code with the **RSUSB backend**. This method of installation bypasses the kernel and so is not dependent on Linux versions or kernel versions and does not require kernel patching.
> 
> An RSUSB backend installation is helpful in enabling librealsense to work in situations where an exotic hardware or software configuration that differs from the average Linux PC is being used.
> 
> You could first try the simple **libuvc backend** SDK build script at the link below.
> 
> https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md
> 
> If the simple libuvc backend build script does not work, there is a more complex procedure for performing an RSUSB build from source code with CMake at [IntelRealSense/librealsense#9931 (comment)](https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692)
> 
> In regard to kernel 5.15, an unofficial test patch to add support for kernels 5.13 and 5.15 is available at [IntelRealSense/librealsense#10439 (comment)](https://github.com/IntelRealSense/librealsense/issues/10439#issuecomment-1171965106)
> 
> I could not find a method to select your own kernel in WSL2 in order to downgrade, only how to update to the latest one.

MartyG-RealSense, 2023-09-30:
Hi @leelening  If you are using WSL with RealSense then it has to be **WSL2** so that it is able to access USB devices such as a RealSense camera.

https://www.omgubuntu.co.uk/how-to-install-wsl2-on-windows-10

Getting RealSense working with WSL2 can be so difficult though (involving installing various missing components manually) that it is far easier and less time consuming just to use a dedicated Ubuntu computer instead of a Windows one with WSL.

leelening, 2023-09-30:
> Hi @leelening If you are using WSL with RealSense then it has to be **WSL2** so that it is able to access USB devices such as a RealSense camera. [...]

[image]
It seems that I am already using WSL 2

MartyG-RealSense, 2023-10-01:
@leelening   I see that you are using Windows 10.  There was a WSL2 user at https://github.com/IntelRealSense/librealsense/issues/10506#issuecomment-1125860744 who was able to get RealSense working with WSL2 on Windows 11 by "completely unplugging and replugging the camera and also restarting WSL" but was not able to get it to work on Windows 10.

MartyG-RealSense, 2023-10-01:
That's great to hear, @leelening  :)