realsenseai/realsense-ros #2485: No pointcloud and no color image when activating PCL

vivek2019, 2022-09-20:
Hello 

I am using Realsense D455, with Nano (Jetpack 4.6)

I am facing issues when testing PCL with the above device. I am using the rs_camera.launch file. When enabling PCL as true, There is NO PCL and NO Color Image and Drop in Depth images from 30 to 15-12hz

I have tested the above above files with below configurations

- Librealsense version  ----     Realsense ros             -----                  fw version

- 2.51             ----              2.3.1/2.3.2/development     ----         [5.13.0.50] /  5.12.14.50

- 2.50       ----                     2.3.1/2.3.2/development   ----          [5.13.0.50] /  5.12.14.50

- 2.48        ----                    2.3.1/2.3.2                       ----              [5.13.0.50] / 5.12.14.50

- 2.45            ----                2.3.0                        ----                      5.12.13.50

**ROS output**

... logging to /home/uvrobot/.ros/log/da6c2594-38d2-11ed-aa09-00044be67567/roslaunch-uvrobot-desktop-11154.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.0.10:43049/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
[... 94 lines cut ...]
[ WARN] [1663671382.323709773]: Param '/camera/stereo_module/auto_exposure_limit' has value 0 that is not in range [1, 200000]. Removing this parameter from dynamic reconfigure options.
[ WARN] [1663671382.328863579]: Param '/camera/stereo_module/auto_gain_limit' has value 0 that is not in range [16, 248]. Removing this parameter from dynamic reconfigure options.
[ WARN] [1663671382.394532286]: Param '/camera/rgb_camera/power_line_frequency' has value 3 that is not in the enum { {50Hz: 1} {60Hz: 2} {Disabled: 0} }. Removing this parameter from dynamic reconfigure options.
[ INFO] [1663671382.450775162]: Done Setting Dynamic reconfig parameters.
[ INFO] [1663671382.452329180]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1663671382.454025551]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1663671382.454155508]: setupPublishers...
[ INFO] [1663671382.463925373]: Expected frequency for depth = 30.00000
[ INFO] [1663671382.576244865]: Expected frequency for color = 30.00000
[ INFO] [1663671382.670860225]: setupStreams...
[ INFO] [1663671382.673420512]: insert Depth to Stereo Module
[ INFO] [1663671382.673576200]: insert Color to RGB Camera
[ INFO] [1663671382.870025510]: SELECTED BASE:Depth, 0
[ INFO] [1663671382.916736038]: RealSense Node Is Up!
[ WARN] [1663671382.952390255]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
 20/09 16:26:22,950 WARNING [546823872896] (ds5-timestamp.cpp:69) UVC metadata payloads not available. Please refer to the installation chapter for details.
 20/09 16:26:23,269 WARNING [546834710912] (ds5-timestamp.cpp:69) UVC metadata payloads not available. Please refer to the installation chapter for details.
^C[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete

[image]

MartyG-RealSense, 2022-09-20:
Hi @vivek2019  I see from your kindly provided image that you are using a Jetson Nano.  

I also note the warning **UVC metadata payloads not available. Please refer to the installation chapter for details** in your log.  This would indicate that your librealsense SDK installation has not had its kernel patched.

May I ask which methods you used to install the librealsense SDK and the RealSense ROS wrapper on your Nano, please?  For example, if your librealsense installation was built from source code then Jetson devices have their own dedicated L4T kernel patching script called **patch-realsense-ubuntu-L4T.sh** and so patching the kernel using the 'patch-realsense-ubuntu-lts.sh' script would not be suitable.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

vivek2019, 2022-09-21:
Hello @MartyG-RealSense I used the 4. Install with Debian Packages to install the SDK. 

what i understood is that i have to execute the patch-realsense-ubuntu-L4T.sh file to patch the kernel and then the system might work as expected ?

MartyG-RealSense, 2022-09-21:
If you installed from packages using the Jetson instructions then the patch should be included in the packages.  This makes it unusual that the log reports that hardware metadata is not available.

After you installed from Jetson packages, which method did you use to install the ROS wrapper please?

vivek2019, 2022-09-21:
I have just git clone the ROS wrapper

vivek2019, 2022-09-21:
Yes, I removed all the files before switching to the new sdk version. Also i updated the relasense-ros wrapper compatible version and build using catkin_make. To check the compatibility i used the release documents. 

I am testing the PCL with the same rs_camera.launch file with updating the parameters below
enable_pointcloud:=true
filters:=pointcloud
enable_sync:=true

vivek2019, 2022-09-21:
I tried the complete setup installation from the start. 

1. Removed all the files using the below command

dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge

2. followed step 4 from the link to install sdk - 

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

3. Installed the ros wrapper using the below command

sudo apt-get install ros-melodic-realsense2-camera

During the 2nd step the files installed the sdk version 2.51 and when executed step 3. it installed the files with version 2.50.

What should I update in the step2 to install SDK version 2.50, As according to the release document SDK 2.50 is the last compatible version with ROS1 Melodic

MartyG-RealSense, 2022-09-21:
In step 3, if you install with apt-get install (known as 'Method 1' in the ROS wrapper instructions) then it is installing librealsense a second time, as this command installs librealsense and the ROS wrapper together at the same time from packages.  So as well as creating a potential conflict between the first Jetson package installation of librealsense and the second one that the apt-get command will install, the apt-get version of librealsense also does not contain support for CUDA.

vivek2019, 2022-09-21:
Okay will test building the wrapper from source after the librealsense

vivek2019, 2022-09-21:
Hello  @MartyG-RealSense I tested the above configuration and the it seems it worked. But it is using sdk 2.51 in ROS 1 melodic stack and the output is also slow.

Below is the test LOG

roslaunch realsense2_camera rs_camera.launch enable_pointcloud:=true... logging to /home/uvrobot/.ros/log/76ec029a-398c-11ed-a835-00044be67567/roslaunch-uvrobot-desktop-31296.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.0.8:41311/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: True
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
[... 79 lines cut ...]
[ INFO] [1663751101.325358310]: Setting Dynamic reconfig parameters.
[ WARN] [1663751101.497531933]: Param '/camera/rgb_camera/power_line_frequency' has value 3 that is not in the enum { {50Hz: 1} {60Hz: 2} {Disabled: 0} }. Removing this parameter from dynamic reconfigure options.
[ INFO] [1663751101.559918979]: Done Setting Dynamic reconfig parameters.
[ INFO] [1663751101.561545525]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1663751101.564788930]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1663751101.564927631]: setupPublishers...
[ INFO] [1663751101.574028716]: Expected frequency for depth = 30.00000
[ INFO] [1663751101.686874526]: Expected frequency for color = 30.00000
[ INFO] [1663751101.782292506]: setupStreams...
[ INFO] [1663751101.920285951]: SELECTED BASE:Depth, 0
[ INFO] [1663751101.965916328]: RealSense Node Is Up!
 21/09 14:35:02,015 WARNING [546850083200] (ds5-timestamp.cpp:69) UVC metadata payloads not available. Please refer to the installation chapter for details.
[ WARN] [1663751102.017399437]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
 21/09 14:35:02,335 WARNING [546861736320] (ds5-timestamp.cpp:69) UVC metadata payloads not available. Please refer to the installation chapter for details.
[ INFO] [1663751253.778391735]: publishing unordered pointcloud.
[ WARN] [1663751253.951885023]: No stream match for pointcloud chosen texture Process - Color
[ WARN] [1663751254.480654376]: No stream match for pointcloud chosen texture Process - Color

Output

The warning No stream match for pointcloud chosen texture Process - Color is continuous. 
Tested it for 30mins
Color image = 10hz
depth image = 30hz
pcl = 6hz

vivek2019, 2022-09-21:
I am planning to test the below configuration

Installing the librealsense2 sdk 2.50 from source and then install the ros wrapper from source compatible with the sdk version and test the configuration

vivek2019, 2022-09-21:
The warning is not present when PCL is off

[ INFO] [1663753807.875800013]: RealSense Node Is Up!
[ WARN] [1663753807.923580954]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
 21/09 15:20:07,922 WARNING [547060617600] (ds5-timestamp.cpp:69) UVC metadata payloads not available. Please refer to the installation chapter for details.
 21/09 15:20:08,242 WARNING [547071455616] (ds5-timestamp.cpp:69) UVC metadata payloads not available. Please refer to the installation chapter for details.
 21/09 15:20:13,887 ERROR [547052224896] (ds5-options.cpp:88) Asic Temperature value is not valid!
 21/09 15:20:14,890 ERROR [547052224896] (ds5-options.cpp:88) Asic Temperature value is not valid!
 21/09 15:20:15,829 ERROR [547197280640] (ds5-options.cpp:88) Asic Temperature value is not valid!
 21/09 15:20:15,893 ERROR [547052224896] (ds5-options.cpp:88) Asic Temperature value is not valid!

MartyG-RealSense, 2022-09-21:
I am unfamiliar with PCL in ROS, unfortunately.  Are you using **pcl_ros** please?

If the pointcloud is being generated by the ROS wrapper's pointcloud filter, what function is PCL performing?

vivek2019, 2022-09-21:
No I am not using pcl_ros.

I am using the ros wrapper to generate the PCL and will be using it for navigation of the device

MartyG-RealSense, 2022-09-21:
The developer of the ROS1 wrapper stated at https://github.com/IntelRealSense/realsense-ros/issues/588#issuecomment-457890773 that the **No stream match for pointcloud chosen texture Process - Color** warning can occur if there is a frame drop.

As you are using a D455 equipped with an IMU, would it be practical for your project to use Intel's SLAM navigation guide for ROS at the link below for navigation instead of PCL?

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i

closed 2022-10-03 by MartyG-RealSense