realsenseai/realsense-ros #2486: D405 fps

DaphnaNanovel, 2022-09-20:
Hello,
we switched from D435 to D405 and faced very strange issue - we can't run D405 faster than 15 FPS.
We tested in both on Jetson NX and on PC.
We installed the ros2_beta and we run the 
`ros2 launch realsense2_camera rs_launch.py`
When we run this on D435 we get 30 FPS on topic /camera/color/image_raw
The same command on D405 provides only 15 FPS on topic /camera/color/image_rect_raw
And any tries to run the command with rgb_camera.profile:=640,480,60 (or 30) also gives only 15FPS.
Moreover, camera reports 30/60 fps by ros logger.

We measure the FPS with **ros2 topic hz** command
Any suggestions?

MartyG-RealSense, 2022-09-20:
Hi @DaphnaNanovel   Do you experience the FPS drop on D405 if you launch with **ros2 run** instead of ros2 launch, please?

`ros2 run realsense2_camera realsense2_camera_node`

DaphnaNanovel, 2022-09-21:
Hi @MartyG-RealSense, 
When we run with ros run, we do get higher fps (~20 on Jetson NX, and ~30 on PC), but we couldn't control the height and width of the frames (both color and depth frames). we tried with ros2 param set and with --ros_args. The ros2 param list shows no option for rgb_camera.profile.We were able to set depth_module.profile with our values, but it does not affect the dimension of the depth image.

MartyG-RealSense, 2022-09-21:
Thank you very much for the feedback.  What happens with ros2 launch if you use **90** FPS for the 640x480 rgb_camera profile and 30 for the depth?

DaphnaNanovel, 2022-09-21:
[image]

MartyG-RealSense, 2022-09-21:
The log says that the color is using 30 FPS instead of 90 FPS.

DaphnaNanovel, 2022-09-21:
yes, but you can see the command above: 
`ros2 launch realsense2_camera rs_launch.py enable_sync:=true rgb_camera.profile:=640,480,90 depth_module.profile:=640,480,30`

DaphnaNanovel, 2022-09-21:
same ...
[image]

MartyG-RealSense, 2022-09-21:
Okay, thanks.  So enable_sync being disabled does not make a difference when using ros2 launch.  And you get good performance with ros2 run but have the problem of how to set a custom stream configuration.

DaphnaNanovel, 2022-09-21:
Correct

MartyG-RealSense, 2022-09-21:
Are you able to configure the streams with ros2 run using the instruction below?

**ros2 run realsense2_camera realsense2_camera_node --ros-args -p depth_width:=640 -p depth_height:=480 -p depth_fps:=30.0 -p color_width:=640 -p color_height:=480 -p color_fps:=30.0**

DaphnaNanovel, 2022-09-21:
The dimension are 848x480, fps is ~2 

[image]

MartyG-RealSense, 2022-09-21:
Okay, that likely means that the custom configuration was deemed invalid by the launch and the camera's default stream configuration was applied instead.

MartyG-RealSense, 2022-09-21:
Going back to **ros2 launch**, let's next try setting the pointcloud texture to '0' instead of the default '2'.

**ros2 launch realsense2_camera rs_launch.py rgb_camera.profile:=640,480,30 depth_module.profile:=640,480,30 pointcloud.stream_filter:=2**

DaphnaNanovel, 2022-09-21:
[image]

MartyG-RealSense, 2022-09-21:
You are not using a pointcloud anyway so that setting would not make a difference.  Apologies for that!

MartyG-RealSense, 2022-09-21:
Given that the ros2_beta wrapper documentation suggests expressing the profile with 'x' between the stream settings instead of commas, can we try it with x's please just to eliminate it as a cause of the problem (as the launch works normally with D435, it probably is not related).

ros2 launch realsense2_camera rs_launch.py rgb_camera.profile:=640x480x30 depth_module.profile:=640x480x30

DaphnaNanovel, 2022-09-21:
Yes, I saw it in the readme and tried it. As you assumed, that's not the problem...
Is there any incompatibility between these versions: RealSense ROS v4.0.4 and Built with LibRealSense v2.51.1?

MartyG-RealSense, 2022-09-21:
The ros2_beta wrapper was updated to 4.51.1 for matching with 2.51.1 about a week ago.   Wrapper 4.0.4 was stable with 2.51.1 but it was designed for use with 2.50.0.

https://github.com/IntelRealSense/realsense-ros/releases/tag/4.51.1

MartyG-RealSense, 2022-09-27:
Hi @DaphnaNanovel  Do you require further assistance with this case, please?  Thanks!

DaphnaNanovel, 2022-09-28:
Hi @MartyG-RealSense, yes we do.. 
same problem.. 
Thank you!

MartyG-RealSense, 2022-09-28:
Were you able to test with librealsense 2.51.1 and wrapper 4.51.1 and found that the same problem was occurring?

DaphnaNanovel, 2022-09-29:
Yes.. It says it is running with 30 FPS but actually when we check with ros2 topic hz we get at max 15 FPS (tested on a PC and not on Jetson)

MartyG-RealSense, 2022-09-29:
In tests with the D405 I observed that the FPS could halve from 30 to 15 if auto-exposure became disabled.  Please test launching with auto-exposure enabled via the launch instruction to see whether the FPS becomes 30.

**ros2 launch realsense2_camera rs_launch.py rgb_camera.profile:=640x480x30 depth_module.profile:=640x480x30 enable_auto_exposure:=true**

MartyG-RealSense, 2022-09-29:
**Device USB Type** is reported as '2.1' in the log, indicating that it is detecting the camera as being on a slower USB 2.1 connection instead of a full-speed USB 3 connection.

If the D405 is plugged into a USB 3 port, please add **initial_reset:=true** to reset the camera at launch to see whether this corrects the detected USB connection to 3.1 or 3.2.

**ros2 launch realsense2_camera rs_launch.py rgb_camera.profile:=640x480x30 depth_module.profile:=640x480x30 enable_auto_exposure:=true initial_reset:=true**

DaphnaNanovel, 2022-09-29:
I'm now trying your suggestions, and I see that it says we are connected to USB 2.1. But if you scroll up to the previous messages you will see that it was detected with USB 3 when I used another Jetson or my PC

MartyG-RealSense, 2022-10-05:
Hi @DaphnaNanovel  My sincere apologies for the delay in responding further.

The log image reports **UVC metadata payloads not available**.  This suggests that a kernel patch might not have been applied.  What methods do you use to install librealsense and the ROS wrapper on Jetson and on PC, please?

DaphnaNanovel, 2022-10-12:
Hi @MartyG-RealSense, unfortunately the installation procedure with the **installation_jetson.md** didn't solve the problem. We are getting the same fps problem. Also on a computer with ros foxy.

[image]

MartyG-RealSense, 2022-10-12:
When using the D435 you were using ros2 topic hz with the **/camera/color/image_raw** topic, whilst on D405 you checked the hz of /camera/color/image_rect_raw

Is /camera/color/image_raw published when launching the ROS wrapper with D405, and if it is then what is its hz rate?

DaphnaNanovel, 2022-10-13:
No,   /camera/color/image_raw topic doesn't exist at all. 
[image]

DaphnaNanovel, 2022-10-18:
Hello @MartyG-RealSense ,
we added cyclone rmw to .bashrc and saw 15FPS on both Jetson and PC.
We can "live" with it, but if you manage to find the solution we'll appriciate it.
Thank you for your support!

MartyG-RealSense, 2022-10-18:
You are very welcome, @DaphnaNanovel - thanks very much for the update!

dheera, 2022-10-20:
I'm also having _major_ issues with FPS with D455, 20.04, foxy, using either *ros2 run* or *ros2 launch*. Has anyone been able to get RS working in ROS2 at all?

MartyG-RealSense, 2022-10-21:
Hi @dheera Please create a new case for your problem by visiting the link below and clicking the green New Issue button. Thanks!

https://github.com/IntelRealSense/realsense-ros/issues

SamerKhshiboun, 2023-01-31:
See my comment in #2507 for more information.

lifeisgood-ai, 2023-02-28:
Hi @MartyG-RealSense 

Could you provide information on the progress of the bug report you [mentionned](https://github.com/IntelRealSense/realsense-ros/issues/2486#issuecomment-1281139873) about setting the FPS rate on D405 ?

I am also facing the issue. `ros2 launch` displays that the setting of my cam is set to 30 fps but in reaity it is blocked to 15 fps and no way to bypass. This is a problem to me as I use the D405 in multicam architecture where I get images at 30 fps and I need to get them synced for processing.

Thank you

MartyG-RealSense, 2023-02-28:
Hi @lifeisgood-ai  This bug report is currently being actively worked on by the RealSense ROS developer team at the time of writing this.

SamerKhshiboun, 2023-03-19:
Hi @lifeisgood-ai , @DaphnaNanovel, @dheera 

My name is Samer and I'm a developer in the RealSense SW team.
I'm investigating issues related to low FPSs in several scenarios, and my suggestion is to make sure that the auto_exposure is on and that the HDR filter is off.
Please read my comment here, and let me know if that helps.
https://github.com/IntelRealSense/realsense-ros/issues/2507#issuecomment-1411214372

You might have to add "depth_module.enable_auto_exposure" to the launch file (which is missing from the launch file currently, and might cause not loading auto_exposure parameter correctly) with default value "True", like this:
`                           {'name': 'depth_module.auto_exposure,      'default': 'true', 'description': '...'},
`
and also, you might want to turn off the depth_module.hdr_enabled and the hdr_merge.enable filters in the launch file.

Let me know if the above suggestions helped you.

Thanks,
Samer

MartyG-RealSense, 2023-03-23:
Hi everyone, is there anybody on this case who requires further assistance after the fix implemented by @SamerKhshiboun at https://github.com/IntelRealSense/realsense-ros/pull/2612 and merged into the **ros2_deveopment** ROS2 wrapper?

lifeisgood-ai, 2023-03-25:
Hi Samer @SamerKhshiboun  

Thank you for taking the issue.

I have been through your comment [#2507 (comment)](https://github.com/IntelRealSense/realsense-ros/issues/2507#issuecomment-1411214372) and applied what you suggest in your [#2486 (answer)](https://github.com/IntelRealSense/realsense-ros/issues/2486#issuecomment-1475272809).

I made a quick acquisition of RGB images to a ros bag using rs_launch.py adapting the `configurable_parameters`.

Simply adding the following line is not enough :

> {'name': 'depth_module.enable_auto_exposure, 'default': 'true', 'description': '...'}, 

Commenting the two lines helps : 

> #{'name': 'depth_module.exposure.1',     'default': '7500', 'description': 'Initial value for hdr_merge filter'},
> #{'name': 'depth_module.exposure.2',     'default': '1', 'description': 'Initial value for hdr_merge filter'},

Running rs_launch,

> ros2 launch realsense2_camera rs_launch.py

 I get 30 fps ! :+1: 

> $ ros2 topic hz /camera/color/image_rect_raw
> average rate: 29.555
>	min: 0.025s max: 0.067s std dev: 0.00636s window: 31

I also get 30 fps using the ros command line 

> $ ros2 run realsense2_camera realsense2_camera_node

This is good for me. I will test with multi cameras later this week if I get time.

Another point that might not be related is a warning of version differences. I compiled librealsense v2.53.1 as I run a Ubuntu 22.04 and my realsense-ros is built over librealsense 2.51.1
> [realsense2_camera_node-1] [INFO] [1679744057.700592306] [camera.camera]: RealSense ROS v4.51.1
> [realsense2_camera_node-1] [INFO] [1679744057.701034637] [camera.camera]: Built with LibRealSense v2.51.1
> [realsense2_camera_node-1] [INFO] [1679744057.701147666] [camera.camera]: Running with LibRealSense v2.53.1
> [realsense2_camera_node-1] [WARN] [1679744057.701233120] [camera.camera]: ***************************************************
> [realsense2_camera_node-1] [WARN] [1679744057.701314466] [camera.camera]: ** running with a different librealsense version **
> [realsense2_camera_node-1] [WARN] [1679744057.701384099] [camera.camera]: ** than the one the wrapper was compiled with!   **
> [realsense2_camera_node-1] [WARN] [1679744057.701460896] [camera.camera]: ***************************************************

Thanks 
LG

lifeisgood-ai, 2023-03-25:
hi @MartyG-RealSense @SamerKhshiboun 
I tested adding a `depth_module.exposure` entry with a value of 7500 with `depth_module.auto_exposure` set to false and it works fine now.
still in my version of rs_launch.py I have 2 entries `depth_module.exposure.1` and `depth_module.exposure.2` by default.
Apart from that, all good for me.

SamerKhshiboun, 2023-03-25:
Hi @lifeisgood-ai,
I'm glad you are able to get 30 FPS now.
I suggest you see my last merged [PR commit](https://github.com/IntelRealSense/realsense-ros/pull/2612/commits/b57eaeea668e276a75c6eeb5cccbb2d9807fecc0
) as I added another parameter named `depth_module.exposure` which we want to override through the rs_launch.py in order to make the 30 FPS works good, specially for D405 model.
see also the [description](https://github.com/IntelRealSense/realsense-ros/pull/2612) in the PR itself to understand more about the FPS decrease reason.

Regarding your librealsense version issue, I guess you might have multiple librealsense versions installed in your system, and our wrapper is finding the first installed version first, according to its search path.
try running `dpkg --list | grep realsense` to see if you have multiple versions installed.

Thanks,
Samer

ohadNanovel, 2023-03-26:
Hi @SamerKhshiboun , @MartyG-RealSense , 
I have just tested your new PR and solution and it worked for me both on my PC and Jetson. Thanks a lot for the investigation and the solution.

DaphnaNanovel, 2023-03-26:
> Hi @SamerKhshiboun , @MartyG-RealSense , I have just tested your new PR and solution and it worked for me both on my PC and Jetson. Thanks a lot for the investigation and the solution.

Hi, I wrote this comment from another account by mistake. Thanks again!

closed 2023-03-26 by DaphnaNanovel

MartyG-RealSense, 2023-03-26:
That's great news that it worked for you, @DaphnaNanovel - thanks very much for the update!

yuqiliang1, 2025-12-26:
thx for your guys' efforts! that helps me a lot