realsenseai/realsense-ros #2488: Pointcloud is almost blank outdoor when using ROS2

tsengapola, 2022-09-23:
We were testing realsense455 outdoor using ROS2 humble/galactic at different platforms including X86/nvidia Xavier.
We found that the realsense455 works fine indoor but detect few points outdoor, here are screenshots.
From the following screenshot, we can see floor and points from pointcloud:
[image]
However, we can not see anything when the door is opened. You can notice that the floor disappeared:
[image]
Meanwhile ROS1 work fines:
[image]

Details of the environment we used:
---
ROS2 galactic
RealSense ROS v4.0.4
Built with LibRealSense v2.50.0
Running with LibRealSense v2.50.0

ROS1 noetic
RealSense ROS v2.3.2
Built with LibRealSense v2.50.0
Running with LibRealSense v2.50.0

## launch file we used in ROS2 galactic:
```
# Copyright (c) 2018 Intel Corporation
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.

"""Launch realsense2_camera node."""
import os
from launch import LaunchDescription
from ament_index_python.packages import get_package_share_directory
import launch_ros.actions
from launch.actions import DeclareLaunchArgument
from launch.substitutions import LaunchConfiguration, PythonExpression
from launch.conditions import IfCondition

configurable_parameters = [{'name': 'camera_name',                  'default': 'camera', 'description': 'camera unique name'},
                           {'name': 'serial_no',                    'default': "''", 'description': 'choose device by serial number'},
                           {'name': 'usb_port_id',                  'default': "''", 'description': 'choose device by usb port id'},
                           {'name': 'device_type',                  'default': "''", 'description': 'choose device by type'},
                           {'name': 'config_file',                  'default': "''", 'description': 'yaml config file'},
                           {'name': 'unite_imu_method',             'default': "0", 'description': '[0-None, 1-copy, 2-linear_interpolation]'},
                           {'name': 'json_file_path',               'default': "''", 'description': 'allows advanced configuration'},
                           {'name': 'log_level',                    'default': 'info', 'description': 'debug log level [DEBUG|INFO|WARN|ERROR|FATAL]'},
                           {'name': 'output',                       'default': 'screen', 'description': 'pipe node output [screen|log]'},
                           {'name': 'depth_module.profile',         'default': '424,240,5', 'description': 'depth module profile'},                           
                           {'name': 'enable_depth',                 'default': 'true', 'description': 'enable depth stream'},
                           {'name': 'rgb_camera.profile',           'default': '0,0,0', 'description': 'color image width'},
                           {'name': 'enable_color',                 'default': 'true', 'description': 'enable color stream'},
                           {'name': 'enable_infra1',                'default': 'false', 'description': 'enable infra1 stream'},
                           {'name': 'enable_infra2',                'default': 'false', 'description': 'enable infra2 stream'},
                           {'name': 'infra_rgb',                    'default': 'false', 'description': 'enable infra2 stream'},
                           {'name': 'tracking_module.profile',      'default': '0,0,0', 'description': 'fisheye width'},
                           {'name': 'enable_fisheye1',              'default': 'true', 'description': 'enable fisheye1 stream'},
                           {'name': 'enable_fisheye2',              'default': 'true', 'description': 'enable fisheye2 stream'},
                           {'name': 'enable_confidence',            'default': 'true', 'description': 'enable depth stream'},
                           {'name': 'gyro_fps',                     'default': '0', 'description': "''"},                           
                           {'name': 'accel_fps',                    'default': '0', 'description': "''"},                           
                           {'name': 'enable_gyro',                  'default': 'false', 'description': "''"},                           
                           {'name': 'enable_accel',                 'default': 'false', 'description': "''"},                           
[... 68 lines cut ...]
                package='realsense2_camera',
                namespace=LaunchConfiguration("camera_name"),
                name=LaunchConfiguration("camera_name"),
                executable='realsense2_camera_node',
                parameters=[set_configurable_parameters(configurable_parameters)
                            ],
                output='screen',
                arguments=['--ros-args', '--log-level', LaunchConfiguration('log_level')],
                emulate_tty=True,
                ),
            launch_ros.actions.Node(
                condition=IfCondition(PythonExpression([LaunchConfiguration('config_file'), " != ''"])),
                package='realsense2_camera',
                namespace=LaunchConfiguration("camera_name"),
                name=LaunchConfiguration("camera_name"),
                executable='realsense2_camera_node',
                parameters=[set_configurable_parameters(configurable_parameters)
                            , PythonExpression([LaunchConfiguration("config_file")])
                            ],
                output='screen',
                arguments=['--ros-args', '--log-level', LaunchConfiguration('log_level')],
                emulate_tty=True,
                ),
        ])
```

MartyG-RealSense, 2022-09-23:
Hi @tsengapola  It appears that your launch file is essentially the same as **rs_launch.py** except that a depth_module.profile of 424,240,5 is set in your launch file instead of the default 0,0,0 of rs_launch.py.

Let's first try launching with rs_launch.py and setting the depth profile and the pointcloud in the launch instruction to see whether it makes a difference.

**ros2 launch realsense2_camera rs_launch.py depth_module.profile:=424x240x5 pointcloud.enable:=true**

tsengapola, 2022-09-23:
Thank you for the prompt reply,
I ran the testing using:
```
ros2 launch realsense2_camera rs_launch.py depth_module.profile:=424x240x5 pointcloud.enable:=true
```
The result remains the same. I think somehow sunlight has effect in the detection. I attach the video of the pointcloud when opening the door.
You can see that the ground disappeared when door opened.

https://user-images.githubusercontent.com/62891637/191933400-2364bd59-0c89-45b2-9bb2-76a8bd3ffdb9.mp4

MartyG-RealSense, 2022-09-23:
The reflectivity of the floor seems to be greatly increasing when the door is opened.  Reflections from surfaces can make those surfaces more difficult for the camera to analyze for depth detail.  As the door rises, the depth detail at floor level can be seen progressively disappearing as the door opens further and more light is cast onto the floor.

As you mention however, the image is not negatively affected by the reflectivity in ROS1, only in the ROS2 wrapper.

Does the effect still occur if you increase the resolution or FPS speed of the depth profile?  For example, 640x480 and 30 FPS.

MartyG-RealSense, 2022-09-29:
Hi @tsengapola  Do you require further assistance with this case, please?  Thanks!

tsengapola, 2022-09-29:
Hi @MartyG-RealSense, just found out the issue.
When we use default launch file on RS 455 in ROS2 say humble/galactic
```
ros2 launch realsense2_camera rs_launch.py depth_module.profile:=424x240x5 pointcloud.enable:=true
```
The enable_auto_exposure in depth module is off when point cloud is enabled (although it is on when using RS 435), which cause the poor outdoor detection.
We try to put following param in launch file, but it does not take effect (although rqt_reconfigure shows it is ticked):
```
{'name': 'depth_module.enable_auto_exposure',         'default': 'true', 'description': 'auto exposure'}
```
The only way it works now from our test is to use ros2 param set to trigger it:
```
ros2 param set <your camera name> depth_module.enable_auto_exposure True
```
The following video shall answer the issue:
 

https://user-images.githubusercontent.com/62891637/193026752-8a1d4cd0-0167-4929-958b-bb49ddab5cdb.mp4

MartyG-RealSense, 2022-09-29:
And auto_exposure_enable does not work if enabled from the launch file, please?

**ros2 launch realsense2_camera rs_launch.py depth_module.profile:=424x240x5 enable_auto_exposure:=true pointcloud.enable:=true**

tsengapola, 2022-09-29:
Unfortunately, seems not working using the command:
```
ros2 launch realsense2_camera rs_launch.py depth_module.profile:=424x240x5 pointcloud.enable:=true enable_auto_exposure:=true
```

https://user-images.githubusercontent.com/62891637/193042084-d0657bb4-2590-48cf-93c3-592e5a5bfbde.mp4

MartyG-RealSense, 2022-09-29:
How does the auto-exposure behave if you launch with **ros2 run** instead of ros2 launch?

ros2 run realsense2_camera realsense2_camera_node --ros-args -p pointcloud.enable:=true

tsengapola, 2022-09-30:
Nothing works by:
```
ros2 run realsense2_camera realsense2_camera_node --ros-args -p pointcloud.enable:=true
```
However, I think it is ok, we can make it work using ros2 param set and this is all we want.
So I will close this issue.

closed 2022-09-30 by tsengapola