realsenseai/realsense-ros #2489: control_transfer returned error, while roslaunch and realsense-viewer

Hubert-1102, 2022-09-26:
hello, I try to run `roslaunch realsense2_camera rs_camera.launch` , there are a lot of warnings,  there are also some `control_transfer returned error ` while running `realsense-viewer`, but  realsense-viewer software seems fine, I can see RGB and depth image with it.
also, running `modinfo uvcvideo | grep "version:"` has no output  with `realsense`
### something about my installation
I install realsense-ros with method 1
by running `uname -a`
Linux Hubert 5.15.0-48-generic #54~20.04.1-Ubuntu SMP Thu Sep 1 16:17:26 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

this is the output 

> roslaunch realsense2_camera rs_camera.launch
... logging to /home/hubert/.ros/log/18b6a154-3d87-11ed-b3f9-3d6204480823/roslaunch-Hubert-5588.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Hubert:46183/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
[... 85 lines cut ...]
[ INFO] [1664188600.892012448]: Done Setting Dynamic reconfig parameters.
[ INFO] [1664188600.892570961]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1664188600.893546769]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1664188600.893585200]: setupPublishers...
[ INFO] [1664188600.896600800]: Expected frequency for depth = 30.00000
[ INFO] [1664188600.940340134]: Expected frequency for color = 30.00000
[ INFO] [1664188600.971061653]: setupStreams...
 26/09 18:36:41,291 WARNING [140507470767872] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 26/09 18:36:41,341 WARNING [140507470767872] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 26/09 18:36:41,392 WARNING [140507470767872] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 26/09 18:36:41,442 WARNING [140507470767872] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1664188601.493428691]: SELECTED BASE:Depth, 0
 26/09 18:36:41,493 WARNING [140507470767872] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1664188601.512668876]: RealSense Node Is Up!
[ WARN] [1664188601.648332275]: 
 26/09 18:36:41,649 WARNING [140507470767872] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 26/09 18:36:41,700 WARNING [140507470767872] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 26/09 18:36:41,752 WARNING [140507470767872] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
^C[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

MartyG-RealSense, 2022-09-26:
Hi @Hubert-1102  Before you installed the wrapper with Method 1, had you already installed the librealsense SDK?  Method 1 installs both librealsense and the wrapper from packages at the same time, so if you had first installed librealsense and then installed it a second time with Method 1 then there would be the potential for conflicts between the two installations.

Does the stability of the launch improve if you add **initial_reset:=true** to your roslaunch instruction to reset the camera at launch, please?

**roslaunch realsense2_camera rs_camera.launch initial_reset:=true**

Hubert-1102, 2022-09-26:
Yes，as you said, I have installed the librealsense SDK before method 1
should I uninstall all the packages and install again?
running `roslaunch realsense2_camera rs_camera.launch initial_reset:=true` , it also has many warnings

> roslaunch realsense2_camera rs_camera.launch initial_reset:=true
... logging to /home/hubert/.ros/log/d4ce7af8-3d8f-11ed-80b3-09ffcf34f0ec/roslaunch-Hubert-3625.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Hubert:36551/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
[... 98 lines cut ...]
 26/09 19:39:19,210 WARNING [140078730958592] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 26/09 19:39:19,261 WARNING [140078730958592] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 26/09 19:39:19,312 WARNING [140078730958592] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[ INFO] [1664192359.362744067]: SELECTED BASE:Depth, 0
 26/09 19:39:19,363 WARNING [140078730958592] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[ INFO] [1664192359.372788950]: RealSense Node Is Up!
[ WARN] [1664192359.489625607]: 
 26/09 19:39:19,490 WARNING [140078730958592] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 26/09 19:39:19,541 WARNING [140078730958592] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 26/09 19:39:19,594 WARNING [140078730958592] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 26/09 19:39:19,644 WARNING [140078730958592] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 26/09 19:39:19,697 WARNING [140078730958592] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 26/09 19:39:19,747 WARNING [140078730958592] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 26/09 19:39:19,899 WARNING [140078730958592] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[... cut ...]
 26/09 19:39:20,152 WARNING [140078730958592] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 26/09 19:39:20,911 WARNING [140078730958592] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
^C[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

MartyG-RealSense, 2022-09-26:
If you installed librealsense the first time from packages then you can uninstall all RealSense related packages with the command below and then repeat the Method 1 installation.

**dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge**

Hubert-1102, 2022-09-26:
after running `dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge` 
I repeat installation in method 1 
run `sudo apt-get install ros-$ROS_DISTRO-realsense2-camera` and `sudo apt-get install ros-$ROS_DISTRO-realsense2-description`

then I reboot and test launch, same error seems happen
> roslaunch realsense2_camera rs_camera.launch
... logging to /home/hubert/.ros/log/f3b39986-3d93-11ed-835a-bf13b18585a6/roslaunch-Hubert-3124.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Hubert:40319/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
[... 88 lines cut ...]
 26/09 20:08:42,626 WARNING [140172557543168] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 26/09 20:08:42,707 WARNING [140172557543168] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 26/09 20:08:42,757 WARNING [140172557543168] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 26/09 20:08:42,808 WARNING [140172557543168] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 26/09 20:08:42,858 WARNING [140172557543168] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1664194122.909212323]: SELECTED BASE:Depth, 0
 26/09 20:08:42,909 WARNING [140172557543168] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1664194122.919087078]: RealSense Node Is Up!
[ WARN] [1664194123.066366150]: 
 26/09 20:08:43,067 WARNING [140172557543168] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 26/09 20:08:43,118 WARNING [140172557543168] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 26/09 20:08:43,170 WARNING [140172557543168] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[... cut ...]
 26/09 20:08:44,029 WARNING [140172557543168] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 26/09 20:08:44,080 WARNING [140172557543168] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
^C[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete

Hubert-1102, 2022-09-26:
And command `realsense-viewer` is not found, is that right?

MartyG-RealSense, 2022-09-26:
I believe that if the ROS wrapper's Method 1 is used for installation then tools and examples such as the Viewer are not included in the installation.  To get the Viewer, you could uninstall everything again with the purge command, then install librealsense the way you did the first time and build the ROS wrapper from source code afterwards.

MartyG-RealSense, 2022-09-26:
The **Resource temporarily unavailable** type of control_transfer returned warning indicates that there may be a communication problem with the camera.

Are you using the official 1 meter USB cable supplied with the camera or a longer cable of your own choice, please?

Also, do the warnings occur with less regularity if you use a lower resolution or FPS?

**roslaunch realsense2_camera rs_camera.launch depth_width:=640 depth_height:=480 depth_fps:=15 color_width:=640 color_height:=480 color_fps:=15**

Hubert-1102, 2022-09-26:
Yes, I use the official USB cable, and warnings didn't decrease with lower resolution

Hubert-1102, 2022-09-26:
after `roslaunch realsense2_camera rs_camera.launch ` , I tried opening rviz to see something, but I recevies some errors.

> rosrun rviz rviz
[ INFO] [1664198621.995089516]: rviz version 1.14.19
[ INFO] [1664198621.995144284]: compiled against Qt version 5.12.8
[ INFO] [1664198621.995156288]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1664198622.001979643]: Forcing OpenGl version 0.
[ INFO] [1664198622.154506639]: Stereo is NOT SUPPORTED
[ INFO] [1664198622.154567802]: OpenGL device: Mesa Intel(R) UHD Graphics (ICL GT1)
[ INFO] [1664198622.154585097]: OpenGl version: 4.6 (GLSL 4.6) limited to GLSL 1.4 on Mesa system.

**is that related with my problem?**

Hubert-1102, 2022-09-27:
Thank you very much, I will try updating Mesa drivers later.
And I found that  I can see the images published by `roslaunch realsense2_camera rs_camera.launch` with rqt_image_viewer even if  there are lots of **control_transfer returned error**. So this problem  doesn't seem to matter.

MartyG-RealSense, 2022-10-02:
hi @Hubert-1102  Do you have an update about this case that you can provide, please?  Thanks!

Akumar201, 2022-10-03:
Hello, I would like to know which USB cable you are using, control transfer error means that the data transfer is not efficient enough and loosing frames, if you are getting control transfer error initially and after which is does not reappear than you can safely ignore  it.

MartyG-RealSense, 2022-10-08:
Hi @Hubert-1102  Do you require further assistance with this case, please?  Thanks!

Hubert-1102, 2022-10-08:
Thank you, I don't need further assistance with this case.

closed 2022-10-08 by MartyG-RealSense